Koenkk/zigbee2mqtt #22628: [Wrong device]: TH008L10RGBCCT showing up as TS0505B_1

Hassi1, 2024-05-14:
### Link

https://www.zigbee2mqtt.io/devices/TH008L10RGBCCT.html#ur-lighting-th008l10rgbcct

### Model

TH008L10RGBCCT

### Description

RGB downlight

### Vendor

Ur lighting

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/TH008L10RGBCCT.png

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0xa4c138a6995ddb24","nwkAddr":20678,"manufId":4417,"manufName":"_TZ3210_hicxa0rh","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1715674317938}

### Notes

I paired a ur lighting downlight and it has shown up as the wrong device. 

Correct device: https://www.zigbee2mqtt.io/devices/TH008L10RGBCCT.html#ur-lighting-th008l10rgbcct

Currently shown as: https://www.zigbee2mqtt.io/devices/TS0505B_1.html#tuya-ts0505b_1

Koenkk, 2024-05-15:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-15 by Koenkk

Hassi1, 2024-05-21:
Hi @Koenkk,

Thanks for fixing this device model. I have noticed though, compared to my Philips hue Akari downlights the red colour is slightly off (more pinkish) at 255 R, 0 G and 0 B.

I also noticed that the colour temperature is slightly different to the Philips downlights. I ordered a Tuya zigbee gateway to debug but wanted to ask if you would happen to have any recommended tweaks to the converter that might help fix.

I attached the link to the product below:

https://urneeds.com.au/10w-90mm-smart-zigbee-rgb-cw-led-downlight-kit-for-echo-plus-smartthings-hub-hue-conbee-ii-tuya-hub/