Koenkk/zigbee2mqtt #22632: [New device support]: TRV602 with OLED

michapr, 2024-05-14:
### Link

https://de.aliexpress.com/item/1005006532990566.html

### Database entry

{"id":9,"type":"EndDevice","ieeeAddr":"0x90ab96fffe721a13","nwkAddr":63442,"manufId":4742,"manufName":"eWeLink","powerSource":"Battery","modelId":"CK-MG22-TRVZB-01(7017)","epList":[1],"endpoints":{"1":{"pr ofId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,6,32,513,64599,64529],"outClusterList":[10,25],"clusters":{"64529":{"attributes":{"24578":700}},"genBasic":{"attributes":{"appVersion":0}},"hvacThermos tat":{"attributes":{"systemMode":0,"localTemp":2250,"occupiedHeatingSetpoint":700,"runningState":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genOnOff":{"attributes":{"onOff":0}},"genPowerCf g":{"attributes":{"batteryPercentageRemaining":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x943469fffe0d65a9","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x9 43469fffe0d65a9","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x943469fffe0d65a9","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval" :65000,"repChange":10,"manufacturerCode":null},{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}}},"appVersion":0,"dateCode":"20230811","swBuil dId":"1.1.1","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1715696543167,"checkinInterval":3600}

### Comments

Have tried to add the device, but it is "unsupported".
Can only add the store file because have not found a product page.

Because it is TRV602 (in normal case supported) - I think that maybe only an "alias" could be added - not sure of course.
(and not created a new definition file)

If I can test/check anything more, please tell me.

### External definition

```shell
const {battery, identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['CK-MG22-TRVZB-01(7017)'],
    model: 'CK-MG22-TRVZB-01(7017)',
    vendor: 'eWeLink',
    description: 'Automatically generated definition',
    extend: [battery(), identify(), onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

Dinth, 2024-05-18:
Hmmm could you post the picture of your TRV? I have been using TRV602 with OLED for the past 6+ months without any additional converters and without any big problems.

michapr, 2024-05-18:
[image]

TRV602-EWE-ZigBee

michapr, 2024-05-23:
wrong topic - device (TRV602) is supported - but not this subtype(?)

closed 2024-05-23 by michapr