Koenkk/zigbee2mqtt #22633: Tuya TZ3000_kqvb5akv seMetering -> notificationControlFlags instead of currentSumDeliverd

FreddyX078, 2024-05-14:
### What happened?

After a new build this morning:  **d39baa6  dev**  my TZ3000_kqvb5akv don't report energy. 
Last energy reading was from before the update.

When looking at Device -> Report, there is Cluster -> seMetering with Attribute -> notificationControlFlags.
But when I apply some modification, I got 

`Request 'z2m_odroidn2/bridge/request/device/configure_reporting' failed with error: 'ZCL command 0xdevice/1 seMetering.configReport([{"attribute":"notificationControlFlags","minimumReportInterval":300,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'
`

And when I try in Developpersconsole seMetering -> currentSummDelivered, the output is: 
_Read result of 'seMetering': {"notificationControlFlags":[0,48]}_

Status from new added device:
{
    "current": 0.12,
    "last_seen": "2024-05-14T14:54:13.319Z",
    "linkquality": 127,
    "power": 16,
    "power_outage_memory": "on",
    "state": "ON",
    "switch_type": "toggle",
    "voltage": 233,
    "energy": null
}

### What did you expect to happen?

I expected energy report like before ;-)

### Zigbee2MQTT version

1.37.1-dev commit: d39baa6

### Adapter firmware version

20240315

### Adapter

Sonoff Dongle-P

### Setup

Odroid N2+ with Armbian

001Tycoon, 2024-05-15:
Hello, I have the same problem with my Smart plug (with power monitoring) (TS011F_plug_1) from TuYa, no update of the total energy.
{
     "child_lock": "UNLOCK",
     "current": 0.523,
     "energy": 68.37,
     "indicator_mode": "off",
     "last_seen": "2024-05-15T09:18:34+02:00",
     "link quality": 18,
     "power": 85,
     "power_outage_memory": "on",
     "state": "ON",
     "update": {
         "installed_version": 192,
         "latest_version": 192,
         "state": "idle"
     },
     "update_available": false,
     "voltage": 231,
     "countdown": zero
}
energy has been at the same value since yesterday
the other values are OK
Zigbee2MQTT-Version 1.37.1-dev commit: [2be947b]
Coordinator-Typ    zStack3x0
Coordinator-Version    20230507
ZBDongle-P

001Tycoon, 2024-05-28:
Thanks, with the last commit it works again.

closed 2024-05-28 by FreddyX078