Koenkk/zigbee2mqtt #22635: z2m: Error: NCP EZSP protocol version of 12 does not match Host version 13

thierryzoller, 2024-05-14:
### What happened?

Homassistant Green

Skyconnect (Updated to latest firmware using Website today - as recommended)

Settings : 
port: /dev/ttyUSB2
adapter: ember
rtscts: true

[17:49:35] INFO: Preparing to start...
[17:49:36] INFO: Socat not enabled
[17:49:40] ERROR: Got unexpected response from the API: Service not enabled
[17:49:40] INFO: Starting Zigbee2MQTT...
[2024-05-14 17:49:50] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-14 17:49:50] info: 	z2m: Starting Zigbee2MQTT version 1.37.1 (commit #unknown)
[2024-05-14 17:49:50] info: 	z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-14 17:49:50] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-14 17:49:50] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-14 17:49:50] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-14 17:49:51] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-14 17:49:51] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-14 17:49:52] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-14 17:49:52] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-14 17:49:52] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-14 17:49:52] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-14 17:49:52] error: 	z2m: Failed to start zigbee
[2024-05-14 17:49:52] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-14 17:49:52] error: 	z2m: Exiting...
[2024-05-14 17:49:52] error: 	z2m: Error: NCP EZSP protocol version of 12 does not match Host version 13
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1633:19)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:784:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2642:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts

### What did you expect to happen?

I expected it functioning and recognising adapter :)

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

20240501.1

### Adapter

Skyconnect

### Setup

Homeassistant Green

wastez, 2024-05-14:
Looks like the firmware is v12 and not v13, which firmware did you install?
This one?
https://github.com/darkxst/silabs-firmware-builder/blob/main/firmware_builds/skyconnect/ncp-uart-hw-v7.4.2.0-skyconnect-115200.gbl

D34DC3N73R, 2024-05-14:
I get this error when using the multipan FW v4.3.1 and Z2M with `adapter: ember` instead of `adapter: ezsp`.

wastez, 2024-05-15:
@D34DC3N73R 
Multipan is not supported on ember and I guess it never will be because of instability in bigger systems. 
Let’s see what will be with multipan in future because as far as I know nabu casa came to the insight that this is not going to work as they thought (because of the slow speed of serial devices) and know they suggest to use a single adapter for every protocol. 
So if @thierryzoller is not using multipan it has not hing to do with it.

thierryzoller, 2024-05-15:
@wastez : It is "Zigbee (EZSP) 7.3.2.0 build 212"

here is a step by step
1. Browse to https://skyconnect.home-assistant.io/firmware-update/
2. Plugin Sykconnect
3. Click Update Firmware
4. Result :  Zigbee (EZSP) 7.3.2.0 build 212

PS - This was the firmware given as "Latest" automatically by the website, and that I was recommended to upgrade to.

[image]

thierryzoller, 2024-05-15:
Should I try to flash the one linked above ? @wastez

wastez, 2024-05-15:
@thierryzoller 
7.3.2 will not work. 
you have to install a 7.4 version.

wastez, 2024-05-15:
@thierryzoller 
Here is also a webflasher like yours which holds the actual firmware (not official)for your device. 
https://darkxst.github.io/silabs-firmware-builder/

thierryzoller, 2024-05-15:
Thank you very much @wastez

dierochade, 2024-05-16:
For Skyconnect: 

I was able to flash the firmware linked by wastez (https://darkxst.github.io/silabs-firmware-builder/) simply by downloading and using in the official web flash tool (https://skyconnect.home-assistant.io/firmware-update/). 

Worked like a charm.

thierryzoller, 2024-05-16:
I can confirm that one can flash custom firmware in general including the one linked above

Midifreakz, 2024-05-16:
> For Skyconnect:
>
> I was able to flash the firmware linked by wastez [...]

This also seemed to have worked for me. Thank you!

closed 2024-05-19 by thierryzoller

jasonhollis, 2024-06-29:
> @thierryzoller Here is also a webflasher like yours which holds the actual firmware (not official)for your device. https://darkxst.github.io/silabs-firmware-builder/
> 
> Edit: Sorry your device is not supported on the webflasher. You have to use the file I posted above and the silabs universal firmware flasher from here: https://github.com/NabuCasa/universal-silabs-flasher

I just used this website to update my Sonoff E Dongle to the latest version and I'm getting the same error:

z2m: Error: NCP EZSP protocol version of 12 does not match Host version 13
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-

I flashed it on my Mac and then shoved it back into the HA Box so I know it got rebooted.  I do have two adapters in the box one for MQTT and another for the regular direct connect Zigbee.

mcicolella, 2024-09-16:
I have the same problem with my Sonoff E Dongle updated to the latest firmware version.

z2m: Error: NCP EZSP protocol version of 12 does not match Host version [13-14]
at EmberAdapter.emberVersion (/app/node_modules/zigbee-

Is there any fix for this?

jasonhollis, 2024-09-16:
Yep,

Update the firmware.  How good are you with SSH?

mcicolella, 2024-09-16:
Just updated with https://darkxst.github.io/silabs-firmware-builder/
I'm good with SSH