Koenkk/zigbee2mqtt #22639: _TZE204_x8fp01wi not working as expected

derkorte, 2024-05-14:
### What happened?

The device shows right values on display, but not in home assistant/ z2m

### How to reproduce it (minimal and precise)

That the device gives out values like consumption (500W) minus PV production (1500W) is value -1000W

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

ConBee_II_DE2686439-if00

### Adapter

ConBee_II

### Setup

Home assistant OS PI 5

derkorte, 2024-05-27:
@Koenkk no idea?

eokgnah, 2024-07-16:
Same here.
I ordered this one:
https://www.aliexpress.com/item/1005005995625555.html

Zigbee2MQTT: _TZE204_x8fp01wi

The values are looking (almost) ok.
```
payload '{"current_a":1.341,"current_b":0.235,"current_c":0.646,"energy":1164.97,"last_seen":"2024-07-16T13:59:38+02:00","linkquality":65,"power":417,"power_a":310,"power_b":29,"power_c":77,"produced_energy":41.46,"voltage_a":235.3,"voltage_b":233.9,"voltage_c":234.4}'
```

I see energy (IN), and produced_energy (OUT).
But beside the global in/out values, there is the problem with the missing negative values (or any other indicator of the powerflow direction).

i tried debugging the raw values, but i dont know what to look for or where to add this to get the searched power direction?
for example i got in debigging:
```
 type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,40,0,2,134,0,0,77],"type":"Buffer"},"datatype":0,"dp":8}],"seq":43634}' from endpoint 1 with groupID 0
 type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,161],"type":"Buffer"},"datatype":2,"dp":9},{"data":{"data":[0,1,199,17],"type":"Buffer"},"datatype":2,"dp":1},{"data":{"data":[0,0,16,50],"type":"Buffer"},"datatype":2,"dp":2}],"seq":43890}' from endpoint 1 with groupID 0
 type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
 type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0

```

If i can help solve the riddle, pls tell me what to do. i will try to get the needed infos.

If it is not possible, is there another 3-phase meter which does the values correct?

derkorte, 2024-07-16:
I think this is a problem of Homeassistent itself. HA doesn't support negative values at all. Just if you create a template sensor and take sensor a minus sensor b you get negative values. But the power output cumulates the power an gives out an positive value, so you can't just take Produktion minus power. Seems like @Koenkk is the only one who may can help.

eokgnah, 2024-07-17:
> I think this is a problem of Homeassistent itself. HA doesn't support negative values at all. Just if you create a template sensor and take sensor a minus sensor b you get negative values. But the power output cumulates the power an gives out an positive value, so you can't just take Produktion minus power. Seems like @Koenkk is the only one who may can help.
> 
> Edit: I've read the modbus output and there's a negative output on channel 45. If I integrate it by esp Home, the negative value is not shown correctly. So this isn't the solution.

I dont use HA. i git the values direct from the Zigbee2MQTT debug-Log.

I think that the used base conversion is not the right one for the meter.

The meter we have problems with was introduced here as a "just add the fingerprint, the values look ok":
https://github.com/Koenkk/zigbee-herdsman-converters/commit/64c18bf408c62d77a86551bb0162224e95533bee

But i think this might be the root of the problem.
Since i dont have a clue how to fix it by myself, i tried to get help with evaluating the conversion by sending the debug output i got.
Maybe a different (own) conversion is needed for _TZE204_x8fp01wi to get the negative values.

_TZE200_x8fp01wi = _TZE204_x8fp01wi is not working correct.

derkorte, 2024-07-17:
The channel 45 for modbus, not zigbee ,,,☝️

Koenkk, 2024-07-18:
What are the values for `produced_energy` and `energy` and what are the expected ones?

eokgnah, 2024-07-19:
> What are the values for `produced_energy` and `energy` and what are the expected ones?

the values look ok.
```
 payload '{
"current_a":0.507,
"current_b":0.233,
"current_c":0.567,
"energy":1209.97,
"last_seen":"2024-07-19T09:41:17+02:00",
"linkquality":69,
"power":133,
"power_a":96,
"power_b":28,
"power_c":8,
"produced_energy":42.87,
"voltage_a":236.3,
"voltage_b":235.2,
"voltage_c":236
}'
```

The problem are NOT the energy and produced_energy values.
The problem is that the mqtt current and power values never go negative, even the display shows them negative (direction).
[image]

So i tried to get the raw values which are coming over zigbee so see if the problem is:
a) the device is sending wrong values over zigbee. (this we canot fix)
b) the sent values are correct, but the conversion in zigbee2mqtt misses something to accomodate the direction. (this we could fix)

I tried setting the loglevel to debug, but i didnt got raw data?
Maybe i'm doing something wrong.

Woild like to help fix or clear this for all, but need a little help myself in setting up the debugging.

eokgnah, 2024-07-19:
Unlogical finding in the json:

```
{"current_a":0.745,"current_b":0.233,"current_c":0.636,"energy":1210.03,"last_seen":"2024-07-19T11:04:55+02:00","linkquality":61,"power":155,"power_a":164,"power_b":28,"power_c":59,"produced_energy":42.99,"voltage_a":234.3,"voltage_b":234.1,"voltage_c":234.5}
```

if i extract the voltage/current/power values and group them by phases.
```
"voltage_a":234.3
"current_a":0.745
"power_a":164

"voltage_b":234.1
"current_b":0.233
"power_b":28

"voltage_c":234.5
"current_c":0.636
"power_c":59

"power":155
```

I expect: P = U * I
A) 234.3 * 0.745 = **174.5W**  ("power_a":**164** ?)
B) 234.1 * 0.233 =  **54.5W**  ("power_b":**28** ?)
C) 234.5 * 0.636 = **149.1W**  ("power_c":**59** ?)

And i expect Pa + Pb + Pc = POWER
MQTT-Values: 164W + 28W + 59W = **251W**
or
calculated values: 174.5W + 54.5W + 149.1W = **378.1W**  
which both differs from the mqtt value ("power":**155** ?)

Looks a bit fishy to me.
I canot pinpoint the error at the moment, but i strongly think there might be one.

Maybe the missing "minus" while producing is the solution?
This would shift the values a bit wround?
Or it is another, unrelated Problem?

Koenkk, 2024-07-21:
> I tried setting the loglevel to debug, but i didnt got raw data?

To get the raw values you have to enable debug logging. Could you share those when the device reports a change in `power` and the should be negative?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

derkorte, 2024-07-21:
Like this?

2024-07-21 14:09:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Smart Meter', payload '{"current_a":5.06,"current_b":1.947,"current_c":0.39,"energy":3129.78,"last_seen":"2024-07-21T14:09:38+02:00","linkquality":255,"power":1591,"power_a":1180,"power_b":400,"power_c":10,"produced_energy":2054.34,"state":"ON","voltage_a":236.1,"voltage_b":235.4,"voltage_c":235.9}'

derkorte, 2024-07-21:
Or did you want the full file for log output?

eokgnah, 2024-07-22:
> To get the raw values you have to enable debug logging. Could you share those when the device reports a change in `power` and the should be negative?

i will send it if the weather gets better.
since i enabled debug, it is rainy and it does not get negative. :-(

eokgnah, 2024-07-22:
Here i had a small brighter window in the clouds,
the energy_produced is rising faster then energy. so there is more produced then consumed.

produced: 44.41 - 44.37 = 40wH produced
consumed: 1232.36 - 1232.36 = 0Wh consumed

This implies to me that there must be negative values in this log.
Canot confirm 100% because im not at home right now to look at the meter.
I only have the log.

I only feed on A and C.
B has no solar pannel attached.
So only A&C can go negative, but not B.

Looking forward to more sun!

```
Jul 22 12:40:22 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:40:22: Received Zigbee message from 'Sicherungskasten', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
Jul 22 12:40:22 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:40:22: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.648,"current_b":0.226,"current_c":0.638,"energy":1232.36,"last_seen":"2024-07-22T12:40:22+02:00","linkquality":61,"power":176,"power_a":133,"power_b":27,"power_c":15,"produced_energy":44.37,"state":"ON","voltage_a":233.2,"voltage_b":232.8,"voltage_c":232.5}'
Jul 22 12:40:55 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:40:55: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,39,0,3,200,0,0,215],"type":"Buffer"},"datatype":0,"dp":6}],"seq":47380}' from endpoint 1 with groupID 0
Jul 22 12:40:55 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:40:55: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.968,"current_b":0.226,"current_c":0.638,"energy":1232.36,"last_seen":"2024-07-22T12:40:55+02:00","linkquality":65,"power":176,"power_a":215,"power_b":27,"power_c":15,"produced_energy":44.37,"state":"ON","voltage_a":234.3,"voltage_b":232.8,"voltage_c":232.5}'
Jul 22 12:40:55 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:40:55: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,37,0,0,225,0,0,26],"type":"Buffer"},"datatype":0,"dp":7}],"seq":47636}' from endpoint 1 with groupID 0
Jul 22 12:40:55 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:40:55: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.968,"current_b":0.225,"current_c":0.638,"energy":1232.36,"last_seen":"2024-07-22T12:40:55+02:00","linkquality":58,"power":176,"power_a":215,"power_b":26,"power_c":15,"produced_energy":44.37,"state":"ON","voltage_a":234.3,"voltage_b":234.1,"voltage_c":232.5}'
Jul 22 12:40:55 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:40:55: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,30,0,2,226,0,0,98],"type":"Buffer"},"datatype":0,"dp":8}],"seq":47892}' from endpoint 1 with groupID 0
Jul 22 12:40:55 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:40:55: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.968,"current_b":0.225,"current_c":0.738,"energy":1232.36,"last_seen":"2024-07-22T12:40:55+02:00","linkquality":61,"power":176,"power_a":215,"power_b":26,"power_c":98,"produced_energy":44.37,"state":"ON","voltage_a":234.3,"voltage_b":234.1,"voltage_c":233.4}'
Jul 22 12:40:56 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:40:56: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,83],"type":"Buffer"},"datatype":2,"dp":9},{"data":{"data":[0,1,225,100],"type":"Buffer"},"datatype":2,"dp":1},{"data":{"data":[0,0,17,86],"type":"Buffer"},"datatype":2,"dp":2}],"seq":48148}' from endpoint 1 with groupID 0
Jul 22 12:40:56 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:40:56: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.968,"current_b":0.225,"current_c":0.738,"energy":1232.36,"last_seen":"2024-07-22T12:40:56+02:00","linkquality":65,"power":339,"power_a":215,"power_b":26,"power_c":98,"produced_energy":44.38,"state":"ON","voltage_a":234.3,"voltage_b":234.1,"voltage_c":233.4}'
Jul 22 12:42:55 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:42:55: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,43,0,2,2,0,0,96],"type":"Buffer"},"datatype":0,"dp":6}],"seq":48404}' from endpoint 1 with groupID 0
Jul 22 12:42:55 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:42:55: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.514,"current_b":0.225,"current_c":0.738,"energy":1232.36,"last_seen":"2024-07-22T12:42:55+02:00","linkquality":69,"power":339,"power_a":96,"power_b":26,"power_c":98,"produced_energy":44.38,"state":"ON","voltage_a":234.7,"voltage_b":234.1,"voltage_c":233.4}'
Jul 22 12:42:55 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:42:55: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,42,0,0,228,0,0,27],"type":"Buffer"},"datatype":0,"dp":7}],"seq":48660}' from endpoint 1 with groupID 0
Jul 22 12:42:55 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:42:55: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.514,"current_b":0.228,"current_c":0.738,"energy":1232.36,"last_seen":"2024-07-22T12:42:55+02:00","linkquality":61,"power":339,"power_a":96,"power [... cut ...]
Jul 22 12:49:00 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:49:00: Received Zigbee message from 'Sicherungskasten', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
Jul 22 12:49:00 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:49:00: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.97,"current_b":0.227,"current_c":0.755,"energy":1232.36,"last_seen":"2024-07-22T12:49:00+02:00","linkquality":61,"power":358,"power_a":216,"power_b":27,"power_c":114,"produced_energy":44.41,"state":"ON","voltage_a":234,"voltage_b":233.6,"voltage_c":233.8}'
```

Koenkk, 2024-07-23:
I don't think it reports negative values, from your log:

```
Jul 22 12:48:55 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-22 12:48:55: Received Zigbee message from 'Sicherungskasten', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,36,0,3,202,0,0,216],"type":"Buffer"},"datatype":0,"dp":6}],"seq":51476}' from endpoint 1 with groupID 0
Jul 22 12:48:55 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-22 12:48:55: MQTT publish: topic 'zigbee2mqtt/Sicherungskasten', payload '{"current_a":0.97,"current_b":0.228,"current_c":0.615,"energy":1232.36,"last_seen":"2024-07-22T12:48:55+02:00","linkquality":65,"power":141,"power_a":216,"power_b":27,"power_c":37,"produced_energy":44.4,"state":"ON","voltage_a":234,"voltage_b":233.8,"voltage_c":233.1}'
```

Here it report datapoint 6, which is the voltage, current and power for phase A, the data is `[9,36,0,3,202,0,0,216]`. The last 2 numbers represent the `power` which is calculated as follows `data[7] | (data[6] << 8)`, this gives us a result of `216` which is published. So I'm not sure how we can determine wether the `power` value is negative or positive

derkorte, 2024-07-23:
I have no idea how to help. 
I just can say that device shows correct value on display and via modbus output.
If I can help, I just need a little more specific manual...
Would it be helpful to connect it to any tuya gateway to check the output value?

Koenkk, 2024-07-24:
Would be indeed interesting to see if the correct values are reported via the Tuya gateway

eokgnah, 2024-07-26:
Unfortunately i dont have access to a Tuya gateway.
Maybe the "number" itself is correct but the "+/-" is hidden as maybe an boolian somwhere else in the dataset?
Already tried a bit to compare logs from day/night to identify possible booleans, but by now without luck.
But i keep digging!
Is there a description beside the one in tuya.ts where i can see which bytes are which known value?
Are all Bits from the data string used or are there unknown bits/bytes which i/we could focus on.

I got some sun the last days.
And i got a 2nd device which is not in "production" use at the moment.
Will try to narrow down the issue on the weekend.

eokgnah, 2024-07-27:
### findings ###
1st - i pulled on each phase about 12A.
i see it clearly in the readings:
```
Jul 27 16:41:06 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 16:41:06: Received Zigbee message from 'Garage', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
Jul 27 16:41:06 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 16:41:06: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.214,"current_b":12.12,"current_c":12.142,"energy":69.66,"last_seen":"2024-07-27T16:41:06+02:00","linkquality":65,"power":8506,"power_a":2857,"power_b":2820,"power_c":2828,"produced_energy":0,"state":"ON","voltage_a":234.2,"voltage_b":233.1,"voltage_c":233.1}'
Jul 27 16:43:02 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 16:43:02: Received Zigbee message from 'Garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,43,0,47,94,0,11,24],"type":"Buffer"},"datatype":0,"dp":6}],"seq":50256}' from endpoint 1 with groupID 0
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 16:43:03: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.126,"current_b":12.12,"current_c":12.142,"energy":69.66,"last_seen":"2024-07-27T16:43:02+02:00","linkquality":29,"power":8506,"power_a":2840,"power_b":2820,"power_c":2828,"produced_energy":0,"state":"ON","voltage_a":234.7,"voltage_b":233.1,"voltage_c":233.1}'
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 16:43:03: Received Zigbee message from 'Garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,28,0,47,53,0,10,254],"type":"Buffer"},"datatype":0,"dp":7}],"seq":50512}' from endpoint 1 with groupID 0
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 16:43:03: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.126,"current_b":12.085,"current_c":12.142,"energy":69.66,"last_seen":"2024-07-27T16:43:03+02:00","linkquality":40,"power":8506,"power_a":2840,"power_b":2814,"power_c":2828,"produced_energy":0,"state":"ON","voltage_a":234.7,"voltage_b":233.2,"voltage_c":233.1}'
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 16:43:03: Received Zigbee message from 'Garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,30,0,47,55,0,11,2],"type":"Buffer"},"datatype":0,"dp":8}],"seq":50768}' from endpoint 1 with groupID 0
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 16:43:03: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.126,"current_b":12.085,"current_c":12.087,"energy":69.66,"last_seen":"2024-07-27T16:43:03+02:00","linkquality":25,"power":8506,"power_a":2840,"power_b":2814,"power_c":2818,"produced_energy":0,"state":"ON","voltage_a":234.7,"voltage_b":233.2,"voltage_c":233.4}'
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 16:43:03: Received Zigbee message from 'Garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,33,32],"type":"Buffer"},"datatype":2,"dp":9},{"data":{"data":[0,0,27,82],"type":"Buffer"},"datatype":2,"dp":1},{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":2}],"seq":51024}' from endpoint 1 with groupID 0
Jul 27 16:43:03 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 16:43:03: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.126,"current_b":12.085,"current_c":12.087,"energy":69.94,"last_seen":"2024-07-27T16:43:03+02:00","linkquality":29,"power":8480,"power_a":2840,"power_b":2814,"power_c":2818,"produced_energy":0,"state":"ON","voltage_a":234.7,"voltage_b":233.2,"voltage_c":233.4}'```

2nd - i reversed in/out on Lb and measured again
```
Jul 27 17:08:50 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 17:08:50: Received Zigbee message from 'Garage', type 'attributeReport', cluster 'genBasic', data '{"65506":56,"65508":0,"appVersion":74}' from endpoint 1 with groupID 0
Jul 27 17:08:50 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 17:08:50: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.177,"current_b":12.094,"current_c":12.148,"energy":70.5,"last_seen":"2024-07-27T17:08:50+02:00","linkquality":65,"power":8464,"power_a":2840,"power_b":2806,"power_c":2814,"produced_energy":0.22,"state":"ON","voltage_a":233.6,"voltage_b":232,"voltage_c":231.8}'
Jul 27 17:10:02 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 17:10:02: Received Zigbee message from 'Garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,28,0,47,155,0,11,14],"type":"Buffer"},"datatype":0,"dp":6}],"seq":7424}' from endpoint 1 with groupID 0
Jul 27 17:10:02 rp4 npm[29483]: Zigbee2MQTT:info  2024-07-27 17:10:02: MQTT publish: topic 'zigbee2mqtt/Garage', payload '{"current_a":12.187,"current_b":12.094,"current_c":12.148,"energy":70.5,"last_seen":"2024-07-27T17:10:02+02:00","linkquality":65,"power":8464,"power_a":2830,"power_b":2806,"power_c":2814,"produced_energy":0.22,"state":"ON","voltage_a":233.2,"voltage_b":232,"voltage_c":231.8}'
Jul 27 17:10:03 rp4 npm[29483]: Zigbee2MQTT:debug 2024-07-27 17:10:03: Received Zigbee message from 'Garage', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[9,21,0,47,114,0,11,2],"type":"Buffer"},"datatype":0,"dp":7}],"seq":7680}' from endpoint 1 with groupID 0
[... cut ...]
V Volt L3 = positive
A Ampere L1 = positive
A Ampere L2 = positive
A Ampere L2 = positive
W Power L1 = positive
W Power L2 = **NEGATIVE**
W Power L3 = positive
PF Powerfactor L1 = positive
PF Powerfactor L2 = **NEGATIVE**
PF Powerfactor L3 = positive
```

### conclusion ###
if there is a chance to see a negative value, we have to check the kWh power readings or we find the powerfactor. only those values are negative.

now i will look into the binary data. 
maybe i will find a boolean for the power sign reading.

eokgnah, 2024-07-27:
i identified:

Datatype 0 - dp 6, 7 and 8
Datatype 2 - dp 1, 2 and 9

```
{"data":{"data":[ 9,28, 0,47,155, 0,11,14],"type":"Buffer"},"datatype":0,"dp":6}]
                  V A     A A      W A
				  
{"data":{"data":[ 9,21, 0,47,114, 0,11,2],"type":"Buffer"},"datatype":0,"dp":7}]
                  V B     A B      W B
				  
{"data":{"data":[ 9,11, 0,47,95, 0,10,245],"type":"Buffer"},"datatype":0,"dp":8}]
                  V C     A C      W C
				  
{"data":{"data":[ 0,0, 33,7],"type":"Buffer"},"datatype":2,"dp":9},
                   ??  POWER
					   
{"data":{"data":[ 0,0, 27,156],"type":"Buffer"},"datatype":2,"dp":1},
                   ??  ENERGY
						
{"data":{"data":[ 0,0, 0,32],"type":"Buffer"},"datatype":2,"dp":2}]
                   ??  PRODUCED
```

Could there maybe be more Datatypes/db which are not parsed by Zigbee2MQTT?

Koenkk, 2024-07-28:
All of these are handled, these are the current mapped datapoints:

```
                [16, 'state', tuya.valueConverter.onOff],
                [1, 'energy', tuya.valueConverter.divideBy100],
                [2, 'produced_energy', tuya.valueConverter.divideBy100],
                [9, 'power', tuya.valueConverter.raw],
                [6, null, tuya.valueConverter.phaseVariant2WithPhase('a')],
                [7, null, tuya.valueConverter.phaseVariant2WithPhase('b')],
                [8, null, tuya.valueConverter.phaseVariant2WithPhase('c')],
```

In case a datapoint is received which is not mapped, you will see the following the z2m debug log: `Datapoint XX not defined for`

eokgnah, 2024-07-29:
> In case a datapoint is received which is not mapped, you will see the following the z2m debug log: `Datapoint XX not defined for`

not quite.

i only have:
```
pi@rp4:~ $ grep -i "defined" /var/log/syslog | tail -n2
Jul 29 07:46:14 rp4 Node-RED[27114]: 29 Jul 07:46:14 - [error] [function:function 1] TypeError: Cannot read properties of undefined (reading '3h')
Jul 29 07:56:14 rp4 Node-RED[27114]: 29 Jul 07:56:14 - [error] [function:function 1] TypeError: Cannot read properties of undefined (reading '3h')
```

but i think this just relates to a spare, offline device and has nothing to do with the missing +/- case.
```
Jul 29 06:56:44 rp4 npm[29483]: Zigbee2MQTT:warn  2024-07-29 06:56:44: Failed to ping 'Solar-reserve' (attempt 1/1, ZCL command 0xa4c13838bf862775/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233)))
```

eokgnah, 2024-07-29:
Now i ordered the WIFI version of it to compare.
https://de.aliexpress.com/item/1005005995625555.html

eokgnah, 2024-07-29:
> All of these are handled, these are the current mapped datapoints:
>
> ```
>                 [16, 'state', tuya.valueConverter.onOff],
>                 [1, 'energy', [...]

While playing around and switching OFF/ON over the 16 datapoint, i saw a new datapoint!

```
Jul 29 12:02:44 rp4 npm[29304]: Zigbee2MQTT:debug 2024-07-29 12:02:44: Received Zigbee message from 'Garage', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":16},{"data":{"data":[50,52,50,51,48,52,51,57],"type":"Buffer"},"datatype":3,"dp":18}],"seq":0}' from endpoint 1 with groupID 0
```

50 looks at first glance like the frequency here in europe, but i dont have a clue why there are so many readings around the value 50 and why this one (18) only comes after the relay is switched off or back on.

The payload [50,52,50,51,48,52,51,57] is in both cases the same.

derkorte, 2024-07-29:
Today I've found the time to connect it to the tuya gateway. 
It seems like there's no negative value via zigbee. 
[image]
[image]
[image]
[image]

derkorte, 2024-07-31:
Earu isn't the manufacturer from the device, it's just the device with a earu brand.
The manufacturer is Xoca. I've wrote a few Mails with the support but they can't help, cause it's a problem of tuya zigbee chip inside. But they send me the modbus register list. Thanks for the great and fast support! 
Now I've had created an bug report via tuya app for the total power output. 
If I get some news I will let you know!

eokgnah, 2024-08-01:
> But they send me the modbus register list.

@derkorte 
Can you send me this list please?
I plan to use a esp32 as alternative bridge to mqtt.

derkorte, 2024-08-02:
Got it working. Just Copy the raws

https://github.com/derkorte/XOCA-DAC7361C-Mod-Bus-EARU-3-phaseTcp-integration

Perfect-Web, 2024-09-20:
I wonder if i am the only one that has this problem. I don't get any readings at all ...
[image]

derkorte, 2024-09-20:
I think so. Check all connections described on my GitHub page.

closed 2025-01-02 by github-actions[bot]