Koenkk/zigbee2mqtt #22643: TuYa Smoke sensor (TS0205) not connect

EnziinSystem, 2024-05-15:
### What happened?

The TuYa Smoke sensor (TS0205) has previously connected and worked normally.

Now when I install on another system and clone the latest zigbee2mqtt, the device connection error occurs.

```
May 15 13:54:24 smarthome node[955]: [2024-05-15 13:54:24] info:         zh:controller: Interview for '0x84fd27fffe694d07' started
May 15 13:54:24 smarthome node[955]: [2024-05-15 13:54:24] info:         z2m: Device '0x84fd27fffe694d07' joined
May 15 13:54:24 smarthome node[955]: [2024-05-15 13:54:24] info:         z2m: Starting interview of '0x84fd27fffe694d07'
May 15 13:54:35 smarthome node[955]: [2024-05-15 13:54:35] info:         zh:controller: Succesfully interviewed '0x84fd27fffe694d07'
May 15 13:54:35 smarthome node[955]: [2024-05-15 13:54:35] info:         z2m: Successfully interviewed '0x84fd27fffe694d07', device has successfully been paired
May 15 13:54:35 smarthome node[955]: [2024-05-15 13:54:35] info:         z2m: Device '0x84fd27fffe694d07' is supported, identified as: TuYa Smoke sensor (TS0205)
May 15 13:54:35 smarthome node[955]: [2024-05-15 13:54:35] info:         z2m: Configuring '0x84fd27fffe694d07'
May 15 13:55:01 smarthome node[955]: [2024-05-15 13:55:01] error:         z2m: Failed to configure '0x84fd27fffe694d07', attempt 3 (Error: ZCL command 0x84fd27fffe694d07/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 129 - 1 - 18 - 1 - 7 after 10000ms)
May 15 13:55:01 smarthome node[955]:     at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
May 15 13:55:01 smarthome node[955]:     at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:498:47)
May 15 13:55:01 smarthome node[955]:     at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
May 15 13:55:01 smarthome node[955]:     at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20)
May 15 13:55:01 smarthome node[955]:     at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:766:28)
May 15 13:55:01 smarthome node[955]:     at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9)
May 15 13:55:01 smarthome node[955]:     at setupAttributes (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:76:13)
May 15 13:55:01 smarthome node[955]:     at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:17)
May 15 13:55:01 smarthome node[955]:     at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
May 15 13:55:01 smarthome node[955]:     at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:121:13))

```

### Zigbee2MQTT version

1.37.1 commit: cda867a

### Adapter firmware version

20230507

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Raspberry Pi-5

Nerivec, 2024-05-18:
Battery-powered devices can require manually pushing a button on them to prevent them from sleeping whenever an action has to be sent by Z2M. You may have to repeat pushing the button at 2-3 seconds interval until the configuring is done.

RuSPanzer, 2024-07-02:
hi. same problem

```
2024-07-02 23:55:43z2m: Failed to configure 'Датчик дыма кухня', attempt 2 (Error: ZCL command 0xbc33acfffe521f0c/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 11230 - 1 - 14 - 1 - 7 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:498:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:767:28) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:557:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:107:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:152:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:190:21) at Configure.configure (/app/lib/extension/configure.ts:126:13))
```

CC2652P  coordinator with FW 20230507 
zStack3x0

smoke detector TS0205 (_TYZB01_iev4cs4x)

closed 2025-01-02 by github-actions[bot]