Koenkk/zigbee2mqtt #22645: Issue I'm losing all the connection between all the devices Z2M Version 1.37.1-1

MiAutomations, 2024-05-15:
### What happened?

Hello 

After a restore to try solve a issue as mentioned in this thread [https://github.com/Koenkk/zigbee2mqtt/issues/22616](url)

I'm facing now a news issue, the devices from the IKEA as successfully recognized as a supported device, but I start facing a issue and a unstable zigbee network with droop connections to all the devices 

After the restore I performed the Update to the last Z2M version Current version: 1.37.1-1 with success, but all the devices are now unlinked or offline 

When I restart the HA or the Host or even the Addon I have again the devices online but only during a little time 30 minute after this all the devices in the map apear unlink and the devices goes to offline 

I try restart the addon, the HA , and also the host and unplug the Sonoff Dongle P plus from the host and plug it again but without success

When I restart the addon or the HA or host this is the map 

[image]

After 30 minute this is the map 

[image]

Can anyone please help me in order to try solve this issue ? Thank you in advance for your help !
       

### What did you expect to happen?

I expect the Z2M works well like until now ...

### How to reproduce it (minimal and precise)

As I mentioned in the last 

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

 20221226

### Adapter

Sonoff Dongle Plus P

### Setup

Mini PC running HAOS

### Debug log

[2024-05-15 09:10:27] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655277/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
[2024-05-15 09:10:30] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655150/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-15 09:10:30] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x84ba20fffe655150:56368/1 (0,0,1)
[2024-05-15 09:10:30] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":56368,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":196,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,144,0,5,5,8,5,11,5]}}
[2024-05-15 09:10:30] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,48,220,1,1,4,11,196,0,30,9,16,144,0,5,5,8,5,11,5,133]
[2024-05-15 09:10:36] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655150/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
[2024-05-15 09:11:06] debug: 	zh:controller:endpoint: ZCL command 0x70ac08fffe997b05/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-15 09:11:06] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x70ac08fffe997b05:53136/1 (0,0,1)
[2024-05-15 09:11:06] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":53136,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":197,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,145,0,5,5,8,5,11,5]}}
[2024-05-15 09:11:06] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,144,207,1,1,4,11,197,0,30,9,16,145,0,5,5,8,5,11,5,54]
[2024-05-15 09:11:12] debug: 	zh:controller:endpoint: ZCL command 0x70ac08fffe997b05/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
[2024-05-15 09:11:21] debug: 	zh:controller:endpoint: ZCL command 0x04cd15fffe02c8ed/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-15 09:11:21] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x04cd15fffe02c8ed:34293/1 (0,0,1)
[2024-05-15 09:11:21] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":34293,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":198,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,146,0,5,5,8,5,11,5]}}
[2024-05-15 09:11:21] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,245,133,1,1,4,11,198,0,30,9,16,146,0,5,5,8,5,11,5,25]
[2024-05-15 09:11:27] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655277/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-15 09:11:27] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x84ba20fffe655277:21313/1 (0,0,2)
[2024-05-15 09:11:27] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":21313,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":199,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,147,0,5,5,8,5,11,5]}}
[2024-05-15 09:11:27] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,65,83,1,1,4,11,199,0,30,9,16,147,0,5,5,8,5,11,5,123]
[... 57 lines cut ...]
[2024-05-15 09:14:46] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655277/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-15 09:14:46] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x84ba20fffe655277:21313/1 (0,0,2)
[2024-05-15 09:14:46] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":21313,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":211,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[16,159,0,5,5,8,5,11,5]}}
[2024-05-15 09:14:46] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,65,83,1,1,4,11,211,0,30,9,16,159,0,5,5,8,5,11,5,99]
[2024-05-15 09:14:46] debug: 	zh:controller:endpoint: ZCL command 0x04cd15fffe02c8ed/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
[2024-05-15 09:14:52] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655277/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
[2024-05-15 09:14:54] debug: 	zh:controller:endpoint: ZCL command 0x84ba20fffe655150/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-15 09:14:54] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x84ba20fffe655150:56368/1 (0,0,1)
[... cut ...]
[2024-05-15 09:15:52] debug: 	zh:zstack:unpi:writer: --> frame [254,19,36,1,65,83,1,1,4,11,215,0,30,9,16,163,0,5,5,8,5,11,5,91]
[2024-05-15 09:15:52] debug: 	zh:controller:endpoint: ZCL command 0x04cd15fffe02c8ed/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)

MiAutomations, 2024-05-15:
And now When I try to add a new device or pairing a new device I have the following error message 

[image]

MiAutomations, 2024-05-15:
I'm now follow this thread 

[https://github.com/Koenkk/Z-Stack-firmware/discussions/478](url)

I will update the status ...

MiAutomations, 2024-05-15:
Solved

closed 2024-05-15 by MiAutomations

Tonio16, 2024-05-15:
What did you do?

MiAutomations, 2024-05-16:
> What did you do?

I changed the fw of the coordinator that I'm using, I follow this information

[](https://github.com/Koenkk/Z-Stack-firmware/discussions/478)