Koenkk/zigbee2mqtt #22650: Error while pairing Develco / Frient SMSZB-120

okastl, 2024-05-15:
### What happened?

When pairing a Develco / Frient SMSZB-120, the following error is thrown:
`
Error while parsing to ZpiObject 'RangeError: The value of "offset" is out of range. It must be >= 0 and <= 5. Received 6 at boundsError (node:internal/buffer:88:9) at Buffer.readUInt8 (node:internal/buffer:254:5) at BuffaloZnp.readUInt8 (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\buffalo\buffalo.ts:34:35) at BuffaloZnp.read (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\znp\buffaloZnp.ts:217:25) at Function.readParameters (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\znp\zpiObject.ts:107:46) at Function.fromUnpiFrame (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\znp\zpiObject.ts:73:30) at Znp.onUnpiParsed (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\znp\znp.ts:93:38) at Parser.emit (node:events:514:28) at Parser.parseNext (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\unpi\parser.ts:45:26) at Parser._transform (D:\NB\z2m\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\unpi\parser.ts:19:14)'
`

### What did you expect to happen?

No error

### How to reproduce it (minimal and precise)

Try to pair a Develco / Frient SMSZB-120

### Zigbee2MQTT version

1.37.1-dev commit: 2be947b2

### Adapter firmware version

20240315

### Adapter

UZG-01

### Setup

plain on Windows & Linux

sjorge, 2024-05-29:
Can confirm I am seeing this with both MOSZB-140 and WISZB-120, but the device does end up joining.

```
[2024-05-29 11:38:46] error:    zh:zstack:znp: Error while parsing to ZpiObject 'RangeError: The value of "offset" is out of range. It must be >= 0 and <= 5. Received 6
    at new NodeError (node:internal/errors:405:5)
    at boundsError (node:internal/buffer:88:9)
    at Buffer.readUInt8 (node:internal/buffer:254:5)
    at BuffaloZnp.readUInt8 (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/buffalo/buffalo.ts:34:35)
    at BuffaloZnp.read (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/buffaloZnp.ts:217:25)
    at Function.readParameters (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:107:46)
    at Function.fromUnpiFrame (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/zpiObject.ts:73:30)
    at Znp.onUnpiParsed (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:93:38)
    at Parser.emit (node:events:517:28)
    at Parser.parseNext (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/unpi/parser.ts:45:26)'
```

Matriciel, 2024-08-17:
I ave same error with MOSZB-141 Develco

RobLeighton67, 2024-10-25:
Also occurs for me on 1.40.0 with WISZB-120 and SMSZB-120, no obvious after effects...

mamrai1, 2024-10-28:
Problem is fixed in latest development z2m

closed 2025-01-05 by github-actions[bot]