Koenkk/zigbee2mqtt #22651: IKEA E2001/E2002 Interview failed

flyingwild, 2024-05-15:
### What happened?

Have a new Styrbar remote control to add to my Z2m network. Attempting to add via the Z2m dashboard in Home Assistant I get the error: Interview failed because can not get active endpoints.

Have attempted to restart Home Assistant as well as Z2m. Everything else is up to date.

### What did you expect to happen?

The switch would be recognised by Z2m

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle E

### Setup

Add-on on HAOS on a NUC

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15321945/log.log)

flyingwild, 2024-05-15:
Debug log
[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15323434/log.log)

getharder, 2024-05-19:
got the same issue here. new styrbar E2001/E2002 (yesterday bought), sonoff dongle. tried 3 times pairing with removing and repair with z2m 1.37.0, finally successfull pairing, but no response in home assistant when pressing a button.
upgraded z2m to 1.37.1 and ha to 2024.5.4, waited some hours, now it works.

firmware from stybar is 2.4.5, date 20221028, no ota updates available. going to try the documented "hack" for updating here https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1692857032 the next days.

jbleyaert, 2024-05-20:
I have the same issue

Jazmodo, 2024-05-20:
Just to add, having the same issue (I think),

Have been using an Ikea E2001 for a few months with 0 issues. Have updated to the latest firmware, but ever since, the E2001 device has stopped working at all, and disconnects regularly. Unsure if broken, an issue with Z2M, or with Home Assistant?

I've read a few posts elsewhere that suggest force removing, restarting Z2M, re-adding - didn't work
I've also read on the Z2M page for the E2001 that 'Note that the remote loses the capability to bind to a group in the latest firmware.' - so is binding the issue? Can this be reversed/fixed?

I've tried a couple of different HA Blueprints too, neither currently work. 

Unsure if it's me/the latest firmware from Ikea/Z2M itself, so apologies if issue is off-topic

steppel, 2024-05-21:
When you remove and re-add the batteries (soft reset) it works in my case for while until it stops again.

ireun, 2024-05-22:
Same here. 

Coordinator SLZB-06 / FW v2.2.2.dev / ZIGBEE FW 20240315
STYRBAR @ FW 20240408 / 2.4.11

No battery reporting, no button press reporting..

devWaves, 2024-05-24:
I have 5 buttons. only one worked after the update. 4 did not. I got them all working this way https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2127507461

Punkado, 2024-05-25:
> When you remove and re-add the batteries (soft reset) it works in my case for while until it stops again.

Exactly same problem, woks when i remove and put the battery again, but just for some hours.

I already tried the remove and bind again as explained on the other posts and all of this problem started after the update to the 2.4.11, would be nice to be able to downgrade

Jazmodo, 2024-05-25:
> I have 5 buttons. only one worked after the update. 4 did not. I got them all working this way [#18515 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2127507461)

This may have been the solution for me,

I ran through your steps yesterday, no change it had seemed. A day later though, and I've just tested, all 4 buttons working again as expected. So working again for me - E2001 4-button remote. Cheers!

Punkado, 2024-06-01:
> I have 5 buttons. only one worked after the update. 4 did not. I got them all working this way [#18515 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-2127507461)

This helped a bit with the problem, now i can use for a day until stops to work... I really dont understand this problem

flyingwild, 2024-06-27:
Am finding that the remotes are losing their connection in some way. Removal/reinstatement of the battery seems to fix the issue for a period of time, but then they stop working correctly.

moedtm, 2024-07-01:
Same here, every x days the remote stops working, With a quick re-pair or change of batteries (that were not even empty) all is fine again, until the connection is lost a few days later

bobmarley2021, 2024-09-02:
I know this is for a different product, but I have just had this exact issue with two of my [Phillips Hue 9290012607](https://www.zigbee2mqtt.io/devices/9290012607.html#philips-9290012607)

Both fell off the network and could not be repaired despite repeated attempts until removing and reinserting the batteries. As soon as that was done, pairing and interviewing completed within a few seconds. The error was the same as in this issue.

Is this a current Z2M issue?

nck974, 2024-09-30:
Same issue, updated the switch, took around 2h to complete. Then it has been left usable. I tried removing it with force and now it will not finish the interview process.

ellnic, 2024-11-15:
Have also just had this with one of my [Phillips motion sensors](https://www.zigbee2mqtt.io/devices/9290012607.html#philips-9290012607), same as bobmarley.

mleiva5, 2025-01-07:
> I have the same issue
> 
> Edit: Removed batteries while it was listed in z2m (but in error state due to failed interview). After inserting the batteries again the interview is successful.

This worked for my sonoff contact sensors with zigbee2mqtt and HA

closed 2025-03-16 by github-actions[bot]

ireun, 2025-05-08:
I've bought new Styrbar form IKEA, this one is revision E2313. 
* FW 2.4.17, no OTA available
* I can bind into a group
* The group seems to not respond

ViktorSchlaffer, 2026-09-17:
I have also ran into this issue, and this is what solved it for me:
1. When I was doing the pairing, I only allowed join via the Coordinator. You can do this by clicking on the down button next to "Permit join" and pick "Coordinator" from the list of Zigbee devices.
2. After this the initial interview was successful, but the actions were not registered. At this point, I clicked "reconfigure", and right after pressing the "Yes" button in the Web UI, I was also pressing the pairing button on the STYRBAR remote once (to wake it), and then the reconfiguration was successful, you will see log entries similar to this:
z2m: Successfully configured '0x....' (definition v0.0.0)
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x...."},"status":"ok","transaction":"2u06b-37"}'