Koenkk/zigbee2mqtt #22652: easyiot ZB-IR01 is working?

lamerjack, 2024-05-15:
### What happened?

I'm trying to use a easyiot zb-ir01
I didn't understand how to send commands
at the end of this page https://www.zigbee2mqtt.io/devices/ZB-IR01.html there is written:
Send command (text)
so i have to send the hex commands as ascii? 0x02,0x00 etc?
Should be comma separated? or should be binary data?

### What did you expect to happen?

That if i formatted wrongly the command i get an error message.
Somebody already use this zigbee item? Can send me a basic example?
I didn't find nothing about on internet...

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

 6.10.0.0 build 169 or  20230507

### Adapter

EZSP v8 or sonoff

### Setup

i run plain on orange pi with armbian

pykiwi, 2024-07-17:
I have successfully used this device. You need to follow the instructions in section 4.2 to match the air conditioner with the IR
 - https://www.zigbee2mqtt.io/devices/ZB-IR01.html. Then send the commands as shown in the climate.py file at the following path: https://github.com/shing6326/hacs-zb-ir01-to-climate/tree/master/custom_components/zb-ir01-to-climate.

johnfromul, 2024-07-25:
hi, i have same problem....have you more steps walkthrough? ....first step in 4.2 is link to section 3.2 (one click matching)..."search for the ID corresponding to the local code library"...super, but where to find this ID?

pykiwi, 2024-07-26:
> hi, i have same problem....have you more steps walkthrough? ....first step in 4.2 is link to section 3.2 (one click matching)..."search for the ID corresponding to the local code library"...super, but where to find this ID?

Try following these steps:
1. Send command 810100000080 to IR to start one-click matching.
2. When the green light on the IR blinks -> Press the 'Power' button on the remote, pointing it towards the IR.
3. IR sends code 0689xxxx00yy back to z2m -> Save xxxx.
4. Use command 8001xxxx00yy to send to IR (this step is to initialize the new remote): xxxx is previously saved, use BCC tools to generate yy.
5. Use the remote to press power, mode, fan, etc. -> observe the information sent from IR to z2m -> 3.10 Report air conditioning status parameter feedback.

johnfromul, 2024-07-26:
it is much better walkthrough....
1) ok
2) ok
3) i got 068900ee0061
4) in BCC tool i fill and get 6F
[image]

result is "final ir code" is:
 800100ee006f

 is it correct?

pykiwi, 2024-07-29:
> it is much better walkthrough....
> 
> 1. ok
> 2. ok
> 3. i got 068900ee0061
> 4. in BCC tool i fill and get 6F
>    ![image](https://private-user-images.githubusercontent.com/47638587/352549347-41e1ad37-27f3-4880-b236-08b42a779f8f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjIyNDE1MDEsIm5iZiI6MTcyMjI0MTIwMSwicGF0aCI6Ii80NzYzODU4Ny8zNTI1NDkzNDctNDFlMWFkMzctMjdmMy00ODgwLWIyMzYtMDhiNDJhNzc5ZjhmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzI5VDA4MjAwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZhNWQ4ZGRiNGU3YjllMzJmMTkzNTMxZTBlYjUwNzRiMzQ2NjY2ZDRhOGU3OTcwMmNlNzUxYTFiNDY5ZDE2NmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.W5Kn2Aoc8Vti5BpBZCTWeSWRrykwDhI-rwYHngz29Zk)
> 
> result is "final ir code" for On/Off of my Toshiba air conditioner is, but this is too short for IR code: 800100ee006f get answer (last received command): 06890000008f
> 
> ...but send command 86010000006F not working (green led no blinking) my Toshiba air conditioner not response this code, what am I doing wrong ?

You have successfully matched, but why did you send code 86010000006F? Try pressing the power on/off, adjusting the temperature, fan mode, etc. on the remote towards the IR to see if there is any response on z2m?
The IR control code via z2m is here:
[image]

zedlauzy, 2024-10-19:
> Try following these steps:
>
> 1. Send command 810100000080 to IR to start one-click matching.
> 2. When the green light on the IR blinks -> Press the 'Power' button on the remote, pointing it towards [...]

Thank you for sharing these steps. I have stuck on step 5 where I am unable to find the code when I press multiple buttons (with IR yellow LED blink) after triggering command from step 4. I look at MQTT logs & sensor.ir_last_received_command with no observation found. Could you share how you find the codes from step 5. Thanks!

pykiwi, 2024-10-29:
> Thank you for sharing these steps. I have stuck on step 5 where I am unable to find the code when I press multiple buttons (with IR yellow LED blink) after triggering command from step 4. I look at [...]

Once you've configured up to step 4, simply use the remote to press buttons like power on/off, cool mode, auto mode, etc. When you see the IR responding in z2m (as shown in the video), you have successfully added the new remote. After that, you just need to parse the codes that IR sent back to z2m to retrieve information such as temperature, mode, fan, power, etc.
https://vioo.cc/v/VzPmV

xsp1989, 2024-11-06:
> Try following these steps:
>
> 1. Send command 810100000080 to IR to start one-click matching.
> 2. When the green light on the IR blinks -> Press the 'Power' button on the remote, pointing it towards [...]

I am the developer of ZB-IR01. I just saw this question. Your steps are correct, but for step 5, I need to clarify:
[3.10 Report air conditioning status parameter feedback.]
This is when you press the button of the original remote control of the air conditioner, ZB-IR01 can receive the infrared signal and then update in reverse, although the plug-in may not realize this function

closed 2025-01-14 by github-actions[bot]

hesham1216, 2025-12-09:
> I am the developer of ZB-IR01. I just saw this question. Your steps are correct, but for step 5, I need to clarify: [3.10 Report air conditioning status parameter feedback.] This is when you press [...]

Hello

I wanted to contact you to discuss about the ZB-IR01. It is from a company EasyIOT right? The internal chip / MCU is not known what it is. Can you help?

xsp1989, 2025-12-10:
> Hello
>
> I wanted to contact you to discuss about the ZB-IR01. It is from a company EasyIOT right? The internal chip / MCU is not known what it is. Can you help?

It uses the ZYZBP008 module manufactured by EASYIOT, with the same design as SM-011, and the Zigbee chip is EFR32MG21A020F768IM32.

hesham1216, 2025-12-10:
> > > > > > > hi, i have same problem....have you more steps walkthrough? ....first step in 4.2 is link to section 3.2 (one click matching)..."search for the ID corresponding to the local code library"...super, but where to find this ID?
> > > > > > 
> > > > > > 
> > > > > > Try following these steps:
> > > > > > 
> > > > > > 1. Send command 810100000080 to IR to start one-click matching.
> > > > > > 2. When the green light on the IR blinks -> Press the 'Power' button on the remote, pointing it towards the IR.
> > > > > > 3. IR sends code 0689xxxx00yy back to z2m -> Save xxxx.
> > > > > > 4. Use command 8001xxxx00yy to send to IR (this step is to initialize the new remote): xxxx is previously saved, use BCC tools to generate yy.
> > > > > > 5. Use the remote to press power, mode, fan, etc. -> observe the information sent from IR to z2m -> 3.10 Report air conditioning status parameter feedback.
> > > > > 
> > > > > 
> > > > > I am the developer of ZB-IR01. I just saw this question. Your steps are correct, but for step 5, I need to clarify: [3.10 Report air conditioning status parameter feedback.] This is when you press the button of the original remote control of the air conditioner, ZB-IR01 can receive the infrared signal and then update in reverse, although the plug-in may not realize this function
> > > > 
> > > > 
> > > > Hello
> > > > I wanted to contact you to discuss about the ZB-IR01. It is from a company EasyIOT right? The internal chip / MCU is not known what it is. Can you help?
> > > 
> > > 
> > > It uses the ZYZBP008 module manufactured by EASYIOT, with the same design as SM-011, and the Zigbee chip is EFR32MG21A020F768IM32.
> > 
> > 
> > Thank you for your response.
> > Zigbee module has the label, but the chip that has the infrared codes library is not know
> > [image]
> > I don't know if the library can be extracted somehow or if even this chip is sold by its own because I want to design another device with infrared module build in
> 
> You can contact [sales@easyiot.tech](mailto:sales@easyiot.tech)

Thanks bro