Koenkk/zigbee2mqtt #22653: [Feature request]: Set light color before turning ON

Hattan-BA, 2024-05-15:
### Is your feature request related to a problem? Please describe

I’m using some IKEA RGB light bulbs, when turning the bulb on with a certain color, it turns on with the previous color for a second, then changes to the desired color. For example, if it was set to **White** then I turned it off, then I called the service to turn it on as **Red**, it turns On as **White** then changes to **Red**.

### Describe the solution you'd like

There is a setting in Z2M in the exposes of the device that says “Execute if off:
Controls whether the color and color temperature can be set while light is off” [See screenshot bellow]

It works, but only when controlling the color from the Z2M dashboard. Meaning, that if I go to the bulb page in the Z2M dashboard, set it to white then turn it Off, I can choose the color Red (while it's Off), and when I turn it On, it turns On as Red directly.

So the problem isn't hardware related with the bulb or Z2M communicating with the bulb, I think the problem could be with the HA "light.turn_on" service giving the call to Z2M via MQTT in the wrong order, i.e. "turn on" then "set color". When it should be "set color" then "turn on".

### Describe alternatives you've considered

ZHA has an option called "Enable enhanced light color/temperature transition from an off-state", which fixes this issue if enabled.

https://github.com/home-assistant/core/blob/dev/homeassistant/components/zha/light.py#L236-L246

### Additional context

[image]

chris-1243, 2024-05-19:
In your home assistant automation, you may try to send two commands like this (do NOT run them in parallel):
```
- service: mqtt.publish
  data:
    qos: '0'
    retain: false
    topic: 'TOPIC/FRIENDLY_NAME/set'
    payload: '{"color_temp": VALUE}'
- service: light.turn_on 
  data:
   entity_id: "the light to turn on"
or
- service: mqtt.publish
  data:
    qos: '0'
    retain: false
    topic: 'TOPIC/FRIENDLY_NAME/set'
    payload: '{"color_xy": {"x": X_VALUE, "y": Y_VALUE}}'
- service: light.turn_on 
  data:
   entity_id: "the light to turn on"
``` 

It is working perfectly and when the light is turned on, the color is set as you wish

Hattan-BA, 2024-05-19:
Hey, thanks for the response, I made a script to solve this issue in the meantime. But it's definitely not an IKEA bulb firmware issue, since this issue was fixed in ZHA a few days ago. I'm not talking about their "Enable enhanced light color/temperature transition from an off-state" workaround where the turn the bulb on at 0 brightness and set the color then turn it on to the desiered brightness, this option was a workaround for all bulbs, but bulbs that has the "ExecuteIfOff" functionality like the IKEA bulb, doesn't need this work around since they can be sent the command to set the color while they're off, then turn them on as normal. ZHA didn't check if the bulb has the "ExecuteIfOff" feauture and assumed all bulbs didn't have them and used their workaround, now they edited the code to check if the bulb can be controlled while off and then controll it accordingly.

That's what I'm hoping Z2M do

nljuba, 2024-11-27:
Bump for this

barisahmet, 2024-12-09:
Bump, same issue.

closed 2025-02-16 by github-actions[bot]

ArbiterGR, 2025-04-12:
any news on this ?