Koenkk/zigbee2mqtt #22654: TS011F _TZ3000_okaz9tjs - no energy report (again?)

michapr, 2024-05-15:
### What happened?

Have tried to integrate the TS011F _TZ3000_okaz9tjs device, but no success.
Like in https://github.com/Koenkk/zigbee2mqtt/issues/20028 described - same problem. No reporting of voltage,current and energy.

Tried to use https://gist.github.com/DataGhost/572e79e08c0e7eb22773f54c1648b263 - but this is outdated (not working more) - as sample 
`const extend = require('zigbee-herdsman-converters/lib/extend');`
the lib does not exist more. Tried anything, but got different errors...

### What did you expect to happen?

reporting values voltage, current, energy

### How to reproduce it (minimal and precise)

include device - no values will be reported.
generated definition:

```
const definition = {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_okaz9tjs',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), identify(), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

### Zigbee2MQTT version

1.36.1 commit: ffc2ff1d

### Adapter firmware version

7.4.1.0 build 0

### Adapter

Z-Wave.Me Z-Station dual Zigbee & Z-Wave

### Setup

plain installatoion, Pi3, bookworm

### Debug log

```
....
2024-05-15T20:56:01.045461+02:00 pi90-200 node[95298]: Zigbee2MQTT:debug 2024-05-15 20:56:01: Received Zigbee message from '0x70b3d52b6009e4a0', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID 0
2024-05-15T20:56:01.067919+02:00 pi90-200 node[95298]: Zigbee2MQTT:info  2024-05-15 20:56:01: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6009e4a0', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","voltage":0}'
2024-05-15T20:56:58.377728+02:00 pi90-200 node[95298]: Zigbee2MQTT:debug 2024-05-15 20:56:58: Received Zigbee message from '0x70b3d52b6009e4a0', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
2024-05-15T20:56:58.392862+02:00 pi90-200 node[95298]: Zigbee2MQTT:info  2024-05-15 20:56:58: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6009e4a0', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":152,"power":0,"power_outage_memory":"restore","state":"ON","voltage":0}'
2024-05-15T20:57:01.158911+02:00 pi90-200 node[95298]: Zigbee2MQTT:debug 2024-05-15 20:57:01: Received Zigbee message from '0x70b3d52b6009e4a0', type 'readResponse', cluster 'seMetering', data '{"currentSummDelivered":[0,0]}' from endpoint 1 with groupID 0
2024-05-15T20:57:03.863464+02:00 pi90-200 node[95298]: Zigbee2MQTT:debug 2024-05-15 20:57:03: Received Zigbee message from '0x70b3d52b6009e4a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
2024-05-15T20:57:03.876265+02:00 pi90-200 node[95298]: Zigbee2MQTT:info  2024-05-15 20:57:03: MQTT publish: topic 'zigbee2mqtt/0x70b3d52b6009e4a0', payload '{"child_lock":"UNLOCK","current":0,"energy":0,"indicator_mode":"off/on","linkquality":152,"power":0,"power_outage_memory":"restore","state":"ON","voltage":0}'
2024-05-15T20:57:41.301935+02:00 pi90-200 node[95298]: Zigbee2MQTT:debug 2024-05-15 20:57:41: Received Zigbee message from '0x70b3d52b6009e4a0', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
....
```

michapr, 2024-05-16:
database value - if this could help....
```
{
    "id": 10,
    "type": "Router",
    "ieeeAddr": "0x70b3d52b6009e4a0",
    "nwkAddr": 1748,
    "manufId": 4660,
    "manufName": "_TZ3000_okaz9tjs",
    "powerSource": "Mains (single phase)",
    "modelId": "TS011F",
    "epList": [1],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devI
            d": 256,
            "inClusterList": [0, 6, 3, 4, 5, 4096, 10, 57345, 57344, 1794, 2820],
            "outClusterList": [25],
            "clusters": {
                "genBasic": {
                    "attributes": {
                        "hwVersion": 0,
                        "manufacturerName": "_TZ3000_okaz9tjs",
                        "zclVersion": 3,
                        "appVersion": 112,
                        "modelId": "TS011F",
                        "pow
                        erSource": 1
                    }
                },
                "haElectricalMeasurement": {
                    "attributes": {
                        "acCurrentDivisor": 1000,
                        "acCurrentMultiplier": 1,
                        "rmsVoltage": 0,
                        "rmsCurrent": 0,
                        "activePower": 0
                    }
                },
                "seMetering": {
                    "attributes": {
                        "divisor": 100,
                        "multiplier": 1,
                        "currentSummDelivered": [
                            0, 0]
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "32768": 0,
                        "onOff": 1,
                        "tuyaBacklightMode": 1,
                        "moesStartUpOnOff": 2,
                        "onTime": 0,
                        "offWaitTime": 0
                    }
                }
            },
            "binds": [{
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x943469fffe0d65a9",
                    "endpointID": 1
                }, {
                    "cluster
                    ": 2820,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x943469fffe0d65a9",
                    "endpointID": 1
                }, {
                    "cluster": 1794,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x943469fffe0d65a9",
                    "endpointID": 1
                }
            ],
            "configuredReportings": [{
                    "cluster": 1794,
                    "attrId": 0,
                    "minRep
                    Intval": 5,
                    "maxRepIntval": 3600,
                    "repChange": [1, 1],
                    "manufacturerCode": null
                }, {
                    "cluster": 2820,
                    "attrId": 1288,
                    "minRepIntval": 5,
                    "maxRepIntval": 3600,
                    "repChange": 50,
                    "manufacturerCode": null
                }, {
                    "cluster": 2820,
                    "attrId": 1285,
                    "minRepIntval": 5,
                    "maxR
                    epIntval": 3600,
                    "repChange": 5,
                    "manufacturerCode": null
                }
            ],
            "meta": {}
        }
    },
    "appVersion": 112,
    "stackVersion": 2,
    "hwVersion": 0,
    "swBuildId": "0122052017",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {
        "configured": 1324213189
    },
    "lastSeen": 1715882
    614486
}

```

Koenkk, 2024-05-19:
Can you provide a screenshot of the device -> about tab with z2m 1.37.1?

michapr, 2024-05-19:
Attached the log after join the plug again - maybe you see something here.
(maybe also other devices are inside...)
[join.log](https://github.com/Koenkk/zigbee2mqtt/files/15370464/join.log)

kirush0280, 2024-05-20:
The same issue, 1.37.1 no energy update at all after upgrade on all my devices with energy monitoring. Energy sum doesn't updates.

Koenkk, 2024-05-21:
Could you try downgrading to z2m, see if it works and then provide a screenshot of the about tab again?

kirush0280, 2024-05-21:
in HomeAssistant was Zigbee2MQTT Edge, I changed it to default (without Edge) and its works again.

michapr, 2024-05-22:
Is there such a way (downgrade) for the plain installation also available?

THANKS!

Koenkk, 2024-05-22:
@michapr You are already on an old version (1.36.1), so this did not break due to an update.

I've checked your log and your device is actually reporting values, e.g. here it reports a voltage of `0`:

```
Received Zigbee message from '0x70b3d52b6009e4a0', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsVoltage":0}' from endpoint 1 with groupID 0
```

Obviously a voltage of `0` doesn't make any sense. Therefore I think that your plug is broken somehow, I don't see how we can fix this from the z2m side.

michapr, 2024-05-23:
>  Therefore I think that your plug is broken somehow, I don't see how we can fix this from the z2m side.

ok, then looks like it is broken. I thought it may be that this report line is wrong because value will not be detected right by z2m.
Will check with other gateway later and then close the issue if really broken...
THANKS!

closed 2024-05-25 by Koenkk

MBRetrofit, 2025-04-06:
My _TZ3000_okaz9tjs report 218-214 Volts and sometimes it became unresponsive maybe because the erroneous voltage reading protection something.

Firmware build date
20210625
Firmware version
500+TZSKT31BS102

Z2M shows that the device does not respond to OTA queries.