Koenkk/zigbee2mqtt #22659: z2m crashing randomly

jchristianj, 2024-05-16:
### What happened?

Hi, I have z2m running in an LXC under Proxmox, installed with tteck's install script. After a few days of reliable operation z2m crashes randomly. After reboot of the LXC everything is fine again.
At the moment no debug logs available, only normal logs. Maybe they already show the reason?

### What did you expect to happen?

Not randomly crashing

### How to reproduce it (minimal and precise)

Start z2m, wait for a few days.

### Zigbee2MQTT version

1.37.1 commit: d043fd56

### Adapter firmware version

7.4.2 [GA]

### Adapter

SONOFF ZBDongle-E Zigbee, ember driver enabled

### Setup

LXC on a Proxmox host

### Debug log

[log_crash.log](https://github.com/Koenkk/zigbee2mqtt/files/15332742/log_crash.log)
[log_reboot_LXC.log](https://github.com/Koenkk/zigbee2mqtt/files/15332744/log_reboot_LXC.log)

jchristianj, 2024-05-18:
@Koenkk: It happened again, with debug logging enabled. I went up to FW version 7.4.3, but as you can see it is still crashing. Log file attached, the crash happened around 17:10h.

I created an automation in home assistant to reboot the Proxmox host in case the z2m bridge gets disconnected (operating a cluster, and z2m is the only container on that host). It is not sufficient to reboot the z2m LXC only, it will not recover then. I need to reboot the complete host machine, then z2m recovers and resumes operation.

What is you recommendation? Should I go back to another earlier FW, earlier z2m version or use ezsp instead of ember?

[crash_log.log](https://github.com/Koenkk/zigbee2mqtt/files/15365990/crash_log.log)

jchristianj, 2024-05-18:
Weird. You are right, somehow the port gets lost on the machine. I think I did not experience this with ezsp drivers. As I first measure I went back to ezsp now and observe, but simultaneously will investigate why this occurs. I have adapters on my USB port to extend the cable an position the stick in the middle of the house. It is from  USB --> Ethernet Cable --> USB. They are not active adapters and of not using ethernet standard, but only ethernet cable.

[Log_host.log](https://github.com/Koenkk/zigbee2mqtt/files/15366215/Log_host.log)

Nerivec, 2024-05-18:
The logs suggest EMI (interferences) as the cause, so you may want to go over every connection, simplify as much as possible, replace cables, and identify anything that may have changed around the location recently...

jchristianj, 2024-05-21:
Just wanted to provide some feedback: Didn't manage to isolate the USB port issue, z2m still crashed every few hours no matter which cable or connection I revised. Ended up in setting up a RPi as ser2net server and connecting z2m to that instead of using a direct USB connection. Up to now fingers crossed, no crash and no more issues. Thanks for pushing me in the right direction!

closed 2024-05-21 by jchristianj