Koenkk/zigbee2mqtt #22661: TS0013 _TZ3000_wyhuocal turns off after 2 minutes

D3vil787, 2024-05-16:
### What happened?

If the device is connected to a Home assistant, it turns off any pressed button exactly 2 minutes after turning it on.
If you disconnect the device from the Home Assisatant, it works fine.

### Zigbee2MQTT version

zha

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Pi3b

001Tycoon, 2024-05-16:
Have you seen this? Maybe it will help
[https://github.com/Koenkk/zigbee2mqtt/issues/6138](https://github.com/Koenkk/zigbee2mqtt/issues/6138)

D3vil787, 2024-05-16:
> Have you seen this? Maybe it will help #6138

did not help me, as well as many in the comments to this incident

closed 2024-12-21 by github-actions[bot]