Koenkk/zigbee2mqtt #22663: zigbee2mqtt & Homebridge - Cannot read properties of null

jarojetu, 2024-05-16:
### What happened?

Hi, 

I'm using Homebridge running on docker (v1.8.1 · UI v4.56.2) together with zigbee2mqtt 1.37.1 commit: ea39d86 and getting these errors while trying to start Homebridge. The reason why I'm trying to get some help here is because when I use zigbee2mqtt 1.34.0 Homebridge is starting properly.

But I need to run the latest zigbee2mqtt because of support of other features I need :).

Can someone please help or suggest something?

Homebridge logs:

[5/16/2024, 6:10:31 PM] TypeError: Cannot read properties of null (reading 'vendor')
    at isDeviceDefinition (/homebridge/node_modules/@zegl/homebridge-z2m/src/z2mModels.ts:302:72)
    at Zigbee2mqttAccessory.updateDeviceInformation (/homebridge/node_modules/@zegl/homebridge-z2m/src/platformAccessory.ts:419:30)
    at new Zigbee2mqttAccessory (/homebridge/node_modules/@zegl/homebridge-z2m/src/platformAccessory.ts:97:10)
    at Zigbee2mqttPlatform.addAccessory (/homebridge/node_modules/@zegl/homebridge-z2m/src/platform.ts:562:19)
    at Zigbee2mqttPlatform.configureAccessory (/homebridge/node_modules/@zegl/homebridge-z2m/src/platform.ts:461:10)
    at /homebridge/node_modules/homebridge/src/bridgeService.ts:328:25
    at Array.filter (<anonymous>)
    at BridgeService.restoreCachedPlatformAccessories (/homebridge/node_modules/homebridge/src/bridgeService.ts:288:69)
    at Server.start (/homebridge/node_modules/homebridge/src/server.ts:188:24)
[5/16/2024, 6:10:31 PM] Got SIGTERM, shutting down Homebridge...
[5/16/2024, 6:10:31 PM] AssertionError [ERR_ASSERTION]: Cannot generate setupURI on an accessory that isn't published yet!
    at Bridge.Accessory.setupURI (/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Accessory.ts:994:11)
    at Server.setServerStatus (/homebridge/node_modules/homebridge/src/server.ts:155:45)
    at Server.teardown (/homebridge/node_modules/homebridge/src/server.ts:199:10)
    at signalHandler (/homebridge/node_modules/homebridge/src/cli.ts:93:12)
    at process.emit (node:events:518:28)
    at process.emit (/homebridge/node_modules/homebridge/node_modules/source-map-support/source-map-support.js:516:21)

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB

### Setup

Docker

wastez, 2024-05-17:
This have nothing to do with z2m, normally it should be closed immediately. 

Just to understand, you have both things installed in one container?
Because otherwise z2m should not have a influence to homebridge. 

If both is in one container i would split it to difference containers and different places on the filesystem.

wastez, 2024-05-17:
Ok I see, seems to be a combo project but then please open there an issue. 
It’s to much to see how this was made. 
The error above sounds like the homebridge wants to access something what isn’t available at this state. Can also be that something was changed on z2m what breaks this package. 
But as is said, you have to contact the package maintainer.

jarojetu, 2024-05-17:
Hi, both running on separated docker containers. What is puzzling for me is that it is all good when I use z2m 1.34.0 docker version.

wastez, 2024-05-17:
> Hi, both running on separated docker containers. What is puzzling for me is that it is all good when I use z2m 1.34.0 docker version.

Sorry but that can’t be. 
If containers and workspace would be sepperated there is no possibility that there is a influence. 
I saw in your logs above that they use the same node_module directory. 
And I also read in the path something about that combo project, homebridge-z2m.

jarojetu, 2024-05-17:
Sure this might be a bit confusing but I'm pretty sure about the setup - I was using it quite a while and after the z2m update it stopped to work:

Not working setup:
1. Docker koenkk/zigbee2mqtt:latest
2. Docker homebridge/homebridge:latest (there is plugin called homebridge-z2m https://z2m.dev/ -> which is causing the log confusion I guess)

Working setup:
1. Docker koenkk/zigbee2mqtt:1.34.0
2. Docker homebridge/homebridge:latest

Here is my docker container ls output (homebridge was stopped because it is not working now because of this null issue I'm facing):

CONTAINER ID   IMAGE                                    COMMAND                  CREATED        STATUS         
46230e11f96f   homebridge/homebridge:latest             "/init"                  29 hours ago   Up 50 seconds
3f9b12e1eb73   koenkk/zigbee2mqtt:latest                "docker-entrypoint.s…"   30 hours ago   Up 30 hours
02443d8dd2a8   eclipse-mosquitto:latest                 "/docker-entrypoint.…"   3 months ago   Up 3 weeks

wastez, 2024-05-17:
Did you already try to disable the plugin to test if this is the problem. (Maybe the plugin need to be updated from the maintainer to work with the newest version)
I don’t know how this plugin is working, I would need to inform me to know how this is working. 
But there is no other connection between this two containers. So try to disable it and start it and look if the problem still exists.

wastez, 2024-05-17:
Edit:
As I said, seem to be the plugin, there is already a issue about it:
https://github.com/itavero/homebridge-z2m/issues/794

mrmaximas, 2024-05-18:
it's hb plugin issue, not z2m.

jarojetu, 2024-05-18:
Thanks for your time spent on it - I re-installed the plugin and the issue is gone!

wastez, 2024-05-18:
> Thanks for your time spent on it - I re-installed the plugin and the issue is gone!

Ok cool that I could help you even if it was not related to z2m

closed 2024-12-21 by github-actions[bot]