Koenkk/zigbee2mqtt #22665: Interview Issues with IKEA E2204 Tretakt Smart Plug

stkr22, 2024-05-16:
### What happened?

The IKEA Tretakt Smart Plug E2204 (https://www.zigbee2mqtt.io/devices/E2204.html) always fails the Interview stage. Getting various messages like "No Active endpoints found" or timeout. 

What resolved it for me was to simply reset it again after it joined and the interview failed. Then it would join again and the interview would pass. Thus this could also be a vendor/firmware issues.

### What did you expect to happen?

E2204 would successfully join and pass the interview.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle

### Setup

Kubernetes/containerized on RPI 4B

stkr22, 2024-05-16:
I initially thought that the plug was the issue but it seems something is fundamentally wrong with zigbee2mqtt. Since I updated from 1.36.1 which was stable for weeks, trying to pair a new device will regularly completely crash the RPI 4B it is running on.

Koenkk, 2024-05-17:
Please provide logs

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

stkr22, 2024-05-18:
So I checked the logs and I did not find anything specific except for these kind of messages:

```
2024-05-17T23:55:37.874243706+02:00 stdout F [2024-05-17 23:55:37] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:37.995055813+02:00 stdout F [2024-05-17 23:55:37] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:38.135354811+02:00 stdout F [2024-05-17 23:55:38] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:38.386682045+02:00 stdout F [2024-05-17 23:55:38] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:39.204587237+02:00 stdout F [2024-05-17 23:55:39] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:39.389942491+02:00 stdout F [2024-05-17 23:55:39] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:39.598009242+02:00 stdout F [2024-05-17 23:55:39] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:55:39.857819901+02:00 stdout F [2024-05-17 23:55:39] error: 	z2m: Entity 'homeassistant/sensor' is unknown
2024-05-17T23:56:24.178467045+02:00 stdout F [2024-05-17 23:56:24] error: 	zh:controller:device: Default response to 0xa4c13816c71f1fe5 failed
```

Especially the "Default response" error messages is something I keep on seeing in the more recent logs but apart from that...

stkr22, 2024-05-18:
The weird thing is that the node was fully stable for month with 1.36.1 and previous version but since then it crashes from time to time. I added a cpu limit to the pod zigbee2mqtt is running in to check if high cpu consumption is an issue. Thanks for the quick response @Koenkk

stkr22, 2024-05-23:
Seems it stabilized

closed 2024-05-23 by stkr22