Koenkk/zigbee2mqtt #22669: [Wrong device]: YSRSAI Wireless remote control 4 button 12 scene

ferarias, 2024-05-17:
### Link

https://www.amazon.es/dp/B0BRV51VLX

### Model

YSR-Z-4K

### Description

Zigbee 4 button remote - 12 scene

### Vendor

YSRSAI

### Picture (link)

https://m.media-amazon.com/images/S/aplus-media-library-service-media/d473028c-f672-4796-9fa3-671e82506d03.__CR0,0,1500,1500_PT0_SX300_V1___.jpg

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0xa4c1382436a5cce6","nwkAddr":65298,"manufId":4417,"manufName":"_TZ3000_j61x9rxn","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[1,6,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u000b�-\u0013�\u000b�-\u0012>\f�-\u0012","65506":49,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_j61x9rxn","zclVersion":3,"appVersion":67,"modelId":"TS0044","powerSource":3}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":64,"batteryVoltage":26}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[6,1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-903893911},"lastSeen":1715942658325}

### Notes

It looks like a slightly modified version of https://www.zigbee2mqtt.io/devices/TS0044_1.html.

Probably just need to add `_TZ3000_j61x9rxn` in src\devices\tuya.ts

```
whiteLabel: [
            {vendor: 'Lonsonho', model: 'TS0044'}, {vendor: 'Haozee', model: 'ESW-OZAA-EU'},
            {vendor: 'LoraTap', model: 'SS6400ZB'}, {vendor: 'Moes', model: 'ZT-SY-EU-G-4S-WH-MS'},
            tuya.whitelabel('Moes', 'ZT-SR-EU4', 'Star Ring 4 Gang Scene Switch', ['_TZ3000_a4xycprs']),
            tuya.whitelabel('TuYa', 'TS0044_1', 'Zigbee 4 button remote - 12 scene',
                ['_TZ3000_dziaict4', '_TZ3000_mh9px7cq', '_TZ3000_j61x9rxn']),
            tuya.whitelabel('TuYa', 'TM-YKQ004', 'Zigbee 4 button remote - 12 scene', ['_TZ3000_u3nv1jwk']),
        ],
```

ferarias, 2024-05-17:
Or maybe this one?
https://www.zigbee2mqtt.io/devices/YSR-MINI-Z.html

Great-Chart, 2024-05-17:
Or perhaps this one?
https://www.zigbee2mqtt.io/devices/TS0044_1.html

ferarias, 2024-05-17:
It's more like the TS0044_1, but the brand is YSRSAI, so the model could be https://es.aliexpress.com/i/1005003639510755.html?gatewayAdapt=glo2esp

> Or perhaps this one? https://www.zigbee2mqtt.io/devices/TS0044_1.html

Koenkk, 2024-05-19:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-19 by Koenkk