Koenkk/zigbee2mqtt #22670: high traffic between zigbee2mqtt and MQTT Server (fhem)  original #22201 marked as solved

matze20211, 2024-05-17:
### What happened?

with 1.37.1 traffic high again? 

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Pi

Nerivec, 2024-05-18:
If you are using `debug` log level, with the setting described [here](https://www.zigbee2mqtt.io/guide/usage/debug.html#publishing-to-mqtt-and-to-frontend) set to `true` (default is `false`), then traffic would be higher. Is that the case?

matze20211, 2024-05-19:
was "debug" always in 1.37. but i go to "info" and we'll see what happens, i changed availability from "simple" to "advanced", but this double the traffic? i set it back to "simple" now.

matze20211, 2024-05-22:
no changes, still high traffic, twice as with 1.37

closed 2024-12-21 by github-actions[bot]

matze20211, 2024-12-21:
With v1.42, the traffic is down to 1/3 of the highest level of v1.37.1.