Koenkk/zigbee2mqtt #22673: TS011F _TZ3000_cicwjqth - no energy report and no update

loicguiraud, 2024-05-17:
### What happened?

Have tried to integrate the TS011F _TZ3000_cicwjqth device, but no success.
Like in https://github.com/Koenkk/zigbee2mqtt/issues/20028 described - same problem. No reporting of energy.

Tried to use https://gist.github.com/DataGhost/572e79e08c0e7eb22773f54c1648b263 - but this is outdated (not working more) - as sample
const extend = require('zigbee-herdsman-converters/lib/extend');
the lib does not exist more. Tried anything, but got different errors...

### What did you expect to happen?

energy working

### How to reproduce it (minimal and precise)

reimporting the device

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

ZBDongle-E

### Setup

add on on HA VM

### Debug log

[2024-05-17 17:42:05] error: 	z2m: Failed to configure 'prise pompe piscine', attempt 3 (Error: ZCL command 0xa4c13886beab60bb/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9)
    at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:224:5)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:4254:13)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
    at Configure.configure (/app/lib/extension/configure.ts:121:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
[2024-05-17 17:43:10] warning: 	zhc:ota:common: Images currently unavailable for device 'TS011F', hardwareVersion '1', manufacturerName _TZ3000_cicwjqth, {"fieldControl":0,"manufacturerCode":4417,"imageType":545,"fileVersion":1946234881}'

loicguiraud, 2024-05-18:
here's the database.bd info : 
{"id":16,"type":"Router","ieeeAddr":"0xa4c13886beab60bb","nwkAddr":46837,"manufId":4417,"manufName":"_TZ3000_cicwjqth","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"haElectricalMeasurement":{"attributes":{"activePower":0,"rmsVoltage":239,"rmsCurrent":0}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"moesStartUpOnOff":2}},"genBasic":{"attributes":{"stackVersion":0,"swBuildId":"","manufacturerName":"_TZ3000_cicwjqth","zclVersion":3,"appVersion":116,"modelId":"TS011F","powerSource":1}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec7a63e","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec7a63e","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec7a63e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":116,"stackVersion":0,"hwVersion":1,"swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1716032621956}

thenoid, 2024-05-28:
Getting the same thing as well

```
zigbee2mqtt       | [2024-05-28 16:09:57] error:        z2m: Failed to configure 'Backyard Zigbee Near Swing', attempt 8 (Error: ZCL command 0x187a3efffecaf398/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
zigbee2mqtt       |     at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28)
zigbee2mqtt       |     at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26)
zigbee2mqtt       |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
zigbee2mqtt       |     at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9)
zigbee2mqtt       |     at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:224:5)
zigbee2mqtt       |     at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:4254:13)
zigbee2mqtt       |     at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
zigbee2mqtt       |     at Configure.configure (/app/lib/extension/configure.ts:121:13)
zigbee2mqtt       |     at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21)
zigbee2mqtt       |     at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

Koenkk, 2024-05-30:
The energy reporting should be fixed in the new release on Saturday, let me know if it works.

loicguiraud, 2024-06-05:
still N/A for energy and the same error with version 1.38.0-1 from Saturday

Koenkk, 2024-06-08:
@loicguiraud could you provide the data/database.db entry of your device?

loicguiraud, 2024-06-09:
{"id":16,"type":"Router","ieeeAddr":"0xa4c13886beab60bb","nwkAddr":31835,"manufId":4417,"manufName":"_TZ3000_cicwjqth","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"haElectricalMeasurement":{"attributes":{"activePower":0,"rmsVoltage":237,"rmsCurrent":0}},"genBasic":{"attributes":{"swBuildId":"","zclVersion":3,"manufacturerName":"_TZ3000_cicwjqth","appVersion":116,"modelId":"TS011F","powerSource":1}},"genOnOff":{"attributes":{"32768":0,"moesStartUpOnOff":2}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x84b4dbfffec7a63e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":116,"stackVersion":0,"hwVersion":1,"swBuildId":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1717920123609}

ftreil, 2024-06-09:
Same issue. Tell me if I can help in providing some info.

ftreil, 2024-06-09:
Oupps ... I'm using ZHA I think I'm wrong in this case. Sorry for inconvenience.

Koenkk, 2024-06-10:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/5999c91632ff7c5997a1680593e942db): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

loicguiraud, 2024-06-15:
the script don't work : 

> [2024-06-15 14:02:40] info: 	z2m: Logging to console, file (filename: log.log)
> [2024-06-15 14:02:40] error: 	z2m: Failed to load external converter file 'ext_converter.js' (ota is not defined)
> [2024-06-15 14:02:40] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
> [2024-06-15 14:02:40] error: 	z2m: Note that external converters are not meant for long term usage, it's meant for local 

When I remove the line " ota: ota.zigbeeOTA," of the script, nothing append and the problem is still here

Koenkk, 2024-06-15:
Doesnt it throw a different error when removing ota?

loicguiraud, 2024-06-15:
no log and no error when remove the line

Koenkk, 2024-06-15:
When changing the description, does that description show up in the z2m frontend, if yes, does the configure succeed?

loicguiraud, 2024-06-15:
yes the description change in the frontend when I change it
the configuration log is ok in the logs

Koenkk, 2024-06-15:
does the energy start reporting now?

loicguiraud, 2024-06-15:
still no value and this error : 

> Error 2024-06-15 14:34:58Failed to configure 'prise pompe piscine', attempt 3 (Error: Bind 0xa4c13886beab60bb/1 genOnOff from '0x84b4dbfffec7a63e/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":249}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:96:31) at EmberAdapter.onMessageSentDeliveryFailed (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:610:30) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:3960:18) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:797:18) at Ezsp.tick (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:451:22) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7))

Koenkk, 2024-06-15:
Previously it managed to bind genOnOff, can you try to re-pair the device?

loicguiraud, 2024-06-15:
still this error after re'pair : 

> Failed to configure 'prise pompe piscine', attempt 1 (Error: ZCL command 0xa4c13886beab60bb/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":39242,"apsFrame":{"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":48},"zclSequence":9,"commandIdentifier":7} timed out after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:194:23) at EmberOneWaitress.startWaitingFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:218:50) at Object.func (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:3676:64) at EmberRequestQueue.dispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/requestQueue.ts:117:46))

Koenkk, 2024-06-15:
Does you converter have line 37 (the one responsible for this error) commented out?

loicguiraud, 2024-06-15:
yes the line 37 is commented, and i try to uncomment but no change

Koenkk, 2024-06-15:
And you confirmed the converter loads correctly by changing the description?

loicguiraud, 2024-06-15:
what do you mean by "the converter loads correctly by changing the description" ?

when I change the device description I have a info 2 logs : 

> [2024-06-15 19:59:01] info: 	z2m: Changed config for device prise pompe piscine
> [2024-06-15 19:59:01] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/options', payload '{"data":{"from":{"description":"test","legacy":false},"id":"prise pompe piscine","restart_required":false,"to":{"description":"test 2","legacy":false}},"status":"ok","transaction":"bmpi7-2"}'

Koenkk, 2024-06-16:
Whoops, I meant the description property in the external definition, if you change it, you should see the description change in the z2m frontend, if not, the z2m startup logs will show what went wrong during loading

Koenkk, 2024-06-17:
In the external converter which is `description: 'Smart plug (with power monitoring)',` now

shortN0te, 2024-06-19:
Got the same problem with the TS011F _TZ3000_cicwjqth. I set up the external converter as described above, but it does not seem to have loaded correctly, since the description is unchanged after editing it in the converter file.

loicguiraud, 2024-06-20:
same here, I've 3 plug TS011F_plug_1 but no change to the description when I reload Z2M with editing the script

loicguiraud, 2024-06-20:
the description is changed when I change it in the script for all my TS011F_plug_1 including the one with the problem and still no energy for this plug

Reinhard-Stoeger, 2024-06-26:
Hi,
I join this thread because I have the same TS011F _TZ3000_cicwjqth device.
Beside of having no energy measurement, the following get/set commands do  not work ("UNSUPPORTED_ATTRIBUTE)":

- get/set power_outage_memory
- get/set indicator_mode
- get state
- get/set countdown
- set child_lock

Is there any work in progess on this issue? I still have some days to think about sending back to Amazon.

Tozzi89, 2024-08-03:
I have the same plugs and the same issue. Bought from Amazon in Sweden.

Sygnus83, 2024-08-13:
After the last z2mqtt update , my TS011F_plug_3 dont show Power, only 0w !

vingerha, 2024-09-15:
A few days ago I discovered that I could turn the plug on but not off (from HA) . Only today I found out that the plug showed offline all the time on the Z2M screen,  I could still trigger stuff but got errors too.
From 1.40.1. I went back to 1.39.0 and now it shows fine and can switch on+ off. It reports power but (indeed) not energy...luckily I donot need the energy for this purpose. I understood from other issues that there is a fix on the way related to version of herdsmann?

vingerha, 2024-10-05:
Checked 1.40.2 just now and dispite of another issue stating it to be working, with me Energy remains 'null'

rammstone83, 2024-11-19:
still same behavior :(

shortN0te, 2024-11-19:
I am running Zigbee2MQTT version [1.41.0]commit: [cc31cea] right now and noticed today that the Energy now gets reported.

So it looks like the only standing issue is, that the Unit seems to reporting mA instead of A and probably the same for the kwh but i yet have to verify that.

[image]

rammstone83, 2025-01-13:
same here now. everything works but the power seems to be mA instead of A

jlintula, 2025-02-08:
I also got couple of these _TZ3000_cicwjqth plugs and with z2m 2.1.1 I'm observing that Power (W) is updated correctly, Current (A) is reporting roughly 1800 times too much (not sure exactly, but when power is 80 W and voltage 230 V, current hovers around 600-620 A). And most importantly, Energy (kWh) is off by exactly 100, I think, reporting 50 kWh when 0.5 kWh would appear correct.

saschadecrignis, 2025-02-12:
Hi, same to me.
Power is updated correctly but current is displayed as mA.
What i observed too, is that i can´t do an OTA with this item.

Mgrela, 2025-02-24:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/5999c91632ff7c5997a1680593e942db):
> 
> * save this as file next to `configuration.yaml` as `ext_converter.js`
> * add it to `configuration.yaml`:
> 
> external_converters:
>   - ext_converter.js
> * start z2m, check if issue is fixed

I ran into this issue as well. I use the latest Z2MQTT version 2.1.1-1 and added recently 
Zigbee Model TS011F
Zigbee Manufacturer _TZ3000_gjnozsaz

The current is reported in [A] while it's mA. As a result, the values are incorrect. ZHA shows it correctly.

Is this fix with external converted still required or in the latest Z2MQTT version should be resolved natively?

closed 2025-05-04 by github-actions[bot]

braekling, 2025-05-10:
> I also got couple of these _TZ3000_cicwjqth plugs and with z2m 2.1.1 I'm observing that Power (W) is updated correctly, Current (A) is reporting roughly 1800 times too much (not sure exactly, but [...]

I'd like to confirm this behavior.

Device:
- Model: TS011F_plug_1 
- Manufacturer: _TZ3000_cicwjqth

Environment:
- z2m 2.3.0-1
- HA Core 2025.5.1

What I notice in particular is the much too high value for energy consumption, e.g. a simple freezer with a supposed consumption of 51 kWh for less than 24 hours. But also that displayed functions such as Indicator Mode do not work - which is generally fine if the device does not support it, but ideally it should not be displayed.

Are these things that can possibly still be corrected on the software side, or do I have to realize that the plugs I bought are simply unusable for consumption measurement?

If I can provide any information such as logs, please let me know.

vingerha, 2025-06-17:
> > > I also got couple of these _TZ3000_cicwjqth plugs ..... 

I reactivated mine in the summer (well/lawn-pump) and noticed the incorrect consumption, I had null last year. As a solution I am using the power which seems to be fine as it indicates about 750 and the pump states a performance of 950W. You can try to use current calbiraiton (settings/specific) but you would still need to know how much the device is pulling. Then you can use a HA integration sensor to calculate the kWh. Another option is to put in in series with another plug that you know is trutsworthy and calibrate from there. All in all this cicwjqth seems to s.ck :)