Koenkk/zigbee2mqtt #22676: External Converter for _TZ3210_it1u8ahz ver TS0505B_1_2 ?

djcrawleravp, 2024-05-17:
### What happened?

Is there a way to roll back with an external converter my _TZ3210_it1u8ahz to be able to use the TS0505B_1_2 version?

TS0505B_2_2 isn't working for me and my adaptive lights. I've been searching for this for a long time I'm stuck in version 1.33.1 because of this

### What did you expect to happen?

be acble to use an external converter to keep the version TS0505B_1_2 instead of the TS0505B_2_2

### How to reproduce it (minimal and precise)

updating to any version after 1.33.1

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20240315

### Adapter

SLZB-06

### Setup

Docker container

djcrawleravp, 2024-06-03:
I did a [video explaining my situation](https://streamable.com/akpedd) I don't know if it helps but just found it again

djcrawleravp, 2024-08-04:
Can I help with something? I'm still stucked on 1.33.1

Koenkk, 2024-08-05:
As a workaround, you can stop z2m, change all occurrences of `_TZ3210_it1u8ahz` to `_TZ3210_it1u8ahz_dummy` in the `data/databse.db` and start z2m.

djcrawleravp, 2024-09-05:
> _TZ3210_it1u8ahz

I'm so sorry, just noticed the message, I'm still on ver 1.37.1.

When opening th database.db I find this:

[image]

Should I just "Find and Replace" every "_TZ3210_it1u8ahz" for "_TZ3210_it1u8ahz_dummy" on my text editor?

Will it Work on the latest version?

Thank you so much in advance

djcrawleravp, 2024-09-05:
Don't mind me!

It's working better than before!!!

from 1.37.1 to 1.40.0 in 5 minutes!!! 

Thank you so much!!

closed 2024-12-30 by github-actions[bot]