Koenkk/zigbee2mqtt #22678: After latest update all plug in devices are going offline constantly. Other devices randomly stop responding

Rainz3, 2024-05-18:
### What happened?

Got the 1.37.1 update a few days ago and my whole network is now on the fritz. This is with a SONOFF Zigbee Dongle-E. I've tried all firmware from 7.4.0 - 7.4.3 with the same results. I've tried Ember and EZSP, I get far more errors with Ember so I switched back but it hasn't fixed the issue. Virtually all of my outlets are now showing as offline. They are a mix of Third Reality and Sonoff. A few Aqara presence sensors randomly go out as well as some other items but they generally come back on their own. I've tried the reconfigure button as well as putting the devices back in pairing mode. The last time I tried this it completely removed the device from Z2MQTT and now it will not pair back.

Here is the latest from the log
```
Warning 2024-05-17 20:18:42Failed to ping '0x385cfbfffee8e95d' (attempt 1/2, ZCL command 0x385cfbfffee8e95d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":192}))
Warning 2024-05-17 20:18:50Failed to ping '0x385cfbfffee8e95d' (attempt 2/2, ZCL command 0x385cfbfffee8e95d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":197}))
Error 2024-05-17 20:20:56Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":220} messageTag=7]
Error 2024-05-17 20:20:56Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":221} messageTag=125]
Info 2024-05-17 20:21:50[STACK STATUS] Network closed.
Error 2024-05-17 20:21:51Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":244} messageTag=20]
Error 2024-05-17 20:21:51Delivery of BROADCAST failed for "65532" [apsFrame={"profileId":0,"clusterId":54,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":245} messageTag=8]
Warning 2024-05-17 20:25:52Failed to ping 'grow lights' (attempt 1/1, ZCL command 0x00124b002a4e9967/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":7}))
Warning 2024-05-17 20:26:02Failed to ping 'front_room_presence' (attempt 1/2, ZCL command 0xa4c138a83de9d124/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":8}))
Warning 2024-05-17 20:26:10Failed to ping 'front_room_presence' (attempt 2/2, ZCL command 0xa4c138a83de9d124/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":11}))
Warning 2024-05-17 20:26:20Failed to ping 'front room outlet' (attempt 1/1, ZCL command 0x00124b00292ad6ce/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":12}))
Warning 2024-05-17 20:26:31Failed to ping 'dryer outlet' (attempt 1/1, ZCL command 0x282c02bfffe8ada4/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":14}))
Warning 2024-05-17 20:27:10Failed to ping 'Dee Desktop Outlet' (attempt 1/1, ZCL command 0x282c02bfffe693f2/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Delivery failed for {"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4160,"groupId":0,"sequence":20}))
```
Here is a copy of my config

```
data_path: /config/zigbee2mqtt
socat:
  enabled: true
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  adapter: ember
  port: /dev/ttyACM0
```

Not sure what else to try at this point.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.2 

### Adapter

SONOFF Zigbee Dongle-E

### Setup

Home assistant in a VM on Ubuntu

Nerivec, 2024-05-18:
`ember` will give you far more feedback than `ezsp` in general, but especially when things go wrong, since the full protocol is supported.
Did you try stopping z2m, unplugging your adapter for a minute or so, then back in and starting z2m again?
Which sensors are the "Aqara presence sensors" you mentioned?

Rainz3, 2024-05-23:
Sorry, it's tuya ZY-M100-24G not aqara. I've tried many times unplugging the adapter for several minutes and restarting. My aqara button (WXKG11LM) has stopped working as well. Shows connected but nothing goes through when pressing the button. Aqara door sensor on changes sporatically, aqara motion sensor also just randomly stops working.

FieldofClay, 2024-06-17:
After updating over the weekend I've also come across this. All plug devices are offline. I have plenty of other battery devices that are reporting in OK. I have multiple different plug devices which are either Tuya or Xiaomi, same behaviour regardless of the brand.
Logs look the same.

```
warning 2024-06-17 11:51:55Failed to ping 'Back Desk Power Plug' (attempt 1/1, ZCL command 0xa4c1387df64b6778/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
warning 2024-06-17 11:52:15Failed to ping 'Front Desk Power Plug' (attempt 1/1, ZCL command 0xa4c1389382b851bf/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
warning 2024-06-17 11:53:21Failed to ping 'TV Unit Power Plug' (attempt 1/1, ZCL command 0xa4c1386f930d0563/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))
```

Br3Ak3R, 2024-06-18:
Hello,

ive got this issue also and cant figure out the source. I've tried changing channel on zigbee and also on wifi to whole separate frequency. Zigbee Channel 11 now and Wifi Channel 11. Repairing devices solve problem just for 1-2 days. Can't isolate the issue. 

Here the Log:

`Error 2024-06-18 18:53:24Failed to read state of 'Büro Licht ' after reconnect (ZCL command 0x0c4314fffebb6fbb/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4484 - 1 - 17 - 6 - 1 after 10000ms))
Warning 2024-06-18 18:53:32Failed to ping 'Wohnzimmer Taster' (attempt 1/1, ZCL command 0xa4c138cf417c6146/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 62956 - 1 - 24 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:53:47Failed to ping 'Eingang Tür ' (attempt 1/1, ZCL command 0xa81710b7b3a30000/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 3562 - 1 - 25 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:54:02Failed to ping 'Wintergarten 1' (attempt 1/1, ZCL command 0x842e14fffe610e63/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 35345 - 1 - 26 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:54:17Failed to ping 'Wintergarten 2' (attempt 1/1, ZCL command 0xcc86ecfffe92b138/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 51049 - 1 - 27 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:54:32Failed to ping 'Wendeltreppe Unten' (attempt 1/1, ZCL command 0x84fd27fffed13ed7/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 47508 - 1 - 28 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:54:48Failed to ping 'Esszimmer Fenster' (attempt 1/1, ZCL command 0x84fd27fffee2ad82/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 30110 - 1 - 29 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:55:03Failed to ping 'RGB LED ' (attempt 1/1, ZCL command 0x90fd9ffffe67964f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 50084 - 1 - 30 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:55:18Failed to ping 'Wendeltreppe Oben' (attempt 1/1, ZCL command 0x680ae2fffec4f177/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 18225 - 1 - 32 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:55:33Failed to ping 'Waschraum Licht ' (attempt 1/1, ZCL command 0x0c4314fffeb5175d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 54471 - 1 - 33 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:55:48Failed to ping 'Garage' (attempt 1/1, ZCL command 0xa4c13814b3405cef/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 63786 - 1 - 34 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:56:03Failed to ping 'Wohnzimmer Taster' (attempt 1/1, ZCL command 0xa4c138cf417c6146/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 62956 - 1 - 35 - 0 - 1 after 10000ms))
Warning 2024-06-18 18:56:18Failed to ping 'Eingang Tür ' (attempt 1/1, ZCL command 0xa81710b7b3a30000/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"dir [... cut ...]
> _TZ3000_q6a3tepg: 1
> _TZ3000_hhiodade: 1
> Nach Modell
> TRADFRI motion sensor: 7
> TRADFRI bulb GU10 WS 400lm: 6
> 01MINIZB: 4
> TRADFRI SHORTCUT Button: 3
> VALLHORN Wireless Motion Sensor: 3
> CK-BL702-AL-01(7009_Z102LG03-1): 3
> TS004F: 2
> TY0202: 2
> TRADFRI Signal Repeater: 1
> ZBT-ExtendedColor: 1
> lumi.switch.l1aeu1: 1
> TRADFRI on/off switch: 1
> SML002: 1
> TRADFRI remote control: 1
> SML004: 1
> TS0505B: 1
> lumi.sensor_magnet.aq2: 1
> lumi.remote.b28ac1: 1
> TS0001: 1
> TS0011: 1
> RWL022: 1

samuele2723, 2024-07-01:
following

closed 2024-12-30 by github-actions[bot]