Koenkk/zigbee2mqtt #22682: Having all kinds of errors after moving from ezsp to ember

xhemp, 2024-05-18:
### What happened?

Having all kinds of errors after moving from ezsp to ember

Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0" and : ~x~> [ZCL BROADCAST] Failed to send with status=NETWORK_BUSY.

Tried with both 7.4.1 and 7.4.2 .
I did repair the devices and they look to be working fine but sometimes, the network seems slow (taking time for the device to act).

### What did you expect to happen?

I expected no errors.

### How to reproduce it (minimal and precise)

Just move to ezsp to the ember drive on the configuration.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.1 (also tried with 7.4.2)

### Adapter

Sonoff Dongle-E

### Setup

Z2M on HA as an add-on

### Debug log

[debuglog.txt](https://github.com/Koenkk/zigbee2mqtt/files/15362907/debuglog.txt)

Rainz3, 2024-05-18:
I'm in the same boat. An absolute ton of errors, devices failing to respond even when showing online, devices going offline and not connecting any more, and devices that are connected take up to a minute to respond to commands.

Nerivec, 2024-05-18:
@xhemp Can you check with `ezsp` but with log level set to debug, can you confirm you see lines like this:
```
handleNetworkStatus: nwk=0, errorCode=13
```

xhemp, 2024-05-18:
> @xhemp Can you check with `ezsp` but with log level set to debug (an old-ish log file will work, if you have one), can you confirm you see lines like this:
> 
> ```
> handleNetworkStatus: nwk=0, errorCode=13
> ```

Hi @Nerivec , just tried and got a bunch after 2 minutes.
Anything I could be missing?

Nerivec, 2024-05-18:
Appears to be identical to https://github.com/Koenkk/zigbee2mqtt/issues/22271 `ezsp` is throwing the same errors, they are just only logged in `debug` (not fully implemented).
Did you ever do any kind of migration from one adapter to another?
Can you confirm which firmware you flashed on your adapter? https://github.com/darkxst/silabs-firmware-builder/raw/main/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.1.0-zbdonglee-115200.gbl This one?

xhemp, 2024-05-18:
> Appears to be identical to https://github.com/Koenkk/zigbee2mqtt/issues/22271 `ezsp` is throwing the same errors, they are just only logged in `debug` (not fully implemented).
> Did you ever do any [...]

I was using multi protocol and migrated to ember , had to pair the devices as they were some of them not working .
I flashed 7.4.2 first and then reading on here, I tried 7.4.1 and it is the one flashed now on the device. 
Should I keep it or move to 7.4.2?

Nerivec, 2024-05-18:
There doesn't seem to be any reason behind which version fixed it from what I saw. My guess would be it's more about the flashing itself, something must be going wrong.
Try leaving the adapter unplugged for a minute or so, then plug it back in and start Z2M again.
Try flashing it with a different software (even the same version again):
https://github.com/NabuCasa/universal-silabs-flasher
https://darkxst.github.io/silabs-firmware-builder/

_You can also try 7.4.3 https://github.com/darkxst/silabs-firmware-builder/raw/4.4.3/firmware_builds/zbdonglee/ncp-uart-hw-v7.4.3.0-zbdonglee-115200.gbl but 7.4.1 is the most tested one._

xhemp, 2024-05-18:
> There doesn't seem to be any reason behind which version fixed it from what I saw. My guess would be it's more about the flashing itself, something must be going wrong.
> Try leaving the adapter [...]

Thanks, @Nerivec , I will give it a go with both and update with my findings.

remb0, 2024-05-19:
I tested it again with: https://github.com/darkxst/silabs-firmware-builder/raw/ember-nohw/firmware_builds/skyconnect/ncp-uart-hw-v7.4.1.0-skyconnect-115200.gbl

but still the same:   typical is that it are all powered router devices that are offline now.

xhemp, 2024-05-19:
I've tested 7.4.1 and 7.4.3 and they both have the same issues but using ezap looks to be more stable than ember at the moment.
I will keep testing for the next couple of days and jumping from ezsp to ember to see how it goes.

closed 2024-10-01 by Nerivec