Koenkk/zigbee2mqtt #22686: Paulmann 500.4x led drivers don't support  "x":0.701,"y":0.299

IetIesAai, 2024-05-18:
### What happened?

See also #13840 and #6059 - if you set the light to RGB 255,0,0 (pure red), home assistant requests this as "x":0.701,"y":0.299. The led driver seems to ignore this. Z2m does report this value as set. If you use Z2M qui to set pure red, the value used is x": 0.7006, "y": 0.2993 (which coincidentally other colored zigbee led bulbs seem to report back as set value when "x":0.701,"y":0.299 is requested)
I have tested with a paulmann 500.47, the mentioned ticket seems to have the same issue with a 500.49. 

### What did you expect to happen?

I expected color red to be set

### How to reproduce it (minimal and precise)

Pair a paulmann led driver such as the 500.49, and set eg {"state":"ON","color":{"x":0.701,"y":0.299},"brightness":254}. The state will not change.

### Zigbee2MQTT version

1.37.1 commit: ea39d86

### Adapter firmware version

7.4.2 [GA]

### Adapter

ember (skyconnect)

### Setup

docker container, x86-64 nas

### Debug log

blue - light changes
[2024-05-18 18:41:58] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bureau/ledstrip', payload '{"brightness":254,"color":{"h":240,"hue":240,"s":100,"saturation":100,"x":0.136,"y":0.04},"color_mode":"xy","color_temp":370,"color_temp_startup":370,"last_seen":"2024-05-18T16:41:58.183Z","linkquality":255,"power_on_behavior":"previous","state":"ON"}'

red - no change:
[2024-05-18 18:45:46] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bureau/ledstrip', payload '{"brightness":254,"color":{"h":0,"hue":0,"s":100,"saturation":100,"x":0.701,"y":0.299},"color_mode":"xy","color_temp":159,"color_temp_startup":370,"last_seen":"2024-05-18T16:45:46.257Z","linkquality":255,"power_on_behavior":"previous","state":"ON"}'

Koenkk, 2024-05-19:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-05-19 by Koenkk

Xyaren, 2025-02-13:
@Koenkk 
I encounter the identical issue on some Paulmann LEDW Strip Controllers. These are identified as the same device (https://www.zigbee2mqtt.io/devices/948.47_29165.html#paulmann-948.47%252F29165) as some Paulmann RGBW Spots that do not suffer from the pure-red issue.

Any suggestion how to proceed?
I probably should not just add the red fix right?