Koenkk/zigbee2mqtt #22687: LEDVANCE 4058075729209  - Update error

groggy99, 2024-05-18:
### What happened?

Error updating firmware of LEDVANCE 4058075729209 to version 52442720 OTA

### What did you expect to happen?

Update LEDVANCE 4058075729209 to version 52442720

### How to reproduce it (minimal and precise)

klicking OTA update button in frontend

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

SONOFF ZB Dongle-P

### Setup

QNAP NAS 

### Debug log

[2024-05-18 20:01:50] debug: 	z2m: Received Zigbee message from 'ab/licht/terrasse', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":51656288,"imageType":208,"manufacturerCode":4489}' from endpoint 1 with groupID 0
[2024-05-18 20:01:50] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/ab/licht/terrasse', payload '{"brightness":254,"last_seen":"2024-05-18T20:01:50+02:00","linkquality":18,"state":"OFF","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":null}'
[2024-05-18 20:01:50] debug: 	zhc:ota:common: Got request '{"fieldControl":0,"manufacturerCode":4489,"imageType":208,"fileVersion":51656288}'
[2024-05-18 20:01:50] debug: 	zhc:ota:ledvance: Call getImageMeta for A60 FIL DIM T
[2024-05-18 20:01:50] debug: 	zhc:ota:common: Getting new image for '0xf0d1b8be240817c7' (A60 FIL DIM T), latest meta {"fileVersion":52442720,"fileSize":188416,"url":"https://api.update.ledvance.com/v1/zigbee/firmwares/download?company=4489&product=208&version=3.32.54.96","sha256":"f53af0b255d589c795913081560de211b162b491dcb5ea865546dec15f433a78"}
[2024-05-18 20:01:50] debug: 	zh:controller:endpoint: ZCL command 0xf0d1b8be2408a95e/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-18 20:01:50] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xf0d1b8be2408a95e:13491/1 (0,0,1)
[2024-05-18 20:01:50] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":13491,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":49,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,43,0,0,0]}}
[2024-05-18 20:01:50] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,179,52,1,1,0,0,49,0,30,5,16,43,0,0,0,188]
[2024-05-18 20:01:50] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-05-18 20:01:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-05-18 20:01:50] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-05-18 20:01:50] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-05-18 20:01:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-05-18 20:01:50] debug: 	zhc:ota:common: Update checksum validation succeeded for '0xf0d1b8be240817c7' (A60 FIL DIM T)
[2024-05-18 20:01:50] debug: 	zhc:ota:common: Partially failed to parse the image, continuing anyway...
[2024-05-18 20:01:50] debug: 	z2m: Update of 'ab/licht/terrasse' failed (AssertionError [ERR_ASSERTION]: Size mismatch)
[2024-05-18 20:01:50] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/ab/licht/terrasse', payload '{"brightness":254,"last_seen":"2024-05-18T20:01:50+02:00","linkquality":18,"state":"OFF","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":null}'
[2024-05-18 20:01:50] debug: 	z2m: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"ab/licht/terrasse"},"error":"Update of 'ab/licht/terrasse' failed (Size mismatch)","status":"error","transaction":"5d556-2"}'
[2024-05-18 20:01:50] error: 	z2m: Update of 'ab/licht/terrasse' failed (Size mismatch)
[2024-05-18 20:01:50] debug: 	z2m: AssertionError [ERR_ASSERTION]: Size mismatch
    at parseImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:205:11)
    at getNewImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:613:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:438:19)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:273:41)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

christopherkutzmann, 2024-06-18:
I have exact the same issue! I am very interested in a solution.

DanielB1982, 2024-08-30:
Hi. I have the same Problem. 

LEDVANCE  4058075729247    Globe125
Zigbee2MQTT-Version 1.39.1 commit: 5bdf90df
Home Assistant SkyConnect USB Hub

`info 2024-08-30 08:29:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Lampe-Stehlampe-Wohnzimmer', payload '{"brightness":51,"last_seen":"2024-08-30T08:29:16+02:00","linkquality":160,"state":"OFF","update":{"installed_version":51656288,"latest_version":52442720,"state":"available"},"update_available":null}'
info 2024-08-30 08:29:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Lampe-Stehlampe-Wohnzimmer"},"error":"Update of 'Lampe-Stehlampe-Wohnzimmer' failed (Size mismatch)","status":"error","transaction":"rppf4-2"}'
error 2024-08-30 08:29:23z2m: Update of 'Lampe-Stehlampe-Wohnzimmer' failed (Size mismatch)`[LEDVANCE](https://www.zigbee2mqtt.io/supported-devices/#v=LEDVANCE)	[4058075729247](https://www.zigbee2mqtt.io/devices/4058075729247.html#ledvance-4058075729247)

itchensen, 2024-09-02:
Hi, I also have the same problem with one of my plugs. The other one processed nicely.
Any help appreciated.

LEDVANCE [4058075729322](https://www.zigbee2mqtt.io/devices/4058075729322.html#ledvance-4058075729322) PLUG COMPACT EU T
Zigbee2MQTT-Version [1.40.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Adapter SONOFF ZB Dongle-P, Firmware: 20210128

`info 2024-09-02 16:03:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Außensteckdose 2', payload '{"device":{"applicationVersion":1,"friendlyName":"Außensteckdose 2","hardwareVersion":0,"ieeeAddr":"0xf0d1b8be2408ab5e","manufacturerID":4489,"manufacturerName":"LEDVANCE","model":"4058075729322","networkAddress":35577,"powerSource":"Mains (single phase)","softwareBuildID":"03203672","stackVersion":2,"type":"Router","zclVersion":8},"last_seen":"2024-09-02T16:03:22+02:00","linkquality":69,"state":"OFF","update":{"installed_version":52442738,"latest_version":53163636,"state":"available"},"update_available":true}'
info 2024-09-02 16:03:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Außensteckdose 2' failed (Size mismatch)","meta":{"device":"Außensteckdose 2","status":"update_failed"},"type":"ota_update"}'
info 2024-09-02 16:03:23z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Außensteckdose 2"},"error":"Update of 'Außensteckdose 2' failed (Size mismatch)","status":"error","transaction":"q2imu-4"}'
error 2024-09-02 16:03:23z2m: Update of 'Außensteckdose 2' failed (Size mismatch)`

dohco-gh, 2024-09-03:
Same problem here https://github.com/Koenkk/zigbee2mqtt/discussions/23534

[image]

Koenkk, 2024-09-03:
For those running in Docker, can you try the following:
1. Enter the z2m container: `docker exec -it Z2M_CONTAINER_NAME sh`
1. `apk add nano`
1. `nano node_modules/zigbee-herdsman-converters/lib/ota/common.js`, remove: ```(0, assert_1.default)(position === header.totalImageSize, `Size mismatch`);```
1. Save, and `exit`
1. `docker restart Z2M_CONTAINER_NAME`

See if the issue is fixed, if it now throws: `Image size mismatch`, then also remove `(0, assert_1.default)(!meta.fileSize || image.header.totalImageSize === meta.fileSize, `Image size mismatch`);`

itchensen, 2024-09-04:
Thanks for your fast response. Tried the modification and can confirm, that the update now succeeds.
In my case, only the first modification is needed - the "image size mismatch" did not occur.

Koenkk, 2024-09-04:
@itchensen many thanks for testing, integrated the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-04 by Koenkk

groggy99, 2024-09-04:
Many thanks to @itchensen and @Koenkk to solve this problem.

pille89, 2024-09-05:
I'm sorry to have to ask you something stupid. I have a small comprehension problem. I use z2m as an AddOn in Home Assistant, so I'm unsure why I should make docker changes if I don't use docker or are the AddOns docker containers? 

Thank you

NickTheRipper, 2024-09-05:
Many thanks from me too. I installed the dev version and all OTA updates for my lamps work now. Thank you for the really great and fast support.

dohco-gh, 2024-09-08:
A little late to the party, but I would also say thanks!
Installed main branch (1.40.1-1) and can confirm that it works now.
Thanks