Koenkk/zigbee2mqtt #22691: TRV602 - OLED version - Value '0' is not allowed, expected one of true,false}

Dinth, 2024-05-18:
### What happened?

I have been readding my TRVs and noticed an error in my logs:
```
Info 2024-05-18 20:10:00Succesfully interviewed '0xa4c1389bd44838a2'
Info 2024-05-18 20:10:00Successfully interviewed '0xa4c1389bd44838a2', device has successfully been paired
Info 2024-05-18 20:10:00Device '0xa4c1389bd44838a2' is supported, identified as: TuYa Thermostatic radiator valve. (TRV602)
Info 2024-05-18 20:10:01Configuring '0xa4c1389bd44838a2'
Info 2024-05-18 20:10:01Successfully configured '0xa4c1389bd44838a2'
Error 2024-05-18 20:10:04Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
Info 2024-05-18 20:10:35Interview for '0xa4c138b8655783d8' started
Info 2024-05-18 20:10:35Device '0xa4c138b8655783d8' joined
Info 2024-05-18 20:10:35Starting interview of '0xa4c138b8655783d8'
Info 2024-05-18 20:10:52Succesfully interviewed '0xa4c138b8655783d8'
Info 2024-05-18 20:10:52Successfully interviewed '0xa4c138b8655783d8', device has successfully been paired
Info 2024-05-18 20:10:52Device '0xa4c138b8655783d8' is supported, identified as: TuYa Thermostatic radiator valve. (TRV602)
Info 2024-05-18 20:10:52Configuring '0xa4c138b8655783d8'
Info 2024-05-18 20:10:52Successfully configured '0xa4c138b8655783d8'
Error 2024-05-18 20:10:55Exception while calling fromZigbee converter: Value '0' is not allowed, expected one of true,false}
```

This being said the device **seems** to be working properly

### What did you expect to happen?

The device should be configured without any errors

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

20230507

### Adapter

ZigStar

### Setup

PoE

Koenkk, 2024-05-21:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

copernic-us, 2024-10-28:
@Koenkk link to debug guide gives 404. I have same error with ZTRV801

Koenkk, 2024-10-29:
See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

closed 2025-01-06 by github-actions[bot]