Koenkk/zigbee2mqtt #22694: Switch Avatto TS0003 "_TZ3000_ryaookhe" does not have Backlight Mode and Power On Behavior available under Configuration

mwramos, 2024-05-19:
### Link

https://www.aliexpress.com/item/1005006692756886.html?spm=a2g0o.productlist.main.7.68205db7fyKCBH&algo_pvid=b70bbf99-a4d7-4ff0-bce0-cef0076bc9c4&algo_exp_id=b70bbf99-a4d7-4ff0-bce0-cef0076bc9c4-3&pdp_npi=4%40dis%21USD%2123.30%2111.53%21%21%21167.49%2182.91%21%402101f04d17160914842625853e076e%2112000038027167661%21sea%21BR%212974720577%21&curPageLogUid=jPeCzbbkwLWB&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xcc86ecfffe3cd3b4","nwkAddr":9388,"manufId":4098,"manufName":"_TZ3000_ryaookhe","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��-\u0012","65506":31,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_ryaookhe","stackVersion":0,"dateCode":"","zclVersion":3,"appVersion":69,"powerSource":1}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightSwitch":1,"tuyaBacklightMode":0,"moesStartUpOnOff":2}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAA"}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1716069582387}

### Comments

I tried the external definition file below, but the "backlight" was not exposed:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel

    fingerprint: [
        {modelID: 'TS0003', manufacturerName: '_TZ3000_ryaookhe'},
    ],
    model: 'TS0003',
    vendor: 'TuYa',
    description: '3-Gang switch with backlight',
    extend: tuya.extend.switch({powerOnBehavior2: true, backlightModeOffOn: true, endpoints: ['left', 'center', 'right']}),
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
    whiteLabel: [
        tuya.whitelabel('Homeetec', '37022474', '3 Gang switch with backlight', ['_TZ3000_pv4puuxi']),
    ],

};

module.exports = definition;

```

These other Avatto models are correct:
- TS0003 / _TZ3000_pv4puuxi

> {"id":29,"type":"Router","ieeeAddr":"0xa4c1386951b48983","nwkAddr":49365,"manufId":4417,"manufName":"_TZ3000_pv4puuxi","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u0006�-\u0012�\u0006�-\u0012\u0005\u0000\u0000\u0000\u0012\u0006\u0000\u0000\u0000\u0012\u0006\u0000\u0000\u0000\u0012R\u0005\u0000\u0000\u0012","65506":48,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_pv4puuxi","zclVersion":3,"appVersion":80,"modelId":"TS0003","powerSource":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAABAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":80,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1716082204341}

- TS0003 / _TZ3000_avky2mvc

> {"id":16,"type":"Router","ieeeAddr":"0xa4c1381f962251e8","nwkAddr":31440,"manufId":4417,"manufName":"_TZ3000_avky2mvc","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_avky2mvc","zclVersion":3,"appVersion":82,"modelId":"TS0003","powerSource":1}},"genOnOff":{"attributes":{"onOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe9f2f13","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33] [... cut ...]
### External definition

```shell
const {deviceEndpoints, identify, onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0003'],
    model: 'TS0003',
    vendor: '_TZ3000_ryaookhe',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3}}), identify(), onOff({"powerOnBehavior":false,"endpointNames":["1","2","3"]})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

renatopagan, 2024-10-16:
I'm facing the same issue.
Backlight and PoweronBehavior not available on TS0003 device.

ManufacturerName: _TZ3000_785olaiq

I can change the backlight settings via Console on Z2M:
[image]

Database Info:
{"id":3,"type":"Router","ieeeAddr":"0xa4c138d69b6c6b59","nwkAddr":44110,"manufId":4417,"manufName":"_TZ3000_785olaiq","powerSource":"Mains (single phase)","modelId":"TS0003","epList":[1,2,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAAAgAABAAA"}},"genBasic":{"attributes":{"65503":"�o�.\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0003","manufacturerName":"_TZ3000_785olaiq","powerSource":1,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030c15e43","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030c15e43","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0030c15e43","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1729034414108}

Pairing logs:
info 2024-10-15 21:17:31zh:controller: Succesfully interviewed '0xa4c138d69b6c6b59'
info 2024-10-15 21:17:31z2m: Successfully interviewed '0xa4c138d69b6c6b59', device has successfully been paired
info 2024-10-15 21:17:31z2m: Device '0xa4c138d69b6c6b59' is supported, identified as: Tuya 3 gang switch (TS0003)
info 2024-10-15 21:17:31z2m: Configuring '0xa4c138d69b6c6b59'
info 2024-10-15 21:17:31z2m:mqtt: MQTT publish: topic 'homeassistant/switch/0xa4c138d69b6c6b59/switch_left/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c138d69b6c6b59/left/set","device":{"identifiers":["zigbee2mqtt_0xa4c138d69b6c6b59"],"manufacturer":"Tuya","model":"3 gang switch (TS0003)","name":"0xa4c138d69b6c6b59","via_device":"zigbee2mqtt_bridge_0x00124b0030c15e43"},"name":"Left","object_id":"0xa4c138d69b6c6b59_left","origin":{"name":"Zigbee2MQTT","sw":"1.40.2","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xa4c138d69b6c6b59","unique_id":"0xa4c138d69b6c6b59_switch_left_zigbee2mqtt","value_template":"{{ value_json.state_left }}"}'
info 2024-10-15 21:17:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"3 gang switch","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","label":"State","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"center","features":[{"access":7,"description":"On/off state of the switch","endpoint":"center","label":"State","name":"state","property":"state_center","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","label":"State","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0003","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","label":"State action","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true}],"supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c138d69b6c6b59","ieee_address":"0xa4c138d69b6c6b59","status":"successful","supported":true},"type":"device_interview"}'
info 2024-10-15 21:17:31z2m:mqtt: MQTT publish: topic 'homeassistant/switch/0xa4c138d69b6c6b59/switch_center/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"command_topic":"zigbee2mqtt/0xa4c138d69b6c6b59/center/set","device":{"identifiers":["zigbee2mqtt_0xa4c138d69b6c6b59"] [... cut ...] T","sw":"1.40.2","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d69b6c6b59","unique_id":"0xa4c138d69b6c6b59_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2024-10-15 21:17:32z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d69b6c6b59', payload '{"linkquality":29,"state_center":null,"state_left":"OFF","state_right":null}'

If I pair it via Tuya, I have all the entities exposed to HA.

renatopagan, 2024-10-16:
@Koenkk I was trying to use the converter below that used to work, but the /lib/extend is not working anymore. 
Do you have any example of converter that I can try to adapt that doesn't use /lib/extend?

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require(**'zigbee-herdsman-converters/lib/extend');**
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel

    fingerprint: [
        {modelID: 'TS0003', manufacturerName: '_TZ3000_785olaiq'},
    ],
    model: 'TS0003',
    vendor: '_TZ3000_785olaiq',
    description: '3-Gang switch with backlight',
    extend: tuya.extend.switch({powerOnBehavior2: true, backlightModeOffOn: true, endpoints: ['left', 'center', 'right']}),
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
    },
    whiteLabel: [
        tuya.whitelabel('Homeetec', '37022474', '3 Gang switch with backlight', ['_TZ3000_pv4puuxi']),
    ],

};
module.exports = definition;

closed 2025-01-01 by github-actions[bot]