Koenkk/zigbee2mqtt #22695: Failed to execute LQI for 'Coordinator'

KevinGaudry, 2024-05-19:
### What happened?

I have an issue where the LQI from the coordinator cannot be retreived during map loading. I tried changing to the Z2M edge branch, tried changing my coordinator (From USB Sonoff to the SLZB-06 via LAN and moved these around) Looks like it is not interferences related as the issue appear regardless on the coordinator I use. I tried as well a bunch of firmware versions but there are no changes either.
Starting all from 0 will be painful as most of the devices are routers connected directly to the mains; no plugs or switches in between (Hue outside lighting). I have as well a few sonoff ZBMINIL2 and sonoff sensors. Around 25 devices in total

I found the issue already on here and on google but for some reason it has never be really solved, people ended up resetting all or did not gave a resolution of the issue. at the end I spent 2 days googling around with no results.

Thanks for your support.

### What did you expect to happen?

The map to properly load.

### How to reproduce it (minimal and precise)

Difficult to say, the issue happened randomly and is not going away

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20230507

### Adapter

Sonoff 3.0 2652P

### Setup

Addon HA, With HA being in a VM under Proxmox.

### Debug log

[2024-05-19 09:22:57] error: 	z2m: Failed to execute LQI for 'Coordinator'
[2024-05-19 09:22:57] debug: 	z2m: Error: AREQ - ZDO - mgmtLqiRsp after 10000ms
    at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)
    at request (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:581:47)
    at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:603:28
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at doRequest (/app/lib/extension/networkMap.ts:218:28)
    at NetworkMap.networkScan (/app/lib/extension/networkMap.ts:231:32)
    at NetworkMap.onMQTTMessage (/app/lib/extension/networkMap.ts:63:34)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17)

KevinGaudry, 2024-05-19:
I found out my coordinator ieee is as well under one of the device in coordinator_backup.json... How can I delete it ??

Here the beggining:
`{
  "metadata": {
    "format": "zigpy/open-coordinator-backup",
    "version": 1,
    "source": "zigbee-herdsman@0.47.2",
    "internal": {
      "date": "2024-05-19T07:15:28.582Z",
      "znpVersion": 1
    }
  },
  "stack_specific": {
    "zstack": {
      "tclk_seed": "4d72f13573bae3a7f8990ae69fdcb951"
    }
  },
  "coordinator_ieee": "00124b002c3a3b22",
  "pan_id": "0833",
  "extended_pan_id": "5162bb6ab82cd0d5",
  "nwk_update_id": 0,
  "security_level": 5,
  "channel": 11,
  "channel_mask": [
    11
  ],`

And further down under devices:
`      }
    },
    {
      "nwk_address": "9f0f",
      "ieee_address": "00124b002c3a3b22",
      "is_child": false,
      "link_key": {
        "key": "bb31dcb3dcf2434d50ca0f5fbaa8b5f8",
        "rx_counter": 0,
        "tx_counter": 7566
      }
    },`

francisp2, 2024-05-20:
I have that error already more then 2 years, it never stopped to load the map.

KevinGaudry, 2024-05-20:
it loads but the links are somehow messed up.

mashiahd, 2024-06-26:
i have his issue as well...
map is messed up and I can't permit join coordinator only, I must choose all.

bricololo44, 2024-07-19:
I have this error aleeady, Addon HA, with HA being in a VM under Proxmox, with d'ongle SLZB-06p7.

mashiahd, 2024-10-26:
> I have this error aleeady, Addon HA, with HA being in a VM under Proxmox, with d'ongle SLZB-06p7.

maybe its a proxmox issue? my hass is also under proxmox

closed 2025-01-03 by github-actions[bot]

namecaps3k, 2025-11-18:
I have the same issue with POE slab-06. Can’t permit join on coordinator only. When doing map I’m getting
z2m: Failed to execute LQI for 'Coordinator'