Koenkk/zigbee2mqtt #22702: Can't start zigbee2mqtt, followed guide, can't find what's the issue

languagemaniac, 2024-05-19:
### What happened?

I received my zigbee usb dongle some days ago, it's a ZB-GW04 (v1.2), updated it to the latest firmware with [this](https://darkxst.github.io/silabs-firmware-builder/) site.  I have it running on a x86_64 machine running debian bookworm. Tried running zigbee2mqtt on docker and I couldn't start it. So just in case it was a docker thing I installed from source, but I'm facing the same issue.

I have verified that the system recognizes it as ttyUSB0, and that I have access to it, I'm part of the tty group and dialout and the other one listed in the guide, I get the success message when trying to verify that I have access to it.

This is what I get when I try to start zigbee2mqtt

`> zigbee2mqtt@1.37.1 start
> node index.js

[2024-05-19 22:08:15] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-19 22:08:15] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a)
[2024-05-19 22:08:15] info:     z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-19 22:08:16] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-05-19 22:08:16] info:     zh:zstack:znp: Serialport opened
[2024-05-19 22:08:16] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-05-19 22:08:18] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-05-19 22:08:36] error:    z2m: Error while starting zigbee-herdsman
[2024-05-19 22:08:36] error:    z2m: Failed to start zigbee
[2024-05-19 22:08:36] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-19 22:08:36] error:    z2m: Exiting...
[2024-05-19 22:08:36] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)`

I have mosquitto running from source too, systemctl status says it's ok, I have added "listener 1883" and "allow_anonymous true" to mosquitto.conf and restarted it.

I don't really know what else to look for.  I'd appreciate any help

### What did you expect to happen?

Should start, I think I followed the guide

### How to reproduce it (minimal and precise)

I just followed the instructions, installed as detailed in zigbee2mqtt and mosquitto websites. Just doesn't work for me. Docker / source. Maybe I missed something, just didn't do any special thing other than what's written in the install instructions.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7. someting, the latest for my device, can't remember what I got and have no idea how to check

### Adapter

ZB-GW04 (v1.2)

### Setup

source install on debian bookworm x86-64 machine

### Debug log

[2024-05-19 22:41:51] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-19 22:41:52] debug:    z2m: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
[2024-05-19 22:41:52] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a)
[2024-05-19 22:41:52] info:     z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-19 22:41:52] debug:    z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqt>
[2024-05-19 22:41:52] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[11,15,148,113,178,24,158,186,220,127,67,32>
[2024-05-19 22:41:52] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyUSB0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-05-19 22:41:52] info:     zh:zstack:znp: Serialport opened
[2024-05-19 22:41:52] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-19 22:41:52] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-19 22:41:53] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-05-19 22:41:53] debug:    zh:zstack:unpi:writer: --> buffer [239]
[2024-05-19 22:41:54] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-19 22:41:54] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-19 22:41:54] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-05-19 22:41:54] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-19 22:41:54] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-19 22:42:00] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-19 22:42:00] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-19 22:42:06] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-19 22:42:06] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-19 22:42:12] error:    z2m: Error while starting zigbee-herdsman
[2024-05-19 22:42:12] error:    z2m: Failed to start zigbee
[2024-05-19 22:42:12] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-19 22:42:12] error:    z2m: Exiting...
[2024-05-19 22:42:12] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:62:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:109:27)
    at start (/opt/zigbee2mqtt/index.js:107:5)

languagemaniac, 2024-05-20:
> >ZB-GW04 (v1.2),
> 
> Add
> `adapter: ezsp`
> 
> to the configuration.yaml

That did it! Although it tried to use port 8080 which is in use. How can I change it?

closed 2024-05-22 by languagemaniac

ozcanakcesme, 2024-12-01:
> > ZB-GW04 (v1.2),
> 
> Add `adapter: ezsp`
> 
> to the configuration.yaml

I'm using Sonoff ZBDongle-E and having same issue, but this solved my problem too, thank's a lot!