Koenkk/zigbee2mqtt #22703: BTICino K4003CWI not reporting correctly, both the hardware and the result from the interview

langas6, 2024-05-20:
### What happened?

OTA not working for K4003CWI, Once updated, showing incorrect firmware details and won't allow normal "on"/"off" functionality with each of the 2 buttons, only "toggle" for both buttons. Also showing the wrong model/manufacturer for the switch. Not sure if this is a z2m reporting/interview error or a firmware error. But this only started when i tried the firmware update. Have a second switch without the OTA update, and it is reporting the same thing. So i'm leaning towards a z2m reporting issue.

### What did you expect to happen?

To update firmware, cos there was a shiny button saying i could do so. Did not expect the model/firmware and manufacturer code to change for the switch.

### How to reproduce it (minimal and precise)

Pressing buttons on the switch to get results shown in the debug logs below.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

EZSP v12

### Setup

Home Assistant OS in a Virtual Machine on Synology NAS

### Debug log

Device is: 0x0004740000801117
[log1.log](https://github.com/Koenkk/zigbee2mqtt/files/15372347/log1.log)

Sequence of pushes on buttons is on/off/on/off, brightness_move_up/Brightness_move_down (2 x) then double tap up and double tap down, which gives a "off" for both double taps (on and off buttons report "off" for double tap")

Nerivec, 2024-06-10:
@FabianMangold Any chance you can test this? Seems firmware versions have some breaking changes or something. [original mention](https://github.com/Koenkk/zigbee2mqtt/issues/22615#issuecomment-2109970892)

FabianMangold, 2024-07-16:
Hi @Nerivec,
Sorry for the delay, I've been out of town for quite a while.
Will take a look ASAP, but this might be tricky since i exclusively have Legrand HW, no BTicino. Let's see what we can do here.

langas6, 2024-07-17:
Thanks so much for the help with this one.

Just a side note, it seemed to be an issue when i tried to do the OTA update, and after that, all the CWI ones are reporting the wrong information. I have also a K4027C and this is showing the correct image and reporting the correct info when the button is pushed.

I've been able to get a work around for the time being (using the "toggle" output from the switch instead of it's intended "on" and "off" outputs. And also not sure if it's just my instance of HA that's stuffing things up. Need to get one of the switches connected to another HA setup to test that out. (hoping to get a chance this weekend)

Thanks again for looking into this :) Much appreciated!!!

FabianMangold, 2024-08-31:
Hi @langas6 ,
Sorry for the delay in getting back to you, unfortunately i have very little time to work on that lately.
Just for the understanding, what does exactly happen when you mention "showing the incorrect firmware".
Is the device showing up as Legrand instead of a BTicino device, or is there anything else ?

Thanks,

langas6, 2024-09-06:
> Hi @langas6 , Sorry for the delay in getting back to you, unfortunately i have very little time to work on that lately. Just for the understanding, what does exactly happen when you mention "showing [...]

Hey Fabian,

Sorry for the late reply to this one, been a lot happening. But thank you so much for getting back to me!

It seems it's only the BTICino K4003CWI that's reporting incorrectly (showing as a legrand 0677773), where i have a K4003C and it's showing correctly and working flawlessly. I've got about 20 of these CWI switches and they're having issues dropping off the network or not getting picked up correctly and i have a strong feeling it's all related to how it's being interviewed and not configured correctly in Z2M, which is causing the issues.

The firmware version out of the switch is showing as -1 rather than an actual number.

I've tried binding certain functions to empty groups as that's what i've seen from a while ago for the 0677773 that it's reporting as, but i feel this is more a bandaid fix rather than fixing the root issue.

It's also only reporting "toggle" as the action rather than "on" and "off" which the CWI is able to do.

Thanks again so much for the help! Hopefully this helps narrow down what the issue might be.

Cheers

Stuart

```
{
    "battery": 80,
    "brightness": 205,
    "linkquality": 140,
    "rate": 100,
    "voltage": 2900,
    "action": "",
    "action_group": 11,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null,
    "action_rate": 100
}
```

closed 2025-01-01 by github-actions[bot]