Koenkk/zigbee2mqtt #22706: Refusing to start with log_level: warn after update

M66B, 2024-05-20:
### What happened?

Does not work:

```
advanced:
  log_level: warn
```

### What did you expect to happen?

Works:
```
advanced:
  log_level: warning
```

### How to reproduce it (minimal and precise)

Updated to 1.37.1-dev commit: 75608c7
Don't know previous version, sorry.

### Zigbee2MQTT version

1.37.1-dev commit: 75608c7

### Adapter firmware version

n/a

### Adapter

n/a

### Setup

docker

LaurentChardin, 2024-05-20:
Hello,

You are using the dev branch and I believe this is an expected behavoir after : https://github.com/Koenkk/zigbee2mqtt/commit/2eec6a4d26a69ed279206c7b988d56a9656ecaa2

Basically `warn` is deprecated and use `warning` instead. I think the documentation will be updated soon.

CC @Nerivec

ytugarev, 2024-05-21:
Yep, and docs should also be upgraded when dev branch makes it to release
`advanced:
  Optional: Logging level, options: debug, info, warn, error (default: info)`

[https://www.zigbee2mqtt.io/guide/configuration/logging.html](url)

Koenkk, 2024-05-22:
@M66B should be fixed now, can you confirm?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

M66B, 2024-05-23:
@Koenkk It's fine now, thanks!