Koenkk/zigbee2mqtt #22707: [New device support]: Müller Licht tint LED-Spotlights white+color 404051

Chrischan-git, 2024-05-20:
### Link

https://www.mueller-licht.de/produktinformationen/artikel/?q=404051

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0x705464fffe1aea4b","nwkAddr":54193,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI Driver 10W","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64599],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI Driver 10W","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":8,"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20181203","swBuildId":"1.0.4"}},"genLevelCtrl":{"attributes":{"currentLevel":127,"onLevel":255}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":16,"stackVersion":103,"hwVersion":1,"dateCode":"20181203","swBuildId":"1.0.4","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1716198900304} {"id":4,"type":"Group","groupID":1,"members":[],"meta":{}} {"id":5,"type":"Router","ieeeAddr":"0x00158d0006d6ed3a","nwkAddr":24439,"manufId":4137,"manufName":"MLI","powerSource":"Mains (single phase)","modelId":"tint-Spotlights","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,4,3,6,8,5,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"tint-Spotlights","manufacturerName":"MLI","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"TDU50E01"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":555,"startUpColorTemperature":65535,"colorMode":1,"currentX":22918,"currentY":23154,"enhancedCurrentHue":10524,"currentSaturation":59,"colorTemperature":202}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":255}},"genLevelCtrl":{"attributes":{"currentLevel":132}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","swBuildId":"TDU50E01","zclVersion":2,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1716200123480}

### Comments

Everything seems to work, please add support for the device.

### External definition

```shell
const {identify, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['tint-Spotlights'],
    model: 'tint-Spotlights',
    vendor: 'MLI',
    description: 'Automatically generated definition',
    extend: [identify(), light({"colorTemp":{"range":[153,555]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};

module.exports = definition;
```

Chrischan-git, 2024-05-20:
added device picture:
https://github.com/Koenkk/zigbee2mqtt.io/pull/2760

Koenkk, 2024-05-21:
Could you fill in a proper `model` (what is written on the device) and `description`?

Chrischan-git, 2024-05-22:
Model: "404051"
Vendor: "Müller Licht"
For the Description I would say: "tint LED-Spotlights, white+color 3x (1800-6500K+RGB) 19W"

Koenkk, 2024-05-22:
Added, thanks!

closed 2024-05-22 by Koenkk