Koenkk/zigbee2mqtt #22713: [New device support]: WiFi Human Presence Sensor (WZ-M100-W)

bchevreau, 2024-05-20:
### Link

https://www.lazada.com.ph/products/i3344108826-s17012981146.html

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0xa4c138ad8623cb59","nwkAddr":13540,"manufId":4417,"manufName":"_TZE204_laokfqwu","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_laokfqwu","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1716221996246}

### Comments

Model : WZ-M100-W
'TS0601' and manufacturer name '_TZE204_laokfqwu'
5.9G Wall Mount version

It seems a new manufacturer was added to the list, but it is not working for me. I added all the converters from the whenzi github link below just in case, but it did not help... Still shows as unsupported.
https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt

Info 2024-05-21 00:19:54Interview for '0xa4c138ad8623cb59' started
Info 2024-05-21 00:19:54Device '0xa4c138ad8623cb59' joined
Info 2024-05-21 00:19:54Starting interview of '0xa4c138ad8623cb59'
Info 2024-05-21 00:19:56Succesfully interviewed '0xa4c138ad8623cb59'
Info 2024-05-21 00:19:56Successfully interviewed '0xa4c138ad8623cb59', device has successfully been paired
Warning 2024-05-21 00:19:56Device '0xa4c138ad8623cb59' with Zigbee model 'TS0601' and manufacturer name '_TZE204_laokfqwu' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### External definition

```shell
Should be the same as https://www.zigbee2mqtt.io/devices/WZ-M100-W.html
```

bchevreau, 2024-05-21:
Manufacturer added a new file for my device today.

However I noticed something, the presence sensor itself doesn't seem to be recognized as a presence sensor so I cannot pick it as such in blueprints and regular triggers:

[image]

There is no domain class attached to it, could I maybe edit the external converter to give it a domain or device_class that's recognized by HA ? I'm not sure how to do that without breaking the converter.

```
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE204_laokfqwu'}],
        model: 'WZ-M100-W',
        vendor: 'Wenzhi',
        description: 'Human presence sensor',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tzDatapoints],
    		onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    		configure: tuya.configureMagicPacket,
        exposes: [
            e.enum('presence_state', ea.STATE, ['none', 'presence']).withDescription('presence_state'),
            e.numeric('sensitivity', ea.STATE_SET).withValueMin(1).withValueMax(9).withValueStep(1).withDescription('sensitivity of the radar'),
       			e.numeric('near_detection', ea.STATE_SET).withValueMin(0).withValueMax(10.0).withDescription('minimum detection range').withValueStep(0.1).withUnit('m'),
        		e.numeric('far_detection', ea.STATE_SET).withValueMin(0).withValueMax(10.0).withDescription('maximum detection range').withValueStep(0.1).withUnit('m'),
            e.numeric('distance', ea.STATE).withValueMin(0).withValueMax(10.0).withDescription('target distance').withValueStep(0.01).withUnit('m'),
            e.numeric('illuminance_value', ea.STATE).withValueMin(0).withValueMax(2000).withDescription('illuminance_lux').withValueStep(1).withUnit('lux'),
            e.numeric('interval_time', ea.STATE_SET).withValueMin(1).withValueMax(3600).withDescription('interval_time').withValueStep(1).withUnit('s'),
            e.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10.0).withValueStep(0.1).withUnit('s').withDescription('detection delay'),
            e.numeric('fading_time', ea.STATE_SET).withValueMax(1500).withValueMin(5).withValueStep(5).withUnit('s').withDescription('presence timeout'),
            
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'presence': tuya.enum(1)})],
                [2, 'sensitivity', tuya.valueConverter.raw],
                [3, 'near_detection', tuya.valueConverter.divideBy100],
                [4, 'far_detection', tuya.valueConverter.divideBy100],
								[9, 'distance', tuya.valueConverter.divideBy100],
								[103, 'illuminance_value', tuya.valueConverter.raw],
								[104, 'interval_time', tuya.valueConverter.raw],
								[105, 'detection_delay', tuya.valueConverter.divideBy10],
								[106, 'fading_time', tuya.valueConverter.divideBy10],
            ],
        },

    },
];
```

Dendrowen, 2024-05-28:
I've got the same issue

rinnegan69, 2024-06-09:
Hello.

Were you able to figure it out? I have the same problem.

Regards.

githubsharezone, 2024-06-26:
i have the same problem . news ?

10der, 2024-07-04:
ping

Koenkk, 2024-07-04:
@bchevreau with the following changes in your converter, does everything work correctly?

- `e.enum('presence_state', ea.STATE, ['none', 'presence']).withDescription('presence_state'),` -> `e.presence()`
- `[1, 'presence_state', tuya.valueConverterBasic.lookup({'none': tuya.enum(0), 'presence': tuya.enum(1)})],` -> `[1, 'presence', tuya.valueConverter.trueFalse1],`

Yoph974, 2024-07-05:
Unless I did something wrong, it did not work for me...

Koenkk, 2024-07-07:
Could you provide the debug log when the presence should change from `false` to `true`?

Yoph974, 2024-07-07:
Is this what you need ? 

Info 2024-07-07 16:09:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8d5738dd0', payload '{"detection_delay":1,"distance":0,"fading_time":6.6,"far_detection":3,"illuminance_value":32,"interval_time":5,"linkquality":168,"near_detection":0.4,"presence_state":"none","sensitivity":3}'
Info 2024-07-07 16:09:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8d5738dd0', payload '{"detection_delay":1,"distance":0,"fading_time":6.6,"far_detection":3,"illuminance_value":32,"interval_time":5,"linkquality":176,"near_detection":0.4,"presence_state":"presence","sensitivity":3}'
Info 2024-07-07 16:09:39z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8d5738dd0', payload '{"detection_delay":1,"distance":2.94,"fading_time":6.6,"far_detection":3,"illuminance_value":32,"interval_time":5,"linkquality":176,"near_detection":0.4,"presence_state":"presence","sensitivity":3}'

The issue Bchevreau said is still here.

Koenkk, 2024-07-09:
I need the debug log

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Yoph974, 2024-07-10:
Hello, Is this any good?

```
[2024-07-10 16:38:23] debug: 	zh:controller:endpoint: ZCL command 0xa4c138d8d5738dd0/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":76,"writeUndiv":false})
[2024-07-10 16:38:23] debug: 	zh:ezsp: sendZclFrameToEndpointInternal 0xa4c138d8d5738dd0:25112/1 (0,0,1), timeout=10000
[2024-07-10 16:38:23] debug: 	z2m: Received Zigbee message from '0xa4c138d8d5738dd0', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":1}],"seq":30208}' from endpoint 1 with groupID 0
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: ==> sendUnicast: {"type":0,"indexOrDestination":25112,"apsFrame":{"profileId":260,"sequence":75,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":76,"message":{"type":"Buffer","data":[16,76,11,2,0]}}
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: ==> {"_cls_":"sendUnicast","_id_":52,"_isRequest_":true,"type":0,"indexOrDestination":25112,"apsFrame":{"profileId":260,"sequence":75,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":76,"message":{"type":"Buffer","data":[16,76,11,2,0]}}
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> DATA (5,5,0): 8500013400001862040100ef0101000100004b4c05104c0b0200
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> [55c721a9602a15aa3b904b25455493499d4e27e0a1cb77c7f6c463e6fd7e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: -?- waiting (6)
[2024-07-10 16:38:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8d5738dd0', payload '{"detection_delay":1,"distance":1.03,"fading_time":6.6,"far_detection":3,"illuminance_value":32,"interval_time":5,"linkquality":196,"near_detection":0.4,"presence_state":"none","sensitivity":3}'
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- [56c7a1a9602a15d45cfa7e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- DATA (5,6,0): 56c7a1a9602a15d45cfa7e
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> ACK  (6)
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> [8610be7e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- ACK (6): 56c7a1a9602a15d45cfa7e
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: <== Frame: 85800134000066
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":102}
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: -+- waiting (6) success
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- [66c7b1a96b2a15aa3b904b25455493499d4e27cda1ce67ac607e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- DATA (6,6,0): 66c7b1a96b2a15aa3b904b25455493499d4e27cda1ce67ac607e
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> ACK  (7)
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> [87009f7e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- ACK (6): 66c7b1a96b2a15aa3b904b25455493499d4e27cda1ce67ac607e
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: <== Frame: 8590013f00001862040100ef010100010000664c0000
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":25112,"apsFrame":{"profileId":260,"sequence":102,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":76,"status":0,"message":{"type":"Buffer","data":[]}}
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- [76c7b1a97d312a15b65894a524ab5593499ce5e366f5ac9874f0cf2e8bfc0936a5ebc9de6f8fffc50f4c7e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- DATA (7,6,0): 76c7b1a9112a15b65894a524ab5593499ce5e366f5ac9874f0cf2e8bfc0936a5ebc9de6f8fffc50f4c7e
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> ACK  (0)
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> [8070787e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: <-- ACK (6): 76c7b1a9112a15b65894a524ab5593499ce5e366f5ac9874f0cf2e8bfc0936a5ebc9de6f8fffc50f4c7e
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: <== Frame: 859001450000040100ef010100010000abc4cd1862ffff0d094d020077090200040000000002
[2024-07-10 16:38:23] debug: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":171,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lastHopLqi":196,"lastHopRssi":-51,"sender":25112,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,77,2,0,119,9,2,0,4,0,0,0,0]}}
[2024-07-10 16:38:23] debug: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":171,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":196,"rssi":-51,"sender":25112,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,77,2,0,119,9,2,0,4,0,0,0,0]}}
[2024-07-10 16:38:23] debug: 	zh:controller: Received payload: clusterID=61184, address=25112, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=196, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":77,"commandIdentifier":2},"payload":{"seq":30464,"dpValues":[{"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dp [... cut ...]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> DATA (6,0,0): 8600013400001862040100ef0101000100004c4d05104d0b0200
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: --> [60c421a9602a15aa3b904b25455493499d4e27e7a0cb77c6f6c463818f7e]
[2024-07-10 16:38:23] debug: 	zh:ezsp:uart: -?- waiting (7)
[2024-07-10 16:38:23] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8d5738dd0', payload '{"detection_delay":1,"distance":0,"fading_time":6.6,"far_detection":3,"illuminance_value":32,"interval_time":5,"linkquality":196,"near_detection":0.4,"presence_state":"none","sensitivity":3}'
```

Koenkk, 2024-07-11:
@Yoph974 can you try with the following [external converter](https://gist.github.com/Koenkk/c4bb9cf177b77d6a2a7d4502c947f16b) and provide the debug logging when presence should change from `false` to `true`?

Yoph974, 2024-07-12:
Hello,
```
[2024-07-12 14:36:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138289650fa9b', payload '{"detection_delay":0.4,"fading_time":6.6,"illuminance_lux":71,"linkquality":196,"maximum_range":2.7,"minimum_range":0.45,"presence":false,"radar_sensitivity":5,"target_distance":0}'
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- [4441b1a96b2a15aa3b904b25455493499d4e2785f9ce67dcce7e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- DATA (4,4,0): 4441b1a96b2a15aa3b904b25455493499d4e2785f9ce67dcce7e
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> ACK  (5)
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> [8520dd7e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- ACK (4): 4441b1a96b2a15aa3b904b25455493499d4e2785f9ce67dcce7e
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: Unexpected packet sequence 4 | 5
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== Frame: 0390013f00001862040100ef0101000100002e140000
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":25112,"apsFrame":{"profileId":260,"sequence":46,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":20,"status":0,"message":{"type":"Buffer","data":[]}}
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- [5546a1a9602a159d5efa7e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- DATA (5,5,0): 5546a1a9602a159d5efa7e
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> ACK  (6)
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> [8610be7e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- ACK (5): 5546a1a9602a159d5efa7e
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== Frame: 0480013400002f
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== 0x34: {"_cls_":"sendUnicast","_id_":52,"_isRequest_":false,"status":0,"sequence":47}
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: -+- waiting (5) success
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- [6546b1a96b2a15d285904b25455493499d4e2784f8ce6707957e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- DATA (6,5,0): 6546b1a96b2a15d285904b25455493499d4e2784f8ce6707957e
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> ACK  (7)
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> [87009f7e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- ACK (5): 6546b1a96b2a15d285904b25455493499d4e2784f8ce6707957e
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== Frame: 0490013f000060dc040100ef0101000100002f150000
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== 0x3f: {"_cls_":"messageSentHandler","_id_":63,"_isRequest_":false,"type":0,"indexOrDestination":56416,"apsFrame":{"profileId":260,"sequence":47,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"messageTag":21,"status":0,"message":{"type":"Buffer","data":[]}}
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- [7546b1a97d312a15b65894a524ab5593499ce0cb7cf5ac9874f0cfef8bfc2a36a5ebc9de6f8f87c585857e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- DATA (7,5,0): 7546b1a9112a15b65894a524ab5593499ce0cb7cf5ac9874f0cfef8bfc2a36a5ebc9de6f8f87c585857e
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> ACK  (0)
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> [8070787e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: <-- ACK (5): 7546b1a9112a15b65894a524ab5593499ce0cb7cf5ac9874f0cfef8bfc2a36a5ebc9de6f8f87c585857e
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== Frame: 049001450000040100ef010100010000aeecd71862ffff0d098c020054090200040000007802
[2024-07-12 14:36:39] debug: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":174,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lastHopLqi":236,"lastHopRssi":-41,"sender":25112,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,140,2,0,84,9,2,0,4,0,0,0,120]}}
[2024-07-12 14:36:39] debug: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":174,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":236,"rssi":-41,"sender":25112,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[9,140,2,0,84,9,2,0,4,0,0,0,120]}}
[2024-07-12 14:36:39] debug: 	zh:controller: Received payload: clusterID=61184, address=25112, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=236, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":140,"commandIdentifier":2},"payload":{"seq":21504,"dpValues":[{"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,120]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2024-07-12 14:36:39] debug: 	zh:controller:endpoint: ZCL command 0xa4c138d8d5738dd0/1 manuSpecificTuya.defaultRsp({"cmdId":2,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":140,"writeUndiv":false})
[2024-07-12 14:36:39] debug: 	zh:ezsp: sendZclFrameToEndpointInternal 0xa4c138d8d5738dd0:25112/1 (0,0,1), timeout=10000
[... cut ...]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> DATA (5,0,0): 0500013400001862040100ef010100010000151605108c0b0200
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: --> [504721a9602a15aa3b904b25455493499d4e27befbcb7707f6c46353b87e]
[2024-07-12 14:36:39] debug: 	zh:ezsp:uart: -?- waiting (6)
[2024-07-12 14:36:39] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138d8d5738dd0', payload '{"detection_delay":1,"distance":1.2,"fading_time":6.6,"far_detection":3,"illuminance_value":375,"interval_time":5,"linkquality":236,"near_detection":0.4,"presence_state":"presence","sensitivity":3}' 
```

Koenkk, 2024-07-13:
Are you sure this is the only message? I see it only updated dp 9 (`{"data":{"data":[0,0,0,120],"type":"Buffer"},"datatype":2,"dp":9}`) which is `distance` and not the presence

Yoph974, 2024-07-17:
The device is shown as not supported again? So I can't provide you a new debug log....

OscarCortesMedina, 2024-07-18:
Here is an external converter that is working for me hope it helps:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_laokfqwu']),
    model: 'WZ-M100',
    vendor: 'Wenzhi',
    description: 'Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            e.illuminance_lux(),
            e.presence(),
            e
                .numeric('target_distance', ea.STATE)
                .withDescription('Distance to target')
                .withUnit('m'),
            e
                .numeric('sensitivity', ea.STATE_SET)
                .withValueMin(1)
                .withValueMax(9)
                .withValueStep(1)
                .withDescription('sensitivity of the radar'),
            e
                .numeric('minimum_range', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(10.0)
                .withDescription('minimum detection range')
                .withValueStep(0.1)
                .withUnit('m'),
            e
                .numeric('maximum_range', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(10.0)
                .withDescription('maximum detection range')
                .withValueStep(0.1)
                .withUnit('m'),
            e
                .numeric('detection_delay', ea.STATE_SET)
                .withValueMin(0)
                .withValueMax(10.0)
                .withValueStep(0.1)
                .withUnit('s')
                .withDescription('detection delay'),
            e
                .numeric('fading_time', ea.STATE_SET)
                .withValueMax(1500)
                .withValueMin(5)
                .withValueStep(5)
                .withUnit('s')
                .withDescription('presence timeout')
        ],
        meta: {
            tuyaDatapoints: [
                [1, 'presence', tuya.valueConverter.trueFalse1],
                [2, 'sensitivity', tuya.valueConverter.raw],
                [3, 'minimum_range', tuya.valueConverter.divideBy100],
                [4, 'maximum_range', tuya.valueConverter.divideBy100],
				[9, 'target_distance', tuya.valueConverter.divideBy100],
				[103, 'illuminance_lux', tuya.valueConverter.raw],
				[105, 'detection_delay', tuya.valueConverter.divideBy10],
				[106, 'fading_time', tuya.valueConverter.divideBy10],
            ],
        },
};

module.exports = definition;

Yoph974, 2024-07-19:
It's perfect. It seems to work perfectly! 
Thank you all.

Koenkk, 2024-07-21:
`_TZE204_laokfqwu` will be supported out-of-the-box in the 1 August release, thanks all!

closed 2024-07-21 by Koenkk