Koenkk/zigbee2mqtt #22715: Ubisys using `configure_device_setup` results in `Cannot convert undefined or null to object`

adriaandotcom, 2024-05-20:
### What happened?

After the fix of [zigbee2mqtt#22610](https://github.com/Koenkk/zigbee2mqtt/issues/22610) (thanks for that, @sjorge), I started configuring my Ubisys S2 and D1 devices with this command:

```json
{
  "configure_device_setup" : {
    "input_action_templates": [
      {
        "type": "dimmer_double"
      }
    ]
  }
}
```

I tried with many configs, and every time it errors with `Cannot convert undefined or null to object` in this file: https://github.com/Koenkk/zigbee-herdsman-converters/blob/d10d6acdc08d44d8c1ee70cdb8a3d64afb28b5b0/src/devices/ubisys.ts#L102-L104C24

Getting this error on dev:

```
TypeError: Cannot convert undefined or null to object
  at Function.values (<anonymous>)
  at /app/node_modules/zigbee-herdsman-converters/src/devices/ubisys.ts:103:39
  at Array.map (<anonymous>)
  at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/ubisys.ts:102:72)
  at Receive.onDeviceMessage (/app/lib/extension/receive.ts:153:51)
  at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23)
  at EventEmitter.emit (node:events:529:35)
  at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:108:22)
  at Controller.<anonymous> (/app/lib/zigbee.ts:117:27)
```

### What did you expect to happen?

A success message with the new config.

### How to reproduce it (minimal and precise)

- Use a D1 / S2 with firmware 2.4.0
- Send command to `zigbee2mqtt/0x00000000000000f7/set`
    ```json
    {
      "configure_device_setup" : {
        "input_action_templates": [
          {
            "type": "dimmer_double"
          }
        ]
      }
    }
    ```

### Zigbee2MQTT version

1.37.1-dev commit: 75608c7 (0.6.168, 19.42.0, 0.48.1)

### Adapter firmware version

0x26580700

### Adapter

ConBee II

### Setup

Add-on on Home Assistant OS on Intel NUC

### Debug log
[... 57 lines cut ...]
debug 2024-05-20 21:47:13 DATA_REQUEST RESPONSE - request id: 198 status: 0
debug 2024-05-20 21:47:13 networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2024-05-20 21:47:13 DEVICE_STATE changed: 10100110
debug 2024-05-20 21:47:13 networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2024-05-20 21:47:13 query aps data confirm
debug 2024-05-20 21:47:13 DATA_CONFIRM - sending data state request - SeqNr. 201
debug 2024-05-20 21:47:13 DATA_CONFIRM RESPONSE - destAddr: 0xc8a4 request id: 198 confirm status: 0
debug 2024-05-20 21:47:13 networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2024-05-20 21:47:13 sendZclFrameToEndpoint - message send with transSeq Nr.: 149
debug 2024-05-20 21:47:13 true, true, false, 10000
debug 2024-05-20 21:47:13 resolve request (149)
debug 2024-05-20 21:47:13 DEVICE_STATE changed: 10101010
debug 2024-05-20 21:47:13 networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
debug 2024-05-20 21:47:13 query aps data indication
debug 2024-05-20 21:47:13 DATA_INDICATION - sending read data request - SeqNr. 202
debug 2024-05-20 21:47:13 DATA_INDICATION RESPONSE - seqNr. 202 srcAddr: 0xc8a4 destAddr: 0x0 profile id: 0x104 cluster id: 0xfc00 lqi: 255
debug 2024-05-20 21:47:13 response payload: 24,149,1,1,0,0,72,0,6,0
debug 2024-05-20 21:47:13 resolve data request with transSeq Nr.: 149
debug 2024-05-20 21:47:13 Received payload: clusterID=64512, address=51364, groupID=null, endpoint=232, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":149,"commandIdentifier":1},"payload":[{"attrId":1,"status":0,"dataType":72,"attrData":[null,null,null,null,null,null]}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
debug 2024-05-20 21:47:13 networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
debug 2024-05-20 21:47:13 response received (149)
debug 2024-05-20 21:47:13 Received Zigbee message from '0x00000000000000f7', type 'readResponse', cluster 'manuSpecificUbisysDeviceSetup', data '{"inputActions":[null,null,null,null,null,null]}' from endpoint 232 with groupID null
error 2024-05-20 21:47:13 Exception while calling fromZigbee converter: Cannot convert undefined or null to object}
debug 2024-05-20 21:47:13 TypeError: Cannot convert undefined or null to object at Function.values (<anonymous>) at /app/node_modules/zigbee-herdsman-converters/src/devices/ubisys.ts:103:39 at Array.map (<anonymous>) at Object.convert (/app/node_modules/zigbee-herdsman-converters/src/devices/ubisys.ts:102:72) at Receive.onDeviceMessage (/app/lib/extension/receive.ts:153:51) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:23) at EventEmitter.emit (node:events:529:35) at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:108:22) at Controller.<anonymous> (/app/lib/zigbee.ts:117:27)
```

sjorge, 2024-05-20:
How are you encoding the payload?

I was testing either this and it works fine for me

```bash
export name=friendly_name
mosquitto_pub -t zigbee2mqtt/${name}/set -m '{"configure_device_setup":{"input_action_templates":[{"type":"dimmer_single", "no_onoff": true}]}}'
```

adriaandotcom, 2024-05-20:
I'm using MQTTX:

[image]

It has worked before with the same app, so I don't think it's an encoding issue. Other commands work fine as well.

When debugging, [`msg` here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/d10d6acdc08d44d8c1ee70cdb8a3d64afb28b5b0/src/devices/ubisys.ts#L102C48-L102C48) has actually nulls:

```js
{
  "type": "readResponse",
  // "device": ...
  // "endpoint": ...
  "data": {
    "inputActions": [
      null,
      null,
      null,
      null,
      null,
      null
    ]
  },
  "linkquality": 255,
  "groupID": null,
  "cluster": "manuSpecificUbisysDeviceSetup",
  // "meta": ...
}
```

sjorge, 2024-05-20:
Does the config actually get applied?

adriaandotcom, 2024-05-20:
I think it does write, but when it reads it fails.

<details markdown="1"><summary>Device state</summary>

```json
{
  "ballast_lamp_alarm_lamp_burn_hours": false,
  "ballast_maximum_level": 254,
  "ballast_minimum_level": 1,
  "ballast_status_lamp_failure": false,
  "ballast_status_non_operational": false,
  "brightness": 25,
  "configure_device_setup": {
    "input_actions": [
      [ 0, 7, 2, 6, 0, 1 ],
      [ 0, 6, 2, 8, 0, 5, 0, 50 ],
      [ 0, 11, 2, 8, 0, 3 ],
      [ 1, 7, 2, 6, 0, 0 ],
      [ 1, 6, 2, 8, 0, 5, 1, 50 ],
      [ 1, 11, 2, 8, 0, 3 ]
    ],
    "input_configurations": [ 0, 0 ]
  },
  "energy": 0,
  "level_config": {
    "current_level_startup": "previous",
    "execute_if_off": true,
    "on_level": "previous",
    "on_off_transition_time": 0
  },
  "linkquality": 255,
  "minimum_on_level": 7,
  "power": 1,
  "state": "ON",
  "update": {
    "installed_version": 37749810,
    "latest_version": 37749810,
    "state": "idle"
  },
  "update_available": false,
  "action": null,
  "capabilities_configurable_curve": null,
  "capabilities_forward_phase_control": null,
  "capabilities_overload_detection": null,
  "capabilities_reactance_discriminator": null,
  "capabilities_reverse_phase_control": null,
  "mode_phase_control": null,
  "status_capacitive_load": null,
  "status_forward_phase_control": null,
  "status_inductive_load": null,
  "status_overload": null,
  "status_reverse_phase_control": null
}
```

</details>

After that, there are no extra binds set:

[image]

So the config does not work on the device. I'm still on dev.

sjorge, 2024-05-20:
You can remove it from the cached state by stopping z2m, removing it from state.json, starting z2m.

Updating the configuration does not add/remove binds. At least they never did for me. That is a separate action.

I'll poke around some more after work tomorrow.

adriaandotcom, 2024-05-20:
Thanks for the tip on the state, removed it from `/root/homeassistant/zigbee2mqtt/state.json` and it's indeed gone.

My goal for the D1 is to have it "dimmer_double" enabled with double push buttons. Have around 30 now in my new house. Before the update of the firmware and z2m, it did just work with `configure_device_setup`. I'm not sure if other things have changed around here. If I can help with some debugging info, please let me know. Thanks for taking a look!

sjorge, 2024-05-21:
> Are those binds required for it to work, or do you only set the `configure_device_setup`?

Yes, it needs the bindings to know where to send the actions. Mine are bound to groups, but if you want the coordinator to get them you need to bind the endpoints to the coordinator.

e.g. one of my S2's
[image]

This one sends directly to the 2 groups and not the coordinator

my C4
[image]

This one sends everything to the coordinator and the first 2 also get send to a group.

Note the endpoints are different between S1, S2, C4 and I think D1 (not sure, don't own the later) (from memory, could be wrong)
S1 -> ep 2
S2 -> ep3 + ep4 
C4 -> ep1, ep2, ep3, ep4 (and ep6 + ep7 for the cover bindings)
J1 -> ep2

Looking at the code, the default bindings are added during interview:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/d10d6acdc08d44d8c1ee70cdb8a3d64afb28b5b0/src/devices/ubisys.ts#L846-L861

You should be able to just restore them using the webui if they are missing.

sjorge, 2024-05-21:
The get step is working for my S1/S2/J1's... maybe it's a D1 specific issue

```
[2024-05-21 16:05:16] debug:    z2m: Publishing get 'get' 'configure_device_setup' to 'switch/office/light'
[2024-05-21 16:05:16] debug:    zh:controller:endpoint: ZCL command 0x1231231231231231/232 manuSpecificUbisysDeviceSetup.read(["inputConfigurations"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-21 16:05:16] debug:    zh:controller: Received payload: clusterID=64512, address=51024, groupID=0, endpoint=232, destinationEndpoint=1, wasBroadcast=false, linkQuality=43, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":143,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":72,"attrData":[0]}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2024-05-21 16:05:16] debug:    z2m: Received Zigbee message from 'switch/office/light', type 'readResponse', cluster 'manuSpecificUbisysDeviceSetup', data '{"inputConfigurations":[0]}' from endpoint 232 with groupID 0
[2024-05-21 16:05:16] debug:    zh:controller:endpoint: ZCL command 0x1231231231231231/232 manuSpecificUbisysDeviceSetup.read(["inputActions"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-05-21 16:05:16] debug:    zh:controller: Received payload: clusterID=64512, address=51024, groupID=0, endpoint=232, destinationEndpoint=1, wasBroadcast=false, linkQuality=36, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":144,"commandIdentifier":1},"payload":[{"attrId":1,"status":0,"dataType":72,"attrData":[{"type":"Buffer","data":[0,7,2,6,0,2]},{"type":"Buffer","data":[0,134,2,8,0,1,0,50]},{"type":"Buffer","data":[0,198,2,8,0,1,1,50]},{"type":"Buffer","data":[0,11,2,8,0,3]}]}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2024-05-21 16:05:16] debug:    z2m: Received Zigbee message from 'switch/office/light', type 'readResponse', cluster 'manuSpecificUbisysDeviceSetup', data '{"inputActions":[{"data":[0,7,2,6,0,2],"type":"Buffer"},{"data":[0,134,2,8,0,1,0,50],"type":"Buffer"},{"data":[0,198,2,8,0,1,1,50],"type":"Buffer"},{"data":[0,11,2,8,0,3],"type":"Buffer"}]}' from endpoint 232 with groupID 0
[2024-05-21 16:05:17] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/switch/office/light', payload '{"configure_device_setup":{"input_actions":[[0,7,2,6,0,2],[0,134,2,8,0,1,0,50],[0,198,2,8,0,1,1,50],[0,11,2,8,0,3]]},"energy":2.45,"last_seen":"2024-05-21T14:05:16.803Z","linkquality":36,"power":1,"power_on_behavior":"on","state":"ON","update":{"installed_version":37749810,"latest_version":37749810,"state":"idle"}}'
```

adriaandotcom, 2024-05-21:
I only have S2 and D1's. With the S2 I had the same issue. Are you doing this on the latest dev?

sjorge, 2024-05-21:
Yes, I updated when I got home.
Also have the latest public firmware on all my Ubisys devices except my H1's which run a dev build.

adriaandotcom, 2024-05-23:
Okay, feel free to close this issue then, as it's maybe only an issue with me of my devices. I will test it later again when everything is in the production master branch. I don't have access to my devices at the moment. Will reopen when issue isn't solved. Thanks for your help!

closed 2024-05-23 by adriaandotcom

adriaandotcom, 2024-09-04:
For future reference, it works fine now when using this config.

For the Ubisys D1, I want:

    short press on 1 = on
    short press on 2 = off
    long press on 1 = brightness up
    long press on 2 = brightness down

`zigbee2mqtt/your_d1_id/set`

```
{
    "configure_device_setup": {
        "input_action_templates": [
            {
                "//_comment": "will automatically use endpoint 1",
                "type": "dimmer_double",
                "inputs": [1, 0]
            }
        ]
    }
}
```

For the Ubisys S2, I want:

    short press on 1 = light 1 on
    short press on 2 = light 1 off
    I don't use light 2 — yet.

`zigbee2mqtt/your_s2_id/set`

```
{
    "configure_device_setup": {
        "input_action_templates": [
            {
                "type": "on"
            },
            {
                "type": "off"
            }
        ]
    }
}
```