Koenkk/zigbee2mqtt #22720: HG08673 has stopped reporting new energy data

Gnarfoz, 2024-05-20:
### What happened?

On May 13, after updating to dev branch version 09e51e0, all of my [HG08673](https://www.zigbee2mqtt.io/devices/HG08673.html) have stopped reporting new energy data. They're stuck on whatever their latest value was.

09e51e0 pulls in [zigbee-herdsman-converters 19.38.0](https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v19.38.0), which is the first version to contain this commit: https://github.com/Koenkk/zigbee-herdsman-converters/commit/4182889448e29e1bcf8e4aae961fe5dea0fa0543
I think it might be related, since these [HG08673](https://www.zigbee2mqtt.io/devices/HG08673.html) are shown as
```
Zigbee Model TS011F
Zigbee Manufacturer _TZ3000_j1v25l17
```
in the frontend.

I'm not sure where the firmware version of the devices is exposed, if at all.

### What did you expect to happen?

Values to keep updating.

### How to reproduce it (minimal and precise)

Just start Z2M and watch the numbers stay the same even though there's energy being consumed.

### Zigbee2MQTT version

1.37.1-dev (commit #09e51e0)

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Docker / Podman

Gnarfoz, 2024-05-21:
Sure: https://gist.github.com/Gnarfoz/712596eb1958d0225bccb98119284e1f

leonidostrovski, 2024-05-21:
Exactly same issue on different devices and networks  (TS011F_plug_1+BSD29_1)
Power: no issue
Current: no issue
Voltage: no issue
Energy: NullkWh or 0kWh

network #1 (1.37.1-dev commit: [d484405])
device:
TS011F_plug_1

network #2 (1.37.1-dev commit: [d484405])
device:
TS011F_plug_1

network #3 (1.37.1-dev commit: [d484405])
device:
TS011F_plug_1
BSD29_1

I can send database.db in private message (not for public use)

Koenkk, 2024-05-22:
@Gnarfoz I don't think it's related to https://github.com/Koenkk/zigbee-herdsman-converters/commit/4182889448e29e1bcf8e4aae961fe5dea0fa0543, this only affects devices detected as `TS011F_plug_3`, yours is detected as `HG08673`. Could you provide a 10 minute debug log? I want to check wether the polling happens.

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable debug logging.

Gnarfoz, 2024-05-22:
Sure thing, here you go: [log.gz](https://github.com/Koenkk/zigbee2mqtt/files/15407677/log.gz)

Gnarfoz, 2024-05-22:
I had singled out https://github.com/Koenkk/zigbee-herdsman-converters/commit/4182889448e29e1bcf8e4aae961fe5dea0fa0543 because it looked vaguely related and fit into the time frame, given that I auto-update daily to the latest dev version.
However, https://github.com/Koenkk/zigbee2mqtt/commit/09e51e064bd4a6759939fbc565f818e6df6d7049 also includes the huge ZigBee spec revamp, so I guess anything is possible. 😀

closed 2024-05-27 by Koenkk

Gnarfoz, 2024-05-28:
Can confirm, it's alive again. :-)
[image]

Big thanks, @Koenkk!