Koenkk/zigbee2mqtt #22721: Aqara FP1 [lumi.motion.ac01] firmware update error

Kevin6872, 2024-05-21:
### What happened?

When checking for an Aqara FP1 [Lumi.motion.ac01] firmware update, the message states: Firmware on '0x54ef441000589610' (lumi.motion.ac01) is newer than the latest firmware online. The current firmware version of my Aqara FP1 is 58, and the firmware build date is 20210121.

### What did you expect to happen?

Correctly check for firmware updates

### How to reproduce it (minimal and precise)

First add Aqara FP1 to Aqara Hub M1S, and then use Aqara Home App to check the firmware update. After upgrading Aqara FP1 to the latest firmware. Then Add it to Zigbee2MQTT, check the firmware update in Zigbee2MQTT, the prompt will appear.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

0x26580700

### Adapter

ConBee II

### Setup

Home Assistant OS on Proxmox

lux73, 2024-05-22:
this is mine FP1 with latest FW from z2m available:

[image]

your Device has newer Firmware - it seems that z2m has only the older 54 Firmware included

Kevin6872, 2024-05-23:
Aqara Home App can update to 58. It seems like Zigbee2MQTT cannot get the right latest firmware for FP1.
[image]

borodaaaaa, 2024-07-21:
Any update on the issue?

stefanuytterhoeven, 2024-10-12:
0x54ef441000516fdd' (lumi.motion.ac01) is newer than latest firmware online.
Please, can someone correct this?

closed 2025-01-01 by github-actions[bot]