Koenkk/zigbee2mqtt #22722: Zigbbee2mqtt does not start setting ember as adapter

Patricen1, 2024-05-21:
### What happened?

Zigbbee2mqtt does not start 

### What did you expect to happen?

Start 🙂

### How to reproduce it (minimal and precise)

Change configuration.yaml, replace adapter from auto to ember, set rtscrs to false, docker compose up.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.3.1.0 build 176

### Adapter

zbdongle-e EZSP v12

### Setup

Z2M on docker container using rpi3b

### Debug log

Tbd

francisp2, 2024-05-21:
> Adapter firmware version
> 7.3.1.0 build 176

You need firmware 7.4.1 or 7.4.3 for ember, with 7.3.1 Zigbee2MQTT will not start

Patricen1, 2024-05-21:
Oh thanks! I flashed my dongle using Silabs firmware flasher and it installed 7.3.1
I went to https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee, downloaded requested gbl file and tried to flash it using custom firmware option but got the following error. Any clue?
[image]

francisp2, 2024-05-21:
Be sure to download raw file

[image]

Patricen1, 2024-05-21:
Thanks a lot fot this valuable help, issue is now solved !

closed 2024-05-21 by Patricen1