Koenkk/zigbee2mqtt #22727: TS011F _TZ3000_cehuw1lw - no energy report

laplac, 2024-05-21:
### What happened?

I have many Tuya smart plugs TS011F and I bought additional 4 from Ali Express recently which came with firmware version 1.0.5. However, they are not reporting energy like the others. All are identified as TS011F_plug_1

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

20210708

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle P

### Setup

Home Assistant VM on Intel Qnap TS-251+

### Debug log

Logs:

 error: 	z2m: Failed to configure '0xa4c138a503ef1d55', attempt 3 (Error: ZCL command 0xa4c138a503ef1d55/1 haElectricalMeasurement.configReport([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":10}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:315:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:770:26)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:556:9)
    at Object.activePower (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:208:5)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:4260:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:172:21)
    at Configure.configure (/app/lib/extension/configure.ts:121:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17

DB:
{"id":89,"type":"Router","ieeeAddr":"0xa4c138a503ef1d55","nwkAddr":41165,"manufId":4107,"manufName":"_TZ3000_cehuw1lw","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"haElectricalMeasurement":{"attributes":{"rmsVoltage":240,"rmsCurrent":0,"activePower":0}},"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_cehuw1lw","powerSource":1,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"1.0.5\u0000"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"onTime":0,"offWaitTime":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0d69","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0d69","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0d69","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null}],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"1.0.5\u0000","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1716294191723}

State:
{
    "child_lock": "UNLOCK",
    "countdown": 0,
    "current": 0,
    "indicator_mode": "off/on",
    "last_seen": "2024-05-21T14:00:26+01:00",
    "linkquality": 87,
    "power": 0,
    "power_outage_memory": "on",
    "state": "ON",
    "voltage": 240,
    "energy": null,
    "update": {
        "installed_version": -1,
        "latest_version": -1,
        "state": null
    },
    "update_available": null
}

[image]

leonidostrovski, 2024-05-21:
It looks like the same issue:
https://github.com/Koenkk/zigbee2mqtt/issues/22720#issue-2306997739

laplac, 2024-05-21:
> It looks like the same issue: [#22720 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/22720#issue-2306997739)

Yes, same issue but I'm not using dev branch and I have older TS011F plugs working fine even after the update.

I think it has something to do with the plugs firmware version as it should be identified by z2m as TS011F_plug_3 not TS011F_plug_1.

https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html

`Starting with firmware version 1.0.5 (which comes pre-flashed on plugs produced since Q4 2021) core functionality on this plug is broken. TuYa has disabled the automatic reporting of power, voltage and current values meaning they need to be polled instead. The poll interval can be controlled through the measurement_poll_interval option.

If your plug is affected, it will be detected as TS011F_plug_3 instead of TS011F_plug_1`

leonidostrovski, 2024-05-21:
If I remember correctly, the last devices firmware update done by me  before a very long time. The issue with energy started before ~1 week ago.

Koenkk, 2024-05-22:
@laplac this one may already have been fixed in https://github.com/Koenkk/zigbee-herdsman-converters/commit/4182889448e29e1bcf8e4aae961fe5dea0fa0543, can you check with the latest [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)?

laplac, 2024-05-22:
Switched to Dev branch , all the new smart plugs with swBuild 1.0.5 have been successfully identified as TS011F_plug_3 and the old plugs stays as TS011F_plug_1 which is good news so my other plugs are not affected. 

The bad news is that energy still not picking up, I have updated the Measurement poll interval to 10 seconds and I got Volt, Current and Power but not Energy. I have tried to re-pair the plug but same outcome. 

On Dev Console when I query Endpoint-1->Cluster-seMetering->attribute-notificationControlFlags I got the below result
Read result of 'seMetering': {"notificationControlFlags":[0,3]} 

Which should indicate that my energy reading is 0.03 kWh but it's not reflecting in the z2m state. 

DB
{"id":89,"type":"Router","ieeeAddr":"0xa4c138a503ef1d55","nwkAddr":18534,"manufId":4107,"manufName":"_TZ3000_cehuw1lw","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"32768":0,"onOff":1,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":239,"rmsCurrent":0,"activePower":0,"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"notificationControlFlags":[0,3],"divisor":100,"multiplier":1}},"genBasic":{"attributes":{"hwVersion":0,"manufacturerName":"_TZ3000_cehuw1lw","zclVersion":3,"appVersion":112,"modelId":"TS011F","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"1.0.5\u0000","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1716406394962}

Logs
Info 2024-05-22 20:42:07MQTT publish: topic 'zigbee2mqtt/0xa4c138a503ef1d55', payload '{"child_lock":null,"current":0,"energy":null,"indicator_mode":"off/on","last_seen":"2024-05-22T20:42:06+01:00","linkquality":87,"power":0,"power_outage_memory":"on","state":"ON","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":238}'

laplac, 2024-05-23:
Update:

It looks like my older working plugs identified as TS011F_plug_1 stopped reporting new energy values and the last value is unchanged since I switched to Dev branch. Switched back to Stable branch and energy started to kick again. 

Looks like the dev branch is breaking old running TS011F_plug_1 .

[image]

laplac, 2024-05-25:
Update : 

After switching back to the stable branch from dev branch, all my plugs now identified as TS011F_plug_1 including the ones with swBuild 1.0.5 . I noticed that the energy now is coming up for the plugs with swBuild 1.0.5 where it's identified as TS011F_plug_1 instead of TS011F_plug_3.  

Not sure what happen. Something fixed it by switching to Dev then switch back to Stable. 
- Can't find any z2m logs for that period. 
- Now Frontend is a bit buggy sometimes it takes too long to load and give me an error "You have found a bug" but when I try to download logs it doesn't show anything.

Anyway the original issue was fixed by unknown method ! 

[image]
[image]

Koenkk, 2024-05-27:
I just pushed a possible fix, can you check with the latest dev?

leonidostrovski, 2024-05-30:
> I just pushed a possible fix, can you check with the latest dev?

Looks did not help (in my case)
**1.37.1-dev commit:9b3a394**
[image]

Thanks for support

laplac, 2024-06-06:
Updated to stable version 1.38.0-1

1.0.5 Tuya plugs correctly identified as TS011F_plug_3
previous version Tuya plugs stays at TS011F_plug_1 

All reporting energy as expected.

Thanks a lot for your support.

closed 2024-06-06 by laplac