Koenkk/zigbee2mqtt #22741: Zigate Parsing error: Error: Message '8008' not found

jmorille, 2024-05-22:
### What happened?

After Update to version 1.37.1 with docker version, 
- all zigbee device change there Adresse IEEE
- and many error in logs appear 
1. error:  `	zh:zigate:driver: Parsing error: Error: Message '8008' not found`
2. error: 	 `source endpoint 242, not supported`	

After repairing all device for the new   Adresse IEEE the schema map become without any link anymore
[image]

I try to the latest dev version with docker tag latest-dev and the error persist

### What did you expect to happen?

No error in IHM like
1. error:  `	zh:zigate:driver: Parsing error: Error: Message '8008' not found`
2. error: 	 `source endpoint 242, not supported`	

A Map Schema with link 

### How to reproduce it (minimal and precise)

My adapter is a `Zigate+ V2 USB` with the following config
```yaml
serial:
  port: /dev/serial/by-id/usb-FTDI_ZiGate_ZIGATE+-if00-port0
  adapter: zigate
  baudrate: 115200
  disable_led: true
  rtscts: true
```

### Zigbee2MQTT version

1.37.1-dev commit: b2d3265

### Adapter firmware version

ZiGateV2 v3.A0

### Adapter

Zigate+ V2 USB

### Setup

Zigbee2MQTT in docker with x86-64 with N100 CPU

### Debug log

[_zigbee2mqtt_logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/15407754/_zigbee2mqtt_logs.txt)

jmorille, 2024-05-22:
After rollback to `zigbee2mqtt 1.36.1`, the zigbee device go back to the all Address 
and after reparing it , the link in schema comme back

[image]

And no error like 

- Parsing error: Error: Message '8008' not found

Rainz3, 2024-05-22:
Do you have directions on rolling back? I'm having a massive amount of issues with 1.37.1

jmorille, 2024-05-23:
I take the direction to replace the zigate by another adapter (more popular)
- drivers problem like this one
- and bad range of communication
- and some feature like OTA are never work on my device

The address IEE change, let me think of a drivers problem like last bit lost
For a device, i have
- In 1.36.1 - The good address : 0x54ef4410007a29cd
- In 1.37.1 -  The bad address 0xcd297a001044ef54 : 

==> in bit
Good 0x54ef4410007a29cd= 0000 0101 0100 1110 1111 0100 0100 0001 0000 0000 0000 0111 1010 0010 1001 1100 1101
Bad    0xcd297a001044ef54= 1100 1101 0010 1001 0111 1010 0000 0000 0001 0000 0100 0100 1110 1111 0101 0100

Try another deivce
Good 0x00158d0009ef3e1e= 0000 0000 0001 0101 1000 1101 0000 0000 0000 1001 1110 1111 0011 1110 0001 1110
Bad    0x1e3eef09008d1500= 0001 1110 0011 1110 1110 1111 0000 1001 0000 0000 1000 1101 0001 0101 0000 0000

Ok bad idea .. no matching pattern

But anyway,
thank to docker instance, is easy to have dual run 
so
I could do some test with zigate I you need to validate a version with the zigate adpater

Neibafsama, 2024-05-30:
I also have the same type of error with a Zigate usb key in 1.37.1

1. Error: "Source Endpoint 242, Not Supported" 
2. Error: Devices that are no longer responding, they are recreated with a new ID address.

superbricolo, 2024-06-08:
I also have the same error with a Zigate Wifi in 1.38.0
- Parsing error: Error: Message '100' not found
- and new ID addres also

aol3556, 2024-07-06:
Hi All,

Same issue since 1.37.0 (I tried all versions to 1.39.0).
- zh:zigate:driver: Parsing error: Error: Message '8008' not found
- zh:zigate: source endpoint 242, not supported
- IEEE Adress change (inverted)
- Loose some device like Ikea and cannot join them again to the network

Thanks for any solution.

aol3556, 2024-07-07:
Hi,

I rollbacked to 1.36-1 and all seems to work well. I can associate all my devices that doesn't join network with >=1.37.0 versions.

Let me know how to help you to debug this issue.

BR

floutchito, 2024-08-11:
Hi all,

Probably linked issue, with the same errors using ZiGate+ (v2) and Z2M v1.37+ :

https://github.com/Koenkk/zigbee2mqtt/issues/22970

aol3556, 2024-09-06:
For information this is merged in version 1.40.0 : https://github.com/Koenkk/zigbee-herdsman/pull/1170

closed 2025-01-01 by github-actions[bot]