Koenkk/zigbee2mqtt #22744: Legrand 067646

cocapic, 2024-05-23:
### What happened?

inconsistant actions results

I have 11 Legrand 067646 shutter switch, firmware 1e or latest 46

actions reported are not consistant ! we need to press multiple time trying all possibility to have a miracle : the right action executed

I recently moved from deconz and switchs have a different behavior on it :

> Montée : 1002 / Montée puis relâché : 1002/3002
> Descente : 2002 / Descente puis relâché : 2002/3003
> Les deux en même temps : 3003

And consistant actions, everything related was working fine

Is there annyway to change z2m interpretation of what the switch is giving ?

support of this product need to be removed, nothing at all can be done

Regards

### What did you expect to happen?

consistant operation

### Zigbee2MQTT version

 1.37.0 commit: ee5cf5a

### Adapter firmware version

20230507

### Adapter

sonoff

### Setup

pi5 jeedom

Tonio16, 2024-05-23:
Hello

Are you using binding? If yes, binding to a group or a device?

Antoine

cocapic, 2024-05-23:
Hi,No binding, automation thru Jeedom behindMerci, Regards

pipolaq, 2024-06-13:
Hello,
Exactly same problem here on 1.36.1
Inconsistent, I need to press 10 or 20 times a button for it to work.
What is weird is the logs :

In exemple, I pressed the close button, nothing happened in Home assistant and I've got these lines in the logs :

`**Info 2024-06-13 19:25:46**MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Volet parents","ieee_address":"0x0004740000a97b5b"},"type":"device_announce"}'

**Debug 2024-06-13 19:25:46**Received Zigbee message from 'Volet parents', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID 0

**Debug 2024-06-13 19:25:46**Received Zigbee message from 'Volet parents', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":29}' from endpoint 1 with groupID 0

**Info 2024-06-13 19:25:46**MQTT publish: topic 'zigbee2mqtt/Volet parents', payload '{"action":null,"battery":null,"linkquality":66,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2900}'`

And each times I have a trouble i have the same 2 infos :

1. I do have the "received Zigbee message" with the correct type
2. I have "MQTT publish" with "action: null"

I think the problem is the "action: null" whereas the right zigbee message seems received

cocapic, 2024-06-14:
Hi pipolaq,

im new on Z2M, did you have this issue previously ?
i have 11 swith, a nightmare, thinking moving back on deconz

regards

pipolaq, 2024-06-14:
Hi @cacapic,
I just received these legrand shutter switch a few days ago, so first time I linked them to Z2M.
I added some legrand wired shutter switch, everything went well (including OTA) and they work perfectly with Z2M.
But I have troubles with my wireless switch. I should receive 4 new wired shutter switch next week, I'll see how it goes !

Does someone understand why Z2M seems to receive the right order, but doen't send it to MQTT ?

Thanks !

cocapic, 2024-06-21:
Hi,

if you need more shutter wireless switch, dont hesitate to come back to me, I sell them :)

regards

cocapic, 2024-07-05:
same issue in 1.39.0

but i noticed something strange, i came home 2 hours after upgrade and tested all switchs and all was working fine, then this morning all dead again
Is it something reset during the update or else that will having all of them working correctly?

regards

cocapic, 2024-07-24:
Hi @collicalex,

You think about something between z2m and jeedom like action too short in duration to trigger in jeedom ?

hum yes something is happening but right action before stopped 100% not sure

++

hlehoux2021, 2024-07-24:
I see that @pipolaq has the same behavior using Home Assistant, so i think this may not be related to jeedom or home assistant but to z2m

collicalex, 2024-07-24:
Hello,

So, after some help of Jeedom staff, Zigbee2MQTT have a default parameter of 0.3s to temporary send back data.
I've put it to 0 (re-launch deamon) and now it's working fine at 99% of time.
Some time it's still sending only "stopped", but I can also see this in Zigbee2MQTT interface...

cocapic, 2024-07-25:
hi,

where is this parameter ?

ty

collicalex, 2024-07-25:
In Jeedom, you have to go in menu Plugins > Programmation > MQTT Manager
Then go to Configuration.
And you have a parameter named Cycle with value 0.3

I've put 0, Save, and restart the deamon.

I've also gone back to Plugins > Protocol Domotique > JeeZigBee and restart the daemon.

cocapic, 2024-07-25:
Hum MQTT Manager is used by other things, hope will not have other impact

will try, many thanks

collicalex, 2024-07-25:
Also it's better in Jeedom...

But the Legrand shutter is still do some incomprehensible message sending...
Sometime push the up or down button send Open/Stopped or Close/Stopped very well.
Sometime it's sending only Stopped...

I do not understand what's Legrand done with this shutter :/
They just have to send 1 command per button : Open / Close or Stop (central button); that's all. Here they try to complicate everything sending muti states per button.

collicalex, 2024-07-25:
Deconz work only with a conbee key?
I have a sonoff zdongle-e

collicalex, 2024-07-25:
Also with Deconz, do you see Open then Stopped / Close then Stopped messages?

cocapic, 2024-07-25:
dont remembrer key, will check once back from vacation :)

code received from switch in deconz
Montée : 1002 / Montée puis relâché : 1002/3002
Descente : 2002 / Descente puis relâché : 2002/3003
Les deux en même temps : 3003

collicalex, 2024-07-26:
Also, I just use a Conbee II usb receiver instead of the Sonoff ZDongle-E.
I've also some troubles with received message.
I think the device send the 2 messages too fast and some are just not seen.

In the device page, Expose page.
In the Action we can see the value.
It's really really quick but 2 values are displayed Open then Stopped (or Close then Stopped), when it's working.
When it's not.. we can see only Stopped value (no quick change).

Also when the message is sent to Jeedom (via MQTT) I can see that sometime it sent Stopped then Open values (i.e. not the correct order) even if that the last message displayed in Device page > Expose > Action is "Stopped".

collicalex, 2024-07-29:
Hello,

So I've made some search...

The decoding of zigbee messages from Legrand device (and this one 67646) is done via 
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/legrand.ts

For our device (search for the id 67646), lines 110-137, there is a 
fz.legrand_binary_input_moving
line 122.

As per a comment on this post https://github.com/Koenkk/zigbee2mqtt/issues/15101  this if for "support binary report on moving state (supposed)".

But the device 67646 is a wireless buttons for shutter. How can the device know if the shutter is moving or stopped? It's impossible.
It must only expose the 3 commands: open, close, stop.

I think the problem come from here, it try to decode something for nothing and add / put some wrong messages...

@Koenkk is it possible to patch this file locally to test? How?

Regards

collicalex, 2024-07-31:
Hello,

We can simplify a little bit more the legrand.js for this device.
Here is what I have, and it's still working (but still no battery display).

[image]

Hope it will be put in an official release soon?

Regards,

Alex

collicalex, 2024-08-01:
Finally... When adding a second shutter... this second one is not working (no action).
When adding a third shutter.. all sutters (first, second and this third) are not working (no action).

I've tried to put back the config I've posted 2 messages above... Restart all. The same.

I'me migrating back to Deconz / Phoscon.

Koenkk, 2024-08-03:
> Hope it will be put in an official release soon?

@collicalex you can submit a pull request by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/legrand.ts)

pikatoste81, 2024-08-04:
I have the same problem. The command has never worked correctly and when buying a second the same thing happened to me.

I hope you solve the problem soon!

CodeForLove83, 2024-08-23:
Hello,
I've been addressing the issues with the converter for both battery and incorrect/missed actions.
I experienced the same problem as you all. I have 3 Legrand 067646 devices, and they've been operating flawlessly for 48 hours now.
Could you test this code as well and let me know if it resolves the issue? If it does, I'll submit a PR.

```
const {fzLegrand, tzLegrand, readInitialBatteryState, legrandOptions} = require('zigbee-herdsman-converters/lib/legrand');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const device = {
        zigbeeModel: [' Shutters central remote switch'],
        model: '067646',
        vendor: 'Legrand',
        description: 'Wireless shutter switch (WIP)',
        ota: ota.zigbeeOTA,
        meta: {multiEndpoint: true, battery: {voltageToPercentage: {min: 2500, max: 3000}}, publishDuplicateTransaction: true},
        fromZigbee: [
            fz.identify,
            fz.ignore_basic_report,
            fz.command_cover_open,
            fz.command_cover_close,
            fz.command_cover_stop,
            fz.battery,
        ],
        toZigbee: [],
        exposes: [e.battery(), e.action(['identify', 'open', 'close', 'stop'])],
        onEvent: async (type, data, device, options, state) => {
            await readInitialBatteryState(type, data, device, options, state);
            if (data.type === 'commandCheckin' && data.cluster === 'genPollCtrl') {
                const endpoint = device.getEndpoint(1);
                await endpoint.command('genPollCtrl', 'fastPollStop', {}, legrandOptions);
            }
        },
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering', 'genIdentify']);
        },
    };

module.exports = device;

cocapic, 2024-08-26:
Hi @CodeForLove83 ,

still working for you ?
also where/how to apply your code ?

ty

CodeForLove83, 2024-08-26:
Hi,

The previous code didn't work as expected; sometimes multiple and spurious actions were received.

By sniffing the communication with the Legrand Gateway, I discovered that Legrand devices are added to a group and the group is then bound to the endpoint Wireless Legrand device.

It seems that binding to a group enables the Legrand devices to handle multiple actions properly, likely discarding them through the firmware of the Legrand devices within the group.

I have also examined issues reported with other Wireless Legrand devices, such as shutters and switches, to validate my assumption that the problem we are experiencing with 067646 is similar to those affecting other wireless endpoints.

Subsequently, I refined my code, although I now believe it may no longer be necessary (see below).

In summary, the remedy is to **create** a **group** with the Legrand devices you wish to manage using the endpoint Wireless Legrand device (067646). This **group** needs to be **bound to the endpoint Legrand Wireless device**.

Please, let me know if it works for you guys.

Currently, my wife has no complaints :-)

[image]

[image]

[image]

Be aware that the endpoint Legrand Wireless device couldn't be awake, press any device button and the click on bind button to success with binding. 

To support new devices or modify an existing one, please visit this [link](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html).

A quick process walkthroughs is illustrated in the following screenshots.

[image]

You need to add to `/homeassistant/zigbee2mqtt/configuration.yaml` (or at the path where you saved the `configuration.yaml` for your z2m installation) the following code snippet:
```
external_converters:
  - 067646.js
```
[image]

The file with the external definition `067646.js` must be created at the path `/homeassistant/zigbee2mqtt/` where the `configuration.yaml` is stored.

The last step is to restart `zigbee2mqtt` process. 

```
legrand_binary_input_moving} = require('zigbee-herdsman-converters/lib/modernExtend');
const {fzLegrand, tzLegrand, readInitialBatteryState, legrandOptions} = require('zigbee-herdsman-converters/lib/legrand');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    command_cover_close: {
        cluster: 'closuresWindowCovering',
        type: 'commandDownClose',
        convert: (model, msg, publish, options, meta) => {
            return {action: 'close'};
        },
    },
    
    command_cover_open: {
        cluster: 'closuresWindowCovering',
        type: 'commandUpOpen',
        convert: (model, msg, publish, options, meta) => {
            return {action: 'open'};
        },
    },
    
    command_cover_stop: {
        cluster: 'closuresWindowCovering',
        type: 'commandStop',
        convert: (model, msg, publish, options, meta) => {
            return {action: 'stop'};
        },
    },
};

const device = {
        zigbeeModel: [' Shutters central remote switch'],
        model: '067646',
        vendor: 'Legrand',
        description: 'Wireless shutter switch (WIP)',
        ota: ota.zigbeeOTA,
        meta: {multiEndpoint: true, battery: {voltageToPercentage: {min: 2500, max: 3000}}}, publishDuplicateTransaction: false},
        fromZigbee: [
            fz.identify,
            fz.ignore_basic_report,
            fzLocal.command_cover_open,
            fzLocal.command_cover_close,
            fzLocal.command_cover_stop,
            //fromZigbee_battery,
            fz.battery,
            fz.legrand_binary_input_moving,
        ],
        toZigbee: [],
        exposes: [e.battery(), e.action(['identify', 'open', 'close', 'stop', 'moving', 'stopped'])],
        onEvent: async (type, data, device, options, state) => {
            await readInitialBatteryState(type, data, device, options, state);
            if (data.type === 'commandCheckin' && data.cluster === 'genPollCtrl') {
                const endpoint = device.getEndpoint(1);
                await endpoint.command('genPollCtrl', 'fastPollStop', {}, legrandOptions);
            }
        },
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genBinaryInput', 'closuresWindowCovering', 'genIdentify']);
        },
    };

module.exports = device;
```

CodeForLove83, 2024-08-27:
Although I have a conbee3, I haven't used deconz yet but based on what I read on other posts deconz seems to create group by default for each device and that's the reason why it probably works. You can avoid to use my code, it basically implements just the battery, and try to create group and binding it to the wireless device.
In the meantime I dig a bit further in the traffic I've been sniffing from the Legrand gateway to understand if everything is properly implemented in z2m.
Waiting for you feedback.

CodeForLove83, 2024-08-27:
> Hi @CodeForLove83
> 
> many thanks for your time
> 
> Im not sure that binding is the solution, i was previously on deconz and all legrand stuff was working fine without any setup (i ddnt saw something so reliable that legrand/deconz :) ) ...
> 
> Also in my case, endpoint are not zigbee and not in Z2m i cannot create an endpoint ...

what do you mean?
I expect you want to control Legrand Shutter ([K4027C/L4027C/N4027C/NT4027C](https://www.zigbee2mqtt.io/devices/K4027C_L4027C_N4027C_NT4027C.html#bticino-k4027c%252Fl4027c%252Fn4027c%252Fnt4027c)) with Legrand 067646. Am I wrong?
If that's the case you can follow the steps I mentioned above.

> 
> Ive set your code in prod, will see at the end of the day
> 
> hope it will work ;)
> 
> ty

cocapic, 2024-08-27:
im using 067646 as wireless switch only to trigger shutter motor wifi relay (shelly) thru my gateway (jeedom)

CodeForLove83, 2024-08-27:
> im using 067646 as wireless switch only to trigger shutter motor wifi relay (shelly) thru my gateway (jeedom)

Another trick I read in other posts is creating an empty group to be used through automation.
Read this [link](https://github.com/Koenkk/zigbee2mqtt/issues/14894)

cocapic, 2024-08-27:
great link,
im not able to build an automation based on group, but from my understanding the only fact that switch is related to a group will work

> As part of its configuration, the remote expects to have a "group" binding configured for each of its button. Otherwise, it will consider itself not fully configured and send unnecessary commands on the zigbee network resulting in noticeable delay when controlling your device(s).
In a nutshell:
- When using devices bindings directly in zigbee2mqtt, create groups with your target devices for each button and bind related groups accordingly.
- When using automations to control your devices (e.g. homeassistant), create a virtual group (empty) and bind it to the buttons. Beware that having both the coordinator and group bound to a remote's button will publish 2 MQTT messages

cocapic, 2024-09-02:
Hi,

nothing relevant on my both setup ... :(
dumb group doesnt help

ive keep only open,close,stop on your external definition, action exposed are now open_1, close_1, stop_1, that is not a issue, but old actions : moving, stopped are still comming back also :)

there is a way to have external config overwrite completely the onboarded definition ?

regards

tchlyah, 2024-09-03:
> Hi,
> 
> The previous code didn't work as expected; sometimes multiple and spurious actions were received.
> 
> By sniffing the communication with the Legrand Gateway, I discovered that Legrand devices are added to a group and the group is then bound to the endpoint Wireless Legrand device.
> 
> It seems that binding to a group enables the Legrand devices to handle multiple actions properly, likely discarding them through the firmware of the Legrand devices within the group.
> 
> I have also examined issues reported with other Wireless Legrand devices, such as shutters and switches, to validate my assumption that the problem we are experiencing with 067646 is similar to those affecting other wireless endpoints.
> 
> In summary, the remedy is to **create** a **group** with the Legrand devices you wish to manage using the endpoint Wireless Legrand device (067646). This **group** needs to be **bound to the endpoint Legrand Wireless device**.
> 
> Please, let me know if it works for you guys.
> 
> Currently, my wife has no complaints :-)
> 
> Be aware that the endpoint Legrand Wireless device couldn't be awake, press any device button and the click on bind button to success with binding.
> 
> If you want to test also my code please read the following paragraph. To support new devices or modify an existing one, please visit this [link](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html).
> 
> A quick process walkthroughs is illustrated in the following screenshots.
> 
> You need to add to `/homeassistant/zigbee2mqtt/configuration.yaml` (or at the path where you saved the `configuration.yaml` for your z2m installation) the following code snippet:

@CodeForLove83 I just tested your solution as I was desperate to make it work reliably. And as of now, it is working perfectly fine. However, now instead of returning `stop`, `open` and `close`, I get `stop_1`, `open_1` and `close_1`. Any idea why ?

Anyway, I have created a blueprint for ease of control of covers, covering the `_1` actions as well : 

```yml
blueprint:
  name: Control Cover with Legrand Shutter Switch
  description: Handles close, open, and stop actions (including variants with "_1") from a Legrand Shutter Switch to control one or more covers.
  domain: automation
  input:
    trigger_device:
      name: Trigger Device
      description: The Legrand Shutter Switch that triggers the cover actions.
      selector:
        device: {}
    covers:
      name: Covers
      description: The covers to be controlled by the trigger device.
      selector:
        target:
          entity:
            domain: cover

trigger:
  - platform: device
    device_id: !input trigger_device
    domain: mqtt
    type: action
    subtype: close
    id: close
  - platform: device
    device_id: !input trigger_device
    domain: mqtt
    type: action
    subtype: close_1
    id: close_1
  - platform: device
    device_id: !input trigger_device
    domain: mqtt
    type: action
    subtype: open
    id: open
  - platform: device
    device_id: !input trigger_device
    domain: mqtt
    type: action
    subtype: open_1
    id: open_1
  - platform: device
    device_id: !input trigger_device
    domain: mqtt
    type: action
    subtype: stop
    id: stop
  - platform: device
    device_id: !input trigger_device
    domain: mqtt
    type: action
    subtype: stop_1
    id: stop_1

condition: []

action:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - close
              - close_1
        sequence:
          - service: cover.close_cover
            target: !input covers
      - conditions:
          - condition: trigger
            id:
              - open
              - open_1
        sequence:
          - service: cover.open_cover
            target: !input covers
      - conditions:
          - condition: trigger
            id:
              - stop
              - stop_1
        sequence:
          - service: cover.stop_cover
            target: !input covers

mode: single
```

cocapic, 2024-09-03:
@tchlyah no more "moving", "stopped" actions ?

tchlyah, 2024-09-03:
> @tchlyah no more "moving", "stopped" actions ?

@cocapic I still have them, but they happen after `open_1` or `close_1` actions, it doesn’t really bother, and the automation is very reliable each time.

CodeForLove83, 2024-09-03:
Hi, I don't think is reliable. The issue seems to be fixed only with groups.

tchlyah, 2024-09-03:
> Hi, I don't think is reliable. The issue seems to be fixed only with groups.

I tested previously the virtual groups alone and I stopped receiving the actions all together... As of now, the groups with conjunction of your external definition is working fine

CodeForLove83, 2024-09-03:
Yes, that's correct. I didn't get you were using virtual groups.

cocapic, 2024-09-03:
groups ddnt solve any issue for me :(

when ur talking about virtual groups you meen on z2m side ?, because you seems HA users

tchlyah, 2024-09-03:
> groups ddnt solve any issue for me :(
> 
> when ur talking about virtual groups you meen on z2m side ?, because you seems HA users

Yes, you only need to create an empty group in z2m, then assign it to your switch, as mentioned in the [post](#issuecomment-2310692925) shared by @CodeForLove83.
<img alt="image" width="1572" src="https://private-user-images.githubusercontent.com/172607490/361513535-37fdc8df-5f0a-4633-a513-bec8d1100d1b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNTU0NDEsIm5iZiI6MTcyNTM1NTE0MSwicGF0aCI6Ii8xNzI2MDc0OTAvMzYxNTEzNTM1LTM3ZmRjOGRmLTVmMGEtNDYzMy1hNTEzLWJlYzhkMTEwMGQxYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwM1QwOTE5MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yMDE2YjE5MmU1NzFkZWM5MzY5MjUxYmU5MjhiNjcwYTAxZGZmODRmYWQxNzc2ZjBhODM4MGE1OTJiOTcxMWY3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.XNfR-8mmN_pcbg-wOc_-nVw4Joixs5Yd8Isf16x8D48">

Don't forget the js external converter as well.

cocapic, 2024-09-04:
Hi,

This morning all my 11x67646 still not working correctly (multiple press to have it wokeup)

Ive removed "stopped" and "moving" from included converter and everywhere in the z2m using grep/sed (because external converter on top of the core one was not persistent)

Also 2x of my switch was related to a dumb group and it doesnt help at all

Same trial on my second setup this morning with same behavior

https://github.com/user-attachments/assets/e4fadf2a-89e2-4040-a819-e0e0aa57b05e

Also ive "smartly loged" this time a phantom press with no action on z2m exept update of LQI "144"
from what i see in logs, z2m is receiving LQI update but also a "downClose" order but ddnt do something with it and not publishing it on mqqt
[log.txt](https://github.com/user-attachments/files/16862435/log.txt)

[image]

`[2024-09-04 08:16:31] debug: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":154,"clusterId":15,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lastHopLqi":140,"lastHopRssi":-65,"sender":6127,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,3,10,85,0,16,1]}}
[2024-09-04 08:16:31] debug: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":154,"clusterId":15,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":140,"rssi":-65,"sender":6127,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[24,3,10,85,0,16,1]}}
[2024-09-04 08:16:31] debug: 	zh:controller: Received payload: clusterID=15, address=6127, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=140, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":3,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":16,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-09-04 08:16:31] debug: 	z2m: Received Zigbee message from '0x0004740000c10db3', type 'attributeReport', cluster 'genBinaryInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
[2024-09-04 08:16:31] debug: 	z2m: No converter available for '067646' with cluster 'genBinaryInput' and type 'attributeReport' and data '{"presentValue":1}'
[2024-09-04 08:16:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0004740000c10db3', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020201217\u0000\u0000\u0000\u0000\u0000","friendlyName":"0x0004740000c10db3","hardwareVersion":7,"ieeeAddr":"0x0004740000c10db3","manufacturerID":4129,"manufacturerName":" Legrand","model":"067646","networkAddress":6127,"powerSource":"Battery","softwareBuildID":"002a","stackVersion":66,"type":"EndDevice","zclVersion":2},"last_seen":"2024-09-04T07:16:31.509Z","linkquality":140,"update":{"installed_version":2769411,"latest_version":4932099,"state":"available"},"voltage":2800}'
[2024-09-04 08:16:31] debug: 	zh:ezsp:uart: <-- [5637b1a97d312a15b658964b24ab5593499cd2b76b02d99874fed76188fe75ec]
[2024-09-04 08:16:31] debug: 	zh:ezsp:uart: <-- [7e]
[2024-09-04 08:16:31] debug: 	zh:ezsp:uart: <-- DATA (5,6,0): 5637b1a9112a15b658964b24ab5593499cd2b76b02d99874fed76188fe75ec7e
[2024-09-04 08:16:31] debug: 	zh:ezsp:uart: --> ACK  (6)
[2024-09-04 08:16:31] debug: 	zh:ezsp:uart: --> [8610be7e]
[2024-09-04 08:16:31] debug: 	zh:ezsp:uart: <-- ACK (6): 5637b1a9112a15b658964b24ab5593499cd2b76b02d99874fed76188fe75ec7e
[2024-09-04 08:16:31] debug: 	zh:ezsp:ezsp: <== Frame: 759001450000040102010101000100009c90c0ef17ffff0311020102
[2024-09-04 08:16:31] debug: 	zh:ezsp:ezsp: <== 0x45: {"_cls_":"incomingMessageHandler","_id_":69,"_isRequest_":false,"type":0,"apsFrame":{"profileId":260,"sequence":156,"clusterId":258,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lastHopLqi":144,"lastHopRssi":-64,"sender":6127,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,2,1]}}
[2024-09-04 08:16:31] debug: 	zh:ezsp: processMessage: {"messageType":0,"apsFrame":{"profileId":260,"sequence":156,"clusterId":258,"sourceEndpoint":1,"destinationEndpoint":1,"groupId":0,"options":256},"lqi":144,"rssi":-64,"sender":6127,"bindingIndex":255,"addressIndex":255,"message":{"type":"Buffer","data":[17,2,1]}}
[2024-09-04 08:16:31] debug: 	zh:controller: Received payload: clusterID=258, address=6127, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=144, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":2,"commandIdentifier":1},"payload":{},"command":{"ID":1,"parameters":[],"name":"downClose"}}
[2024-09-04 08:16:31] debug: 	z2m: Received Zigbee message from '0x0004740000c10db3', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID 0
[2024-09-04 08:16:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0004740000c10db3', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020201217\u0000\u0000\u0000\u0000\u0000","friendlyName":"0x0004740000c10db3","hardwareVersion":7,"ieeeAddr":"0x0004740000c10db3","manufacturerID":4129,"manufacturerName":" Legrand","model":"067646","networkAddress":6127,"powerSource":"Battery","softwareBuildID":"002a","stackVersion":66,"t [... cut ...]
[2024-09-04 08:16:41] debug: 	zh:ezsp:uart: <-- DATA (6,7,0): 6734a1a9512a1bc87e
[2024-09-04 08:16:41] debug: 	zh:ezsp:uart: --> ACK  (7)
[2024-09-04 08:16:41] debug: 	zh:ezsp:uart: --> [87009f7e]
[2024-09-04 08:16:41] debug: 	zh:ezsp:uart: <-- ACK (7): 6734a1a9512a1bc87e
[2024-09-04 08:16:41] debug: 	zh:ezsp:ezsp: <== Frame: 7680010500
[2024-09-04 08:16:41] debug: 	zh:ezsp:ezsp: <== 0x5: {"_cls_":"nop","_id_":5,"_isRequest_":false}
[2024-09-04 08:16:41] debug: 	zh:ezsp:uart: -+- waiting (7) success
`

cocapic, 2024-09-10:
Hi,

Could some one explain me how converters work ?
where is the code or matrix who handle data received and what fonction is called

ty

closed 2025-01-01 by github-actions[bot]