Koenkk/zigbee2mqtt #22746: Sonoff ZBDongle-P not working in Windows at all, Sonoff ZBDongle-E working only with ember driver

LamerTex, 2024-05-23:
### What happened?

I've thoroughly followed [the installation instruction for Windows](https://www.zigbee2mqtt.io/guide/installation/05_windows.html), but even after trying everything I'm not able to correctly start Zigbee2Mqtt on a Windows machine using the Sonoff ZBDongle-P and I have no more idea on what can I try to do.

Here is a little on what I've tried:
I have bought both a Sonoff ZBDongle-P and a Sonoff ZBDongle-E.

Sonoff ZBDongle-P: 
- Driver not found by Windows and also not working with "standard" installation of [VCP-Driver from FTDI](https://ftdichip.com/drivers/vcp-drivers/), I had to manually select the FTDI "USB Serial port" with Windows telling me it cannot be sure the driver is right. I've tried several other drivers without differences in the result
[image]
- Z2M in Windows is not working, I've pasted its output under the "Debug Log" section
- Z2M in WSL (Ubuntu 18.04) with USB passthrough using "usbipd" works without any problem
- Updated dongle firmware to the latest available (20230507) -> also done in WSL because in Windows I was not able to update it in any way
- Z2M still works in WSL 
- Z2M still not working in Windows

Sonoff ZBDongle-E: 
- Driver installed directly by Windows and displayed as "USB-Enhanced-SERIAL CH9102"
- Z2M on Windows is not working with the same output as the Sonoff ZBDongle-P
- Z2M in WSL (Ubuntu 18.04) with USB passthrough using "usbipd" works without any problem
- Updated dongle firmware to the lastest available (7.4.2 [GA]) -> done directly from the [Silabs firmware builder website](https://darkxst.github.io/silabs-firmware-builder/)
- Z2M still works in WSL 
- Z2M still not working in Windows
- Added "adapter: ember" in configuration.yaml
- Z2M work in Windows

 

### What did you expect to happen?

From what I can see I thought there was a problem with the Windows driver for communicating with the ZBDongle-P (since it works without problems under WSL and I was not even able to update its firmware under Windows), but since the ZBDongle-E also doesn't work until the activation of the ember driver maybe there is also a problem in the "standard" driver is not working at all in Windows?

Next week I will return the ZBDongle-P and in the future I will use only the ZBDongle-E.
If you have some other idea or test I can do to understand why the ZBDongle-P (and also the ZBDongle-E without activating the ember driver) is not working in Windows I'm always available.

Thanks for the amazing project!

### How to reproduce it (minimal and precise)

configuration.yaml used (I've tried both \\.\COM4 and COM4 directly):

```
homeassistant: false
frontend: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost
serial:
  port: \\.\COM4
advanced:
  network_key:
    - 119
    - 255
    - 138
    - 155
    - 110
    - 89
    - 154
    - 132
    - 54
    - 17
    - 35
    - 238
    - 232
    - 5
    - 81
    - 219
  pan_id: 10650
  ext_pan_id:
    - 206
    - 207
    - 244
    - 17
    - 87
    - 204
    - 227
    - 232
```

### Zigbee2MQTT version

1.37.1 (commit #cda867a3)

### Adapter firmware version

20230507

### Adapter

ZBDongle-P

### Setup

Plain on x64 Windows laptop 

### Debug log

npm start

> zigbee2mqtt@1.37.1 start
> node index.js

[2024-05-23 10:48:31] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-23 10:48:31] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a3)
[2024-05-23 10:48:31] info:     z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-23 10:48:31] error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: EINVAL: invalid argument, lstat '\\.\COM4\''
[2024-05-23 10:48:31] info:     zh:zstack:znp: Opening SerialPort with {"path":"\\\\.\\COM4","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-05-23 10:48:31] info:     zh:zstack:znp: Serialport opened
[2024-05-23 10:48:32] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-05-23 10:48:33] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-05-23 10:48:51] error:    z2m: Error while starting zigbee-herdsman
[2024-05-23 10:48:51] error:    z2m: Failed to start zigbee
[2024-05-23 10:48:51] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-23 10:48:51] error:    z2m: Exiting...
[2024-05-23 10:48:51] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\zStackAdapter.ts:101:27)
    at Controller.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:124:29)
    at Zigbee.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\lib\zigbee.ts:62:27)
    at Controller.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\lib\controller.ts:109:27)
    at start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\index.js:107:5)

Nerivec, 2024-05-25:
Can you provide a log with `debug` level?

I don't have the Dongle-P, so no idea there, but I can confirm the Dongle-E works on Windows, with both `ezsp` and `ember`. Auto-detection does not work however (due to OS driver overwriting some parameters), you need to specify `adapter: xyz`.

LamerTex, 2024-05-25:
> Can you provide a log with `debug` level?
> 
> I don't have the Dongle-P, so no idea there, but I can confirm the Dongle-E works on Windows, with both `ezsp` and `ember`. Auto-detection does not work however (due to OS driver overwriting some parameters), you need to specify `adapter: xyz`.

Yes you are right, the Dongle-E works correctly with both `ezsp` and `ember` only without the option it doesn't works, thanks!

### Dongle-P
The Dongle-P instead still doesn't work, when starting Z2M with the options:
```
serial:
  adapter: zstack
  port: \\.\COM4
advanced:
  log_level: debug
```
This is the log if I install the Windows driver "CDM-v2.12.36.4.U-WHQL-Certified"  from [FTDI chip's website](https://ftdichip.com/drivers/vcp-drivers/) as recommended in the installation instruction for the Dongle-P since it is based on the CC2652P chipset. 
Windows doesn't recognize it as the correct driver for the device and I have to "force it" to install, then the dongle is displayed just as "USB Serial Port (COM4)"
```
npm start

> zigbee2mqtt@1.37.1 start
> node index.js

[2024-05-25 17:13:38] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-25 17:13:38] debug:    z2m: Loaded state from file D:\Working\LocalProjects\MQTT\zigbee2mqtt\data\state.json
[2024-05-25 17:13:38] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a3)
[2024-05-25 17:13:38] info:     z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-25 17:13:38] debug:    z2m: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"D:\\Working\\LocalProjects\\MQTT\\zigbee2mqtt\\data\\coordinator_backup.json","databaseBackupPath":"D:\\Working\\LocalProjects\\MQTT\\zigbee2mqtt\\data\\database.db.backup","databasePath":"D:\\Working\\LocalProjects\\MQTT\\zigbee2mqtt\\data\\database.db","network":{"channelList":[11],"extendedPanID":[206,207,244,17,87,204,227,232],"networkKey":"HIDDEN","panID":10650},"serialPort":{"adapter":"zstack","path":"\\\\.\\COM4"}}'
[2024-05-25 17:13:39] error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: EINVAL: invalid argument, lstat '\\.\COM4\''
[2024-05-25 17:13:39] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[119,255,138,155,110,89,154,132,54,17,35,238,232,5,81,219],"panID":10650,"extendedPanID":[206,207,244,17,87,204,227,232],"channelList":[11]},"serialPort":{"path":"\\\\.\\COM4","adapter":"zstack"},"databasePath":"D:\\Working\\LocalProjects\\MQTT\\zigbee2mqtt\\data\\database.db","databaseBackupPath":"D:\\Working\\LocalProjects\\MQTT\\zigbee2mqtt\\data\\database.db.backup","backupPath":"D:\\Working\\LocalProjects\\MQTT\\zigbee2mqtt\\data\\coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
[2024-05-25 17:13:39] info:     zh:zstack:znp: Opening SerialPort with {"path":"\\\\.\\COM4","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-05-25 17:13:39] info:     zh:zstack:znp: Serialport opened
[2024-05-25 17:13:39] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-25 17:13:39] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-25 17:13:39] info:     zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-05-25 17:13:39] debug:    zh:zstack:unpi:writer: --> buffer [239]
[2024-05-25 17:13:40] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-25 17:13:40] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-25 17:13:40] info:     zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-05-25 17:13:41] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-25 17:13:41] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-25 17:13:47] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-25 17:13:47] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-25 17:13:53] debug:    zh:zstack:znp: SREQ: --> SYS - ping - {"capabilities":1}
[2024-05-25 17:13:53] debug:    zh:zstack:unpi:writer: --> frame [254,0,33,1,32]
[2024-05-25 17:13:59] error:    z2m: Error while starting zigbee-herdsman
[2024-05-25 17:13:59] error:    z2m: Failed to start zigbee
[2024-05-25 17:13:59] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-25 17:13:59] error:    z2m: Exiting...
[2024-05-25 17:13:59] error:    z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
```

This instead is the (much longer) log if I install the "wrong" driver "CP210x_Universal_Windows_Driver" from [Silicon Labs](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads) , in this case Windows install the driver without the need to force it and then the dongle is displayed as "Silicon Labs CP210x USB to UART Bridge (COM4)".
Configuration.yaml is the same as the previous log.

```
npm start

> zigbee2mqtt@1.37.1 start
> node index.js

[2024-05-25 17:18:54] info:     z2m: Logging to console, file (filename: log.log)
[2024-05-25 17:18:54] debug:    z2m: Loaded state from file D:\Working\LocalProjects\MQTT\zigbee2mqtt\data\state.json
[2024-05-25 17:18:54] info:     z2m: Starting Zigbee2MQTT version 1.37.1 (commit #cda867a3)
[2024-05-25 17:18:54] info:     z2m: Starting zigbee-herdsman (0.46.6)
[... cut ...]
    at ZnpAdapterManager.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\src\adapter\z-stack\adapter\manager.ts:86:17)
    at Controller.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\node_modules\zigbee-herdsman\src\controller\controller.ts:124:29)
    at Zigbee.start (D:\Working\LocalProjects\MQTT\zigbee2mqtt\lib\zigbee.ts:62:27)
```

In both cases, Z2M is not able to start correctly.

I hope the debug log can help find something wrong on my setup and I you have some other test that can help debug the problem I'm always available, thanks again

MartinVerges, 2024-10-07:
Mine won't work as well, but due to not using the correct serial port.

```
Starting Zigbee2MQTT without watchdog.
[2024-10-06 20:30:40] info:     z2m: Logging to console, file (filename: log.log)
[2024-10-06 20:30:40] debug:    z2m: Can't load state from file C:\Program Files\zigbee2mqtt\data\state.json (doesn't exist)
[2024-10-06 20:30:40] info:     z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-06 20:30:40] debug:    z2m: sd-notify loaded
[2024-10-06 20:30:40] info:     z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-06 20:30:40] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"C:\\\\Program Files\\\\zigbee2mqtt\\\\data\\\\database.db\",\"databaseBackupPath\":\"C:\\\\Program Files\\\\zigbee2mqtt\\\\data\\\\database.db.backup\",\"backupPath\":\"C:\\\\Program Files\\\\zigbee2mqtt\\\\data\\\\coordinator_backup.json\",\"serialPort\":{\"path\":\"COM6\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2024-10-06 20:30:41] error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat 'C:\Program Files\zigbee2mqtt\COM6''
[2024-10-06 20:30:41] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"COM6","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"C:\\Program Files\\zigbee2mqtt\\data\\database.db","databaseBackupPath":"C:\\Program Files\\zigbee2mqtt\\data\\database.db.backup","backupPath":"C:\\Program Files\\zigbee2mqtt\\data\\coordinator_backup.json"}'
[2024-10-06 20:30:41] info:     zh:zstack:znp: Opening SerialPort with {"path":"COM6","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-10-06 20:30:41] error:    z2m: Error while starting zigbee-herdsman
[2024-10-06 20:30:41] error:    z2m: Failed to start zigbee
[2024-10-06 20:30:41] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-06 20:30:41] error:    z2m: Exiting...
[2024-10-06 20:30:41] error:    z2m: Error: Opening COM6: Access denied
``` 

I tried it with `\\.\COM6` and just `COM6` but the path in the error message is always incorrect (`C:\Program Files\zigbee2mqtt\COM6`) and does not represent a Windows Serial device

Nerivec, 2024-10-07:
@MartinVerges Without the slashes is the best way, `COM3`, `COM6`, etc..
Since yours is failing with your installation in `Program Files` folder, and the error somehow adds `Program Files` in the port path, I'd say this has something to do with the fact that `Program Files` is a special folder, and might be messing with Z2M. As a first step, I'd suggest taking the installation out of that folder, and putting it in a regular folder, where you are sure Z2M has the correct permissions & such (like `C:\Users\User\yourusename\zigbee2mqtt`).

MartinVerges, 2024-10-07:
I tried to move it into c:\zigbee2mqtt but the path still gets prefixed to the COM6 port name.

```
PS C:\zigbee2mqtt> cat .\data\configuration.yaml
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  adapter: zstack
  port: COM6

advanced:
  log_level: debug
PS C:\zigbee2mqtt> npm start

> zigbee2mqtt@1.40.2 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2024-10-07 07:52:07] info:     z2m: Logging to console, file (filename: log.log)
[2024-10-07 07:52:07] debug:    z2m: Can't load state from file C:\zigbee2mqtt\data\state.json (doesn't exist)
[2024-10-07 07:52:07] info:     z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-07 07:52:07] debug:    z2m: sd-notify loaded
[2024-10-07 07:52:07] info:     z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-07 07:52:07] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"C:\\\\zigbee2mqtt\\\\data\\\\database.db\",\"databaseBackupPath\":\"C:\\\\zigbee2mqtt\\\\data\\\\database.db.backup\",\"backupPath\":\"C:\\\\zigbee2mqtt\\\\data\\\\coordinator_backup.json\",\"serialPort\":{\"path\":\"COM6\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false}}"'
[2024-10-07 07:52:08] error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat 'C:\zigbee2mqtt\COM6''
[2024-10-07 07:52:08] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"COM6","adapter":"zstack"},"adapter":{"disableLED":false},"databasePath":"C:\\zigbee2mqtt\\data\\database.db","databaseBackupPath":"C:\\zigbee2mqtt\\data\\database.db.backup","backupPath":"C:\\zigbee2mqtt\\data\\coordinator_backup.json"}'
[2024-10-07 07:52:08] info:     zh:zstack:znp: Opening SerialPort with {"path":"COM6","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-10-07 07:52:08] error:    z2m: Error while starting zigbee-herdsman
[2024-10-07 07:52:08] error:    z2m: Failed to start zigbee
[2024-10-07 07:52:08] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-07 07:52:08] error:    z2m: Exiting...
[2024-10-07 07:52:08] error:    z2m: Error: Opening COM6: Access denied
```

the same happens in the suggested User folder: `[2024-10-07 07:54:38] error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat 'C:\Users\Administrator\zigbee2mqtt\COM6''`

MartinVerges, 2024-10-08:
Hello,

thanks, it was indeed just a wrong port. I guess I got confused by the error message with the incorrect path that maybe should be fixed, besides that it's working great. Thank you so much!

Regards
Martin

closed 2025-01-01 by github-actions[bot]