Koenkk/zigbee2mqtt #22748: TRV602 (OLED) (vendor eweLink) is not working

michapr, 2024-05-23:
### What happened?

Have included device and see in about screen:

[image]

But got message
"Device '0x90ab96fffe721a13' with Zigbee model 'CK-MG22-TRVZB-01(7017)' and manufacturer name 'eWeLink' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html"

Device does not report values and setpoint etc. cannot be set. Battery value will be reported.

(made wrong topic before - it is not "new device", but TRV602 is supported - in general...

### What did you expect to happen?

Work like other TRV602 ;)

### How to reproduce it (minimal and precise)

include device only.

### Zigbee2MQTT version

 1.37.1 commit: cda867a3

### Adapter firmware version

7.4.1.0 build 0

### Adapter

Z-Wave.Me Z-Station dual Zigbee & Z-Wave

### Setup

Z2M plain, Pi3, bookworm

### Debug log

as sample:
```
2024-05-23T16:23:02.130462+02:00 pi90-200 node[112070]: [2024-05-23 16:23:02] #033[34mdebug#033[39m: #011z2m: Received Zigbee message from '0x90ab96fffe721a13', type 'attributeReport', cluster '64529', data '{"24585":1700}' from endpoint 1 with groupID 0
2024-05-23T16:23:02.131389+02:00 pi90-200 node[112070]: [2024-05-23 16:23:02] #033[34mdebug#033[39m: #011z2m: No converter available for 'CK-MG22-TRVZB-01(7017)' with cluster '64529' and type 'attributeReport' and data '{"24585":1700}'

```
[TRV602_converter_errors.log](https://github.com/Koenkk/zigbee2mqtt/files/15418357/TRV602_converter_errors.log)

[TRV602_device.log](https://github.com/Koenkk/zigbee2mqtt/files/15418359/TRV602_device.log)

michapr, 2024-06-06:
database entry:
```
{
    "id": 9,
    "type": "EndDevice",
    "ieeeAddr": "0x90ab96fffe721a13",
    "nwkAddr": 40761,
    "manufId": 4742,
    "manufName": "eWeLink",
    "powerSource": "Battery",
    "modelId": "CK-MG22-TRVZB-01(7017)",
    "epList": [1],
    "endpoints": {
        "1": {
            "profId": 260,
            "epId": 1,
            "devId": 769,
            "inClusterList": [0, 1, 3, 6, 32, 513, 64599, 64529],
            "outClusterList": [10, 25],
            "clusters": {
                "64529": {
                    "attributes": {
                        "0": 0,
                        "16": 513,
                        "24576": 0,
                        "24577": 1500,
                        "24578": 700,
                        "24579": 0,
                        "24580": 308,
                        "24581": 0,
                        "24582": 2604,
                        "24583": 1356,
                        "24584": 127,
                        "24585": 1250
                    }
                },
                "genBasic": {
                    "attributes": {
                        "appVersion": 0
                    }
                },
                "hvacThermostat": {
                    "attributes": {
                        "systemMode": 4,
                        "localTemp": 2160,
                        "occupiedHeatingSetpoint": 1250,
                        "runningState": 0,
                        "localTemperatureCalibration": 0
                    }
                },
                "genPollCtrl": {
                    "attributes": {
                        "checkinInterval": 14400
                    }
                },
                "genOnOff": {
                    "attributes": {
                        "onOff": 1
                    }
                },
                "genPowerCfg": {
                    "attributes": {
                        "batteryPercentageRemaining": 198
                    }
                }
            },
            "binds": [{
                    "cluster": 32,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x943469fffe0d65a9",
                    "endpointID": 1
                }, {
                    "cluster": 1,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x943469fffe0d65a9",
                    "endpointID": 1
                }, {
                    "cluster": 6,
                    "type": "endpoint",
                    "deviceIeeeAddress": "0x943469fffe0d65a9",
                    "endpointID": 1
                }
            ],
            "configuredReportings": [{
                    "cluster": 1,
                    "attrId": 33,
                    "minRepIntval": 3600,
                    "maxRepIntval": 65000,
                    "repChange": 10,
                    "manufacturerCode": null
                }, {
                    "cluster": 6,
                    "attrId": 0,
                    "minRepIntval": 0,
                    "maxRepIntval": 65000,
                    "repChange": 1,
                    "manufacturerCode": null
                }
            ],
            "meta": {}
        }
    },
    "appVersion": 0,
    "dateCode": "20230811",
    "swBuildId": "1.1.1",
    "zclVersion": 8,
    "interviewCompleted": true,
    "meta": {
        "configured": 332242049
    },
    "lastSeen": 1717654460858,
    "checkinInterval": 3600
}

```

@Koenkk : is there a chance to make an external definition? I have tried, but no success.
Or better to forget about this device? ;)

railin, 2024-09-12:
Hallo Michael,
konntest Du Deinen TRV CK-MG22-TRVZB-01 von eWeLink irgendwie einbinden, oder hast Du das Gerät in die Ecke gelegt?
Stehe vor dem gleichen Problem, da ich es als TRV602 gekauft habe, nun aber feststellen musste, dass es das gar nicht ist.

Danke für die Info, Rainer

michapr, 2024-09-13:
Hallo,
liegt *noch* in der Ecke - wartet darauf, dass ich nochmal versuche einen entsprechenden Konverter zu schreiben...
Kannst ja nochmal einen request für ein neues Device aufmachen, vielleicht erhört man uns diesmal oder es findet sich jemand, der das schon gemacht hat... ;)

railin, 2024-09-13:
Hallo Michael,
danke für die Info.
Werde das Teil auch erst einmal weg legen :-).

VG, Rainer

genmllc, 2024-11-02:
Hello, I got it mostly working using the Sonoff TRVZB as a model. Here's my external converter : 

```
const {
    Zcl
} = require('zigbee-herdsman');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const constants = require('zigbee-herdsman-converters/lib/constants');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const {
    battery,
    binary,
    bindCluster,
    commandsOnOff,
    customTimeResponse,
    deviceAddCustomCluster,
    enumLookup,
    forcePowerSource,
    humidity,
    iasZoneAlarm,
    numeric,
    occupancy,
    onOff,
    ota,
    temperature,
} = require('zigbee-herdsman-converters/lib/modernExtend');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {
    DefinitionWithExtend,
    Fz,
    KeyValue,
    KeyValueAny,
    ModernExtend,
    Tz
} = require('zigbee-herdsman-converters/lib/types');
const utils = require('zigbee-herdsman-converters/lib/utils');

const e = exposes.presets;
const ea = exposes.access;

// const sonoffExtend = {
    // weeklySchedule: (): ModernExtend => {
        // const exposes = e
            // .composite('schedule', 'weekly_schedule', ea.STATE_SET)
            // .withDescription(
                // 'The preset heating schedule to use when the system mode is set to "auto" (indicated with ⏲ on the TRV). ' +
                // "Up to 6 transitions can be defined per day, where a transition is expressed in the format 'HH:mm/temperature', each " +
                // 'separated by a space. The first transition for each day must start at 00:00 and the valid temperature range is 4-35°C ' +
                // '(in 0.5°C steps). The temperature will be set at the time of the first transition until the time of the next transition, ' +
                // "e.g. '04:00/20 10:00/25' will result in the temperature being set to 20°C at 04:00 until 10:00, when it will change to 25°C.",
            // )
            // .withFeature(e.text('sunday', ea.STATE_SET))
            // .withFeature(e.text('monday', ea.STATE_SET))
            // .withFeature(e.text('tuesday', ea.STATE_SET))
            // .withFeature(e.text('wednesday', ea.STATE_SET))
            // .withFeature(e.text('thursday', ea.STATE_SET))
            // .withFeature(e.text('friday', ea.STATE_SET))
            // .withFeature(e.text('saturday', ea.STATE_SET));

        // const fromZigbee: Fz.Converter[] = [{
            // cluster: 'hvacThermostat',
            // type: ['commandGetWeeklyScheduleRsp'],
            // convert: (model, msg, publish, options, meta) => {
                // const day = Object.entries(constants.thermostatDayOfWeek).find((d) => msg.data.dayofweek & (1 << +d[0]))[1];

                // const transitions = msg.data.transitions
                    // .map((t: {
[... 274 lines cut ...]
                'Note: Only version v1.1.4 or higher is supported.',
            valueMin: 0.0,
            valueMax: 100.0,
            valueStep: 1.0,
            unit: '%',
        }),
        //sonoffExtend.weeklySchedule(),
        customTimeResponse('1970_UTC'),
        ota(),
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatSystemMode(endpoint);
        await endpoint.read('hvacThermostat', ['localTemperatureCalibration']);
        await endpoint.read(0xfc11, [0x0000, 0x6000, 0x6002, 0x6003, 0x6004, 0x6005, 0x6006, 0x6007]);
    },
};

module.exports = definition;
```

I was not able to make the schedule feature work (this part is commented in the converter). Otherwise, most features seem to work fine.

michapr, 2025-01-02:
@genmllc - thanks, looks like it is working fine for me too. Will make more tests in next days.

koverg70, 2025-01-05:
> @genmllc - thanks, looks like it is working fine for me too. Will make more tests in next days.

Do you have firmware version 1.1.4? Mine is on 1.1.1 and do not know, how to update.

genmllc, 2025-01-12:
For version 2 of zigbe2mqtt : 
```
const {
    Zcl
} = require('zigbee-herdsman');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const constants = require('zigbee-herdsman-converters/lib/constants');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const {
    battery,
    binary,
    bindCluster,
    commandsOnOff,
    customTimeResponse,
    deviceAddCustomCluster,
    enumLookup,
    forcePowerSource,
    humidity,
    iasZoneAlarm,
    numeric,
    occupancy,
    onOff,
    ota,
    temperature,
} = require('zigbee-herdsman-converters/lib/modernExtend');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {
    DefinitionWithExtend,
    Fz,
    KeyValue,
    KeyValueAny,
    ModernExtend,
    Tz
} = require('zigbee-herdsman-converters/lib/types');
const utils = require('zigbee-herdsman-converters/lib/utils');

const e = exposes.presets;
const ea = exposes.access;

// const sonoffExtend = {
    // weeklySchedule: (): ModernExtend => {
        // const exposes = e
            // .composite('schedule', 'weekly_schedule', ea.STATE_SET)
            // .withDescription(
                // 'The preset heating schedule to use when the system mode is set to "auto" (indicated with ⏲ on the TRV). ' +
                // "Up to 6 transitions can be defined per day, where a transition is expressed in the format 'HH:mm/temperature', each " +
                // 'separated by a space. The first transition for each day must start at 00:00 and the valid temperature range is 4-35°C ' +
                // '(in 0.5°C steps). The temperature will be set at the time of the first transition until the time of the next transition, ' +
                // "e.g. '04:00/20 10:00/25' will result in the temperature being set to 20°C at 04:00 until 10:00, when it will change to 25°C.",
            // )
            // .withFeature(e.text('sunday', ea.STATE_SET))
            // .withFeature(e.text('monday', ea.STATE_SET))
            // .withFeature(e.text('tuesday', ea.STATE_SET))
            // .withFeature(e.text('wednesday', ea.STATE_SET))
            // .withFeature(e.text('thursday', ea.STATE_SET))
            // .withFeature(e.text('friday', ea.STATE_SET))
            // .withFeature(e.text('saturday', ea.STATE_SET));

        // const fromZigbee: Fz.Converter[] = [{
            // cluster: 'hvacThermostat',
            // type: ['commandGetWeeklyScheduleRsp'],
            // convert: (model, msg, publish, options, meta) => {
                // const day = Object.entries(constants.thermostatDayOfWeek).find((d) => msg.data.dayofweek & (1 << +d[0]))[1];

                // const transitions = msg.data.transitions
                    // .map((t: {
                        // heatSetpoint: number;transitionTime: number
[... 273 lines cut ...]
                'Note: Only version v1.1.4 or higher is supported.',
            valueMin: 0.0,
            valueMax: 100.0,
            valueStep: 1.0,
            unit: '%',
        }),
        //sonoffExtend.weeklySchedule(),
        customTimeResponse('1970_UTC'),
    ],
   ota: true
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat']);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatSystemMode(endpoint);
        await endpoint.read('hvacThermostat', ['localTemperatureCalibration']);
        await endpoint.read(0xfc11, [0x0000, 0x6000, 0x6002, 0x6003, 0x6004, 0x6005, 0x6006, 0x6007]);
    },
};

module.exports = definition;
```

Not much has changed, only the ota part.

genmllc, 2025-01-12:
> > @genmllc - thanks, looks like it is working fine for me too. Will make more tests in next days.
> 
> Do you have firmware version 1.1.4? Mine is on 1.1.1 and do not know, how to update.

Mine is 1.1.1 as well.

rantwolf, 2025-02-25:
> For version 2 of zigbe2mqtt :
>
> ```
> const {
>     Zcl
> } = require('zigbee-herdsman');
>
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

Hi. 
Thanks for your script.
It works basically but the 'ota' section creates an error:
```
error: 	z2m: Failed to call 'ExternalConverters' 'start' (/app/data/extension/ck-mg22-trvzb-01.js:352
    configure: async (device, coordinatorEndpoint) => {
    ^^^^^^^^^
SyntaxError: Unexpected identifier 'configure'
    at new Script (node:vm:117:7)
    at createScript (node:vm:269:10)
    at runInNewContext (node:vm:310:10)
    at ExternalConverters.loadModuleFromText (/app/lib/extension/externalJS.ts:187:24)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:52)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
```

After commenting out this section z2m is starting and and trv is working.
But after testing all functions of the valve I got another error:
```
z2m: Publish 'set' 'frost_protection_temperature' to '0x28dba7fffe56cd07' failed: 'Error: 'frost_protection_temperature' is not a number, got string ()'
```

closed 2025-05-05 by github-actions[bot]