Koenkk/zigbee2mqtt #22751: Ubysis C4 missing cluster

chris-1243, 2024-05-23:
### What happened?

I tried to configure an Ubysis C4 as described in the documentation.

I got this error:

```
[2024-05-23 15:54:57] error: 	z2m: Publish 'set' 'configure_device_setup' to 'name of the device' failed: 'Error: Cluster with name 'manuSpecificUbisysDeviceSetup' does not exist'
``` 

### What did you expect to happen?

Using Z2M 1.37.0 and everything went fine as expected. I guess there was an change between version 1.37.0 and 1.37.1

### How to reproduce it (minimal and precise)

Send a MQTT payload with 

```
{"configure_device_setup": {"input_action_templates": [{"type": "dimmer_single", "no_onoff_down": 1, "rate": 70, "input": 0}, {"type": "dimmer_single", "no_onoff_down": 1, "rate": 70, "input": 1}]}}
``` 
v1.37.1, I have the error
v1.37.0, no error and configuration seems to be taken

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20240315

### Adapter

Sonoff Dongle-P

### Setup

Docker container on an intel NUC

sjorge, 2024-05-25:
This should already be fixed in dev, see https://github.com/Koenkk/zigbee2mqtt/issues/22610

chris-1243, 2024-05-26:
I will have a check. 

I did not search the issues really well. Thanks

closed 2024-05-26 by chris-1243