Koenkk/zigbee2mqtt #22758: Increased CPU from version 1.37.0 with the Ember driver

Miesje, 2024-05-24:
### What happened?

Increased CPU from version 1.37.0 with the Ember driver.
I have tried it on two adapters both both have the same problem.

[image]

### What did you expect to happen?

nvt

### How to reproduce it (minimal and precise)

If you turn back Auto (EZSP) then the CPU decreases.

### Zigbee2MQTT version

1.37.0

### Adapter firmware version

7.4.2.0 build 0

### Adapter

ember

### Setup

Docker on Ubuntu with domoticz - Zigbee 3.0 USB Dongle Plus

### Debug log

nvt

Nerivec, 2024-05-28:
Is there an actual problem linked to this or it's just the CPU being a couple of points higher?
In all the tests I ran, the impact on CPU is minimal compared to the gain of having the protocols fully implemented (ASH/EZSP). Lots of stuff can affect CPU usage though, log level (`debug` being the most demanding ofc), a very quiet versus a very noisy timespan (less or more processing required)...

lubomirp, 2024-06-24:
the same here,
- SONOFF Zigbee 3.0 USB Dongle Plus V2 with ember 7.4.2.0
- Zigbee2MQTT version 1.38.0
 
after update approx. 5time higher CPU load
[image]

treitlis, 2024-07-10:
It seems that this is also an issue reported on the home assistant forums. 

https://community.home-assistant.io/t/skyconnect-v7-3-2-0-ezsp-to-ember-driver-depricated/730866/21?page=2

I have the same issue, running HA under proxmox. Using SkyConnect with the 7.4.3 firmware.
The CPU is always using around 2% of after enabling ember drivers. i switched back to ezsp, as for me a 25% increase in power budget doesn't seem to be worth it. (I went from 8W power usage to 10W power usage just by enabling the ember driver). 
At least in my naive view, I don't see any benefit of the ember driver. I have 7 zigbee devices on the network.

Giel538, 2024-07-29:
I can confirm this same problem. I have the Sonoff zigbee 3.0 stick with silicon labs chip. 
I made 2 screen prints:

ezsp driver:
[image]

ember driver:
[image]

With ember driver the cpu load is always 3.5 - 4.5% (from 2 cores of an intel i5-i13500 cpu, which is quite a lot).

Zigbee2MQTT versie
    [1.39.0]
Coördinator type
    EmberZNet
Coordinator revisie
    7.4.3 [GA]

In case you need (dutch) support to figure this out i am also available on t.net and search for user Mich.

Nerivec, 2024-07-29:
@Giel538 Can you report back with latest dev (or in a couple days after next release)? The system to process messages for ember is now aligned with zstack's (and with support for concurrency, and other fixes from v8 support). It should be "leaner and meaner", so-to-speak. 😉
Though in all the tests/profiling I've done, I've never seen this high a CPU usage, even before.
How many devices? And how many do you know to be spammers, if any?
What logging settings? Anything else that seems relevant in your setup?
I'll re-do a profiling with your settings, see if I can find a bottleneck somewhere.

Giel538, 2024-07-31:
@Nerivec I updated to the dev version and what i can say after 30 minutes running it seems to be fixed.  I will check again tomorrow morning and let you know.

lubomirp, 2024-08-01:
[image]
with dev version it seems to be much more better:)

Giel538, 2024-08-01:
Yes it is definitely a lot better. I think its more or less the same cpu usage as the ezsp driver.

closed 2024-08-01 by Koenkk