Koenkk/zigbee2mqtt #22763: MQTT InvalidStateError with length 320

contactmetherenow, 2024-05-24:
### What happened?

I brought up this issue on the HA core github and posted:
"I noticed an issue in my log files regarding the following: homeassistant.exceptions.InvalidStateError: Invalid state with length 320. State max length is 255 characters. This is in regards to a ir blaster device (ZS06) using MQTT. The device does function properly even though I receive this problem but I wanted to see a resolution can be found to eliminate the error message that occurs any time i restart my HA."

After reviewing the issue I was informed that the issue was in regards to Z2M as shown:
"The issue is that a sensor or text entity cannot have a state with more then 255 chars. If the length exceeds this limit we should log a better error message showing the entity causing the issue. That the state exceeds the limit is not an issue of MQTT. This is just a bad configuration."

The discussion and the details about this issue was originally first reported on:
https://github.com/home-assistant/core/issues/117861

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20240501.1

### Adapter

Tuya ZS06

### Setup

Docker container ron x86-64

contactmetherenow, 2024-05-25:
Let me know if any more information is needed.

closed 2024-05-26 by contactmetherenow

HWiese1980, 2024-05-27:
Completed? How?