Koenkk/zigbee2mqtt #22766: Ubisys C4 does not report all commmand to coordinator

chris-1243, 2024-05-24:
### What happened?

Commands `brightness_move_down` and `brightness_stop` are not reported to the coordinator. If the device is bound to a light, all commands are sent correctly and the dimming works as expected.

I do use `dimmer_single` as configuration for the two first inputs.

My S2 are working fine and do report all commands.

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

20240315

### Adapter

Sonoff Dongle-P

### Setup

Docker container

### Debug log

[herdsmann-debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/15435596/herdsmann-debug.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/15435597/log.txt)

sjorge, 2024-05-24:
What do your bindings look like ? 

Old fw revisions came pre-bound to the coordinator and we just fill in the info. I wonder if newer device do not do this and we need to actually bind them. (That used to fail)

chris-1243, 2024-05-24:
My bindings look like yours. All the endpoints are bound to the coordinator

[image]

(Sorry for the quality, cell phone screenshot)

What strange, I receive via mqtt `brightness_move_up` but not the `down` or `stop`...
As I have a spare one, I had a try with a Domgle-E and `ember` driver. Same result.

closed 2024-10-02 by chris-1243