Koenkk/zigbee2mqtt #22773: Enable/disable devices using automation script

E30M20B20, 2024-05-25:
### Is your feature request related to a problem? Please describe

If I switch some devices on in a pair times a year, I can do this manually using checkbox in device's settings page. But if I have several losses of power supply a day (like I have it now in Kiev), it would be very useful to disable/enable devices automatically when 220V lost/reconnects

### Describe the solution you'd like

So I need an instrument to check/uncheck "disable" checkbox at device's settings using automation script, and next is restart z2m using existing feature, and here we go!

### Describe alternatives you've considered

Any kind of automatic switch on/off pings to router devices using automation script

### Additional context

[image]

ghoz, 2024-06-03:
Hi
Not tested but it might already be possible by sending an mqtt message to `zigbee2mqtt/bridge/request/device/options` with `disabled: true` as option

See:  https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

E30M20B20, 2024-09-27:
Yes, it works! Payload must be {"id": "friendly name of device", "options":{"disabled":1}}
Thank you!

closed 2024-09-27 by E30M20B20

E30M20B20, 2024-09-29:
{"id": "friendly name of device", "options":{"disabled":true}} is better. Works both variants, but with "disabled":1 z2m do not start with errors "... must be boolean"