Koenkk/zigbee2mqtt #22777: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT

smartmatic, 2024-05-25:
### What happened?

One week ago i have update to ember and flashed the firmware 7.4.2 on my Sonoff Dongle-E. Today i am no longer able to add two new IKEA devices. Interview can not finished.

### What did you expect to happen?

Add device with successful interview

### How to reproduce it (minimal and precise)

1. Flash 7.4.2
2. Set adapter to ember
3.  Try to add IKEA Barding or IKEA Parasoll

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

7.4.2

### Adapter

ember

### Setup

Home Assistant Add-On

### Debug log

`[2024-05-25 22:41:31] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":3} after 4000ms
[2024-05-25 22:41:31] error: 	zh:ezsp:uart: -!- break waiting (3)
[2024-05-25 22:41:31] error: 	zh:ezsp:uart: Can't send DATA frame (2,7,0): 3a000164000efdff0100
[2024-05-25 22:41:35] error: 	zh:ezsp:uart: --> Error: Error: {"sequence":3} after 4000ms
[2024-05-25 22:41:35] error: 	zh:ezsp:uart: -!- break rewaiting (3)
[2024-05-25 22:41:35] error: 	zh:ezsp:uart: Can't resend DATA frame (2,7,1): 3a000164000efdff0100
[2024-05-25 22:41:35] error: 	zh:ezsp:cast: No more available slots MulticastId subscription
[22:42:13] INFO: Preparing to start...
[22:42:14] INFO: Socat not enabled
[22:42:17] INFO: Starting Zigbee2MQTT...
[2024-05-25 22:42:28] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-05-25 22:42:28] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-25 22:42:28] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-05-25 22:42:33] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":41440,"clusterId":33,"sourceEndpoint":242,"destinationEndpoint":242,"options":256,"groupId":0,"sequence":210} messageTag=255]
[2024-05-25 22:47:40] error: 	z2m: Publish 'set' 'state' to 'Zwischenstecker E-MTB' failed: 'Error: ZCL command 0xa4c1386609828f2a/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed ({"target":22161,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":2},"zclSequence":3,"commandIdentifier":11} timed out after 10000ms)'
[2024-05-25 22:48:40] error: 	zh:ember: Delivery of BROADCAST failed for "65533" [apsFrame={"profileId":0,"clusterId":19,"sourceEndpoint":0,"destinationEndpoint":0,"options":1024,"groupId":0,"sequence":37} messageTag=255] `

smartmatic, 2024-05-26:
Today i am able to add IKEA Parasoll and i recorded it via debug log.
But there is a an Info Error inside the log: 
2024-05-26 10:48:37Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "62659".
[45df7312_zigbee2mqtt_2024-05-26T08-49-48.823Z.log](https://github.com/Koenkk/zigbee2mqtt/files/15446812/45df7312_zigbee2mqtt_2024-05-26T08-49-48.823Z.log)

Koenkk, 2024-05-27:
You are using the old `ezsp` driver, it's recommended to use the new `ember` drivers ([info](https://github.com/Koenkk/zigbee2mqtt/discussions/21462))

smartmatic, 2024-05-28:
No, i switched to ember

[image]

Nerivec, 2024-06-10:
Can you confirm this interview issue is still present with 1.38? If yes, please provide a full `debug` log of the pairing procedure.

smartmatic, 2024-06-12:
I've deleted the device and re-added it without the reported error from this issue. So it is solved!

Thank you!

closed 2024-06-12 by Nerivec

ka0r5, 2024-09-16:
I also encountered this error with SLZB-06M adapter, but solved by rebooting it (by clicking Reboot in the UI)

zer013, 2025-01-05:
In my case, for it to work again I need to completely turn off the raspberry otherwise it won't work again. I think the problem may even be energy