Koenkk/zigbee2mqtt #22779: Kiwkset Convert Lock status not updating

monogoat, 2024-05-26:
### What happened?

I have had two of these locks (Amazon Edition Kwikset Convert, shows up as Model 66492-001 in z2m) working with zigbee2mqtt for several months without issue. While messing with the configuration file my install went in to "recovery mode" (aka delete everything mode) and I did not have a recent enough full backup to restore everything and took the opportunity to upgrade from a Pi4 to a Pi5 and move the install from 32 to 64bit for more addon compatibility. Have started the install fresh from scratch.

I am trying to get the locks working again. Using the same adapter (CC26X2R1), I have gotten a lock paired and it can lock and unlock but the state does not update in the z2m dashboard or home assistant. It also does not update if I lock or unlock it manually. It will update if you click the refresh button (and I went ahead and made an automation to send the service call to update every 5 minutes for now).

This was working perfectly fine before, but I have no access to the prior configuration to see if anything is different. It is possible that it was on the previous version of z2m. I did find this in the debug logs that seems to point to z2m as being the problem:

`[2024-05-25 19:41:31] debug: 	z2m: Received Zigbee message from '0x002446fffd034e8b', type 'commandUnlockDoorRsp', cluster 'closuresDoorLock', data '{"status":0}' from endpoint 2 with groupID 0
[2024-05-25 19:41:31] debug: 	z2m: No converter available for '66492-001' with cluster 'closuresDoorLock' and type 'commandUnlockDoorRsp' and data '{"status":0}'`
`[2024-05-25 19:45:35] debug: 	zh:controller: Received payload: clusterID=257, address=53415, groupID=0, endpoint=2, destinationEndpoint=1, wasBroadcast=false, linkQuality=114, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":102,"commandIdentifier":0},"payload":{"status":0},"command":{"ID":0,"parameters":[{"name":"status","type":32}],"name":"lockDoorRsp"}}
[2024-05-25 19:45:35] debug: 	zh:controller: Skipping command 'lockDoorRsp' because it is missing from the lookup
[2024-05-25 19:45:35] debug: 	zh:zstack:unpi:parser: --- parseNext []`

### What did you expect to happen?

I expected the locks to properly update their state as they have been in the past.

### How to reproduce it (minimal and precise)

Pair Kwikset Convert zigbee lock to system and lock or unlock it.

### Zigbee2MQTT version

1.37.1-1

### Adapter firmware version

20210120

### Adapter

CC26X2R1

### Setup

64bit Hassio Add-on, Pi 5 8gb

### Debug log

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15445484/log.log)

monogoat, 2024-05-26:
Just noticed that most recent log is empty, here is one with a lot of stuff. One of the locks is the only thing in the zigbee network here:

[log.log](https://github.com/Koenkk/zigbee2mqtt/files/15445494/log.log)

monogoat, 2024-05-26:
Quick update, was looking into another issue I had that I thought was unrelated and decided to try eqmx for mqtt instead of mosquito, and lo and behold the lock status seems to updating perfectly. It is late so I will be playing with it in the morning to be sure the issue is completely gone and I'll close this.

jacobefaust, 2024-11-18:
I am also running into this issue on two Kwikset Convert Locks (64492-001):
 - Mosquitto 6.4.1
 - Zigbee2Mqtt 1.41.0-1

I can remotely control the lock, but the status is not reporting up and making it into my server's logs.here 

Are there any logs I can share that would be useful to start debugging @dale3h ?

closed 2025-01-25 by github-actions[bot]

dschaper, 2025-08-07:
Can we reopen this issue?

I have this lock and I see the same effects. Triggering a lock will update everything visually to show lock status. Triggering an unlock will unlock but it doesn't update any visual status, all toggles and states show locked. If you refresh (click on the cycle arrows) then the status updates.

I have a snippet of logs below and what stood out to me is that every working function, manual_lock, manual_unlock, and lock have an `/action` payload showing the command. Triggering and unlock does not have that topic published. I've added a comment to the log that shows where I think there should be a payload topic for the unlock command.

```
[2025-08-07 15:10:55] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action":"manual_unlock","action_source":2,"action_source_name":"manual","action_user":65535,"battery":70,"last_seen":"2025-08-07T22:10:54+00:00","linkquality":112,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:10:55] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7/action', payload 'manual_unlock'

[2025-08-07 15:11:16] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action":"manual_lock","action_source":2,"action_source_name":"manual","action_user":65535,"battery":70,"last_seen":"2025-08-07T22:11:15+00:00","linkquality":112,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:11:16] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7/action', payload 'manual_lock'

[2025-08-07 15:11:31] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action_source_name":null,"action_user":null,"battery":70,"last_seen":"2025-08-07T22:11:31+00:00","linkquality":100,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:12:02] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action_source_name":null,"action_user":null,"battery":70,"last_seen":"2025-08-07T22:12:01+00:00","linkquality":92,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:12:43] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action_source_name":null,"action_user":null,"battery":70,"last_seen":"2025-08-07T22:12:43+00:00","linkquality":88,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:12:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action":"lock","action_source":1,"action_source_name":"rf","action_user":65535,"battery":70,"last_seen":"2025-08-07T22:12:48+00:00","linkquality":72,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:12:49] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7/action', payload 'lock'

[2025-08-07 15:13:40] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action_source_name":null,"action_user":null,"battery":70,"last_seen":"2025-08-07T22:13:40+00:00","linkquality":92,"lock_state":"locked","sound_volume":"silent_mode","state":"LOCK","voltage":5000}'

[2025-08-07 15:14:07] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action_source_name":null,"action_user":null,"battery":70,"last_seen":"2025-08-07T22:14:06+00:00","linkquality":88,"lock_state":"unlocked","sound_volume":"silent_mode","state":"UNLOCK","voltage":5000}'

### Note: No action paylod of 'unlocked'

[2025-08-07 15:14:24] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7', payload '{"action":"manual_lock","action_source":2,"action_source_name":"manual","action_user":65535,"battery":70,"last_seen":"2025-08-07T22:14:24+00:00","linkquality":96,"lock_state":"unlocked","sound_volume":"silent_mode","state":"UNLOCK","voltage":5000}'

[2025-08-07 15:14:24] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x002446fffd038aa7/action', payload 'manual_lock'
```