Koenkk/zigbee2mqtt #22782: Ember not working for Sonoff E dongle

robinwo, 2024-05-26:
### What happened?

Runs in a docker container, on a Synology NAS (with USB drivers installed & enabled).
When I set adapter to 'ezsp' it works, when changing this to 'ember' it does not (see below).

```
[2024-05-26 10:08:09] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-26 10:08:09] info: 	z2m: Starting Zigbee2MQTT version 1.37.1 (commit #ea39d86)
[2024-05-26 10:08:09] info: 	z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-26 10:08:09] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-05-26 10:08:09] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-05-26 10:08:09] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-26 10:08:09] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-05-26 10:08:09] info: 	zh:ember:uart:ash: Serial port opened
[2024-05-26 10:08:09] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-26 10:08:09] error: 	zh:ember:uart:ash: Received frame with CRC error
[2024-05-26 10:08:09] error: 	zh:ember:uart:ash: Received ERROR from NCP while connecting, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2024-05-26 10:08:09] error: 	zh:ember:uart:ash: ASH disconnected | NCP status: ASH_NCP_FATAL_ERROR
[2024-05-26 10:08:09] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-05-26 10:08:09] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2024-05-26 10:08:09] info: 	zh:ember:uart:ash: ======== ASH NCP reset ========
[2024-05-26 10:08:09] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-05-26 10:08:10] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-05-26 10:08:10] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-05-26 10:08:10] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-05-26 10:08:10] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-26 10:08:10] error: 	z2m: Failed to start zigbee
[2024-05-26 10:08:10] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-26 10:08:10] error: 	z2m: Exiting...
[2024-05-26 10:08:11] error: 	z2m: Error: NCP EZSP protocol version of 8 does not match Host version 13
    at EmberAdapter.emberVersion (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1633:19)
    at EmberAdapter.initEzsp (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:784:9)
    at EmberAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:2642:24)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)
```

### Zigbee2MQTT version

1.37.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EZSP v8 (SONOFF Zigbee 3.0 USB Dongle Plus V2)

### Setup

Synology DSM7.2, Docker container

francisp2, 2024-05-26:
>[2024-05-26 10:08:11] error: 	z2m: Error: NCP EZSP protocol version of 8 does not match Host version 13

You need to update the firmware of the adapter to 7.4.1 or 7.4.3

distante, 2024-05-27:
> > [2024-05-26 10:08:11] error: 	z2m: Error: NCP EZSP protocol version of 8 does not match Host version 13
> 
> You need to update the firmware of the adapter to 7.4.1 or 7.4.3

In the official docs says that just 7.4.1 is supported

https://www.zigbee2mqtt.io/guide/adapters/emberznet.html

closed 2024-06-10 by Nerivec