Koenkk/zigbee2mqtt #22784: [New device support]: unsupported device manufacturer with Zigbee model 'TS0601' and manufacturer name '_TZE284_cjbofhxw'

lyonelf, 2024-05-26:
### Link

https://fr.aliexpress.com/item/1005006942840986.html?spm=a2g0o.order_list.order_list_main.17.21ef5e5bU2MF9s&gatewayAdapt=glo2fra

### Database entry

Where find it?

### Comments

Device '0x842712fffe481b72' with Zigbee model 'TS0601' and manufacturer name '_TZE284_cjbofhxw' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### External definition

```shell
..
```

Boba1982, 2024-05-31:
hi,

i have same device, an cannot integrate.
can someone please integrate it

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_cjbofhxw',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},

lyonelf, 2024-06-01:
> hi,
> 
> i have same device, an cannot integrate. can someone please integrate it
> 
> const definition = { zigbeeModel: ['TS0601'], model: 'TS0601', vendor: '_TZE284_cjbofhxw', description: 'Automatically generated definition', extend: [], meta: {},

I found the external unsupported settings in another page, here is:

`const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    gateway_connection_status: {
        cluster: 'manuSpecificTuya',
        type: ['commandMcuGatewayConnectionStatus'],
        convert: async (model, msg, publish, options, meta) => {
            const payload = {payloadSize: 1,payload: 1}
            await msg.endpoint.command('manuSpecificTuya', 'mcuGatewayConnectionStatus', payload, {});
        },
    },
}

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'PJ-1203-W',
    vendor: 'TuYa',
    description: 'Electricity Energy Monitor',
    extend: [],
    fromZigbee: [tuya.fz.datapoints, fzLocal.gateway_connection_status],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [e.current(), e.power(), e.voltage(), e.energy()],
    meta: {
                tuyaDatapoints: [
            [18, 'current', tuya.valueConverter.divideBy1000],
            [19, 'power', tuya.valueConverter.divideBy10],
            [20, 'voltage', tuya.valueConverter.divideBy10],
            [101, 'energy', tuya.valueConverter.divideBy1000],
        ],  
    },
};

module.exports = definition;
`

Koenkk, 2024-06-01:
@lyonelf does that external converter work?

lyonelf, 2024-06-01:
Yes, my clamp is in product via Jeedom

Koenkk, 2024-06-02:
Added it, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-06-02 by Koenkk

Boba1982, 2024-06-02:
thank you,
i try to add custom quirk but with no luck.
my clamp is still unsupported.
this is my custum quirk:

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
gateway_connection_status: {
cluster: 'manuSpecificTuya',
type: ['commandMcuGatewayConnectionStatus'],
convert: async (model, msg, publish, options, meta) => {
const payload = {payloadSize: 1,payload: 1}
await msg.endpoint.command('manuSpecificTuya', 'mcuGatewayConnectionStatus', payload, {});
},
},
}

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_cjbofhxw',
    description: 'Automatically generated definition',
extend: [],
fromZigbee: [tuya.fz.datapoints, fzLocal.gateway_connection_status],
toZigbee: [tuya.tz.datapoints],
configure: tuya.configureMagicPacket,
exposes: [e.current(), e.power(), e.voltage(), e.energy()],
meta: {
tuyaDatapoints: [
[18, 'current', tuya.valueConverter.divideBy1000],
[19, 'power', tuya.valueConverter.divideBy10],
[20, 'voltage', tuya.valueConverter.divideBy10],
[101, 'energy', tuya.valueConverter.divideBy1000],
],
},
};

module.exports = definition;

and i input this lines in HA Config.yaml

zha:
  database_path: /config/zigbee.db 
  enable_quirks: true
  custom_quirks_path: /config/custom_zha_quirks/